nickg/nvc #854: *** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x56276b03e762] ***

m-kru, 2024-02-29:
```
*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x56276b03e762] ***

[0x56276b08e02d] ../src/util.c:870 signal_handler
[0x7faef35e908f] (/usr/lib/x86_64-linux-gnu/libc-2.31.so)
[0x56276b03e762] /usr/include/x86_64-linux-gnu/bits/stdio2.h:578 tree_kind
[0x56276b03e762] ../src/tree.c:469 find_element_mode_indication
[0x56276b016b5a] ../src/lower.c:11441 lower_map_view_field_cb
[0x56276b038da1] ../src/lower.c:903 lower_for_each_field.lto_priv.0
[0x56276b016c47] ../src/lower.c:11474 lower_map_view_field_cb
[0x56276b038da1] ../src/lower.c:903 lower_for_each_field.lto_priv.0
[0x56276b01d234] ../src/lower.c:11777 lower_port_map.lto_priv.0
[0x56276b050731] ../src/eval.c:12190 lower_instance
[0x56276b050731] ../src/lower.c:1043 elab_lower.lto_priv.0
[0x56276b04e711] ../src/elab.c:1286 elab_architecture
[0x56276b04bb82] ../src/elab.c:1359 elab_stmts
[0x56276b04e13e] ../src/elab.c:1866 elab_top_level.lto_priv.0
[0x56276b092bed] ../src/names.c:1939 elab
[0x56276b092bed] ../src/elab.c:464 elaborate
[0x56276b093f60] ../src/nvc.c:1921 process_command
[0x56276afa44f1] ../src/nvc.c:2059 main

Please report this bug at https://github.com/nickg/nvc/issues
nvc::elaborate: tb_two_requesters_one_completer elaboration failed with exit status 1
```

nickg, 2024-02-29:
Can you provide a minimal example or a link to the code that causes this?

m-kru, 2024-03-01:
https://github.com/m-kru/nvc854 not a minimal, but crashes. Simply run `run.sh`.

commit 696495d1 (2024-03-02): Fix crash when view element indication references whole record

Fixes #854
closed 2024-03-02 by nickg