nickg/nvc #857: cocotb stop working

scolabirra, 2024-03-05:
I recently update to the latest development commit and my testbench in cocotb doesn't work anymore. The error that I get is:

`4225.00ns ERROR    cocotb.scheduler                   Failing test at simulator request before test run completion: Simulator shut down prematurely`

I did some regression and the issue appeared with commit b52c5763 . Versions before this commit doesn't shown the issue.
I looked at the VHPI debug information but there is no difference between versions.

nickg, 2024-03-06:
> 
> I did some regression and the issue appeared with commit [b52c576](https://github.com/nickg/nvc/commit/b52c57635f1a48d0a5d9df224fb72366eacc2439) . Versions before this commit doesn't shown the issue. 

I fixed the regression introduced by this commit, thanks for tracking it down. Could you test again? 

However I also just ran the cocotb test suite and noticed there were still a few failures which I need to investigate.

scolabirra, 2024-03-07:
All my tests work again, I don't see any failure. Thanks for fixing it.

nickg, 2024-03-25:
All the cocotb regression tests are passing now.

closed 2024-03-25 by nickg