nickg/nvc #866: OSVVM co-simulation run-time crash on loading shared object

wyvernSemi, 2024-03-13:
Running the OSVVM co-simulation tests causes a crash on loading <tt>VProc.so</tt> foreign code. The output is as shown below:

**MSYS2**:

```
nvc --std=08 -H 128m --work=osvvm_CoSim_TbAxi4Lite:C:/git/OSVVM/nvc/VHDL_LIBS/nvc-1.12-devel/OSVVM_COSIM_TBAXI4LITE.08 -L C:/git/OSVVM/nvc/VHDL_LIBS/nvc-1.12-devel -r --ieee-warnings=off --load=./VProc.so TbAb_CoSim
Error: ** Note: loading VHPI plugin ./VProc.so
Error: ** Fatal: (init): unsupported constraint subkind 2
Error: [00007ff77f52c1ef]
Error: [00007ff77f5e5805]
Error: [00007ff77f522164]
Error: [00007ff77f6248f3] _nvc_get_object+0x106d3
Error: [00007ff77f624b1b] _nvc_get_object+0x108fb
Error: [00007ff77f624c95] _nvc_get_object+0x10a75
Error: [00007ff77f62877c] _nvc_get_object+0x1455c
Error: [00007ff77f52a835]
Error: [00007ff77f5259d4]
Error: [00007ff77f65b4ee] vhpi_is_printable+0x2f7ee
Error: [00007ff77f5212ee]
Error: [00007ff77f521406]
Error: [00007ffb26fd7344] BaseThreadInitThunk+0x14
Error: [00007ffb283026b1] RtlUserThreadStart+0x21
Error:
Error: Please report this bug at https://github.com/nickg/nvc/issues
Error: child process exited abnormally
```

**Linux**:

```
nvc --std=08 -H 128m --work=osvvm_CoSim_TbAxi4Lite:/home/simon/OSVVM/nvc/VHDL_LIBS/nvc-1.12-devel/OSVVM_COSIM_TBAXI4LITE.08 -L /home/simon/OSVVM/nvc/VHDL_LIBS/nvc-1.12-devel -r --ieee-warnings=off --load=./VProc.so TbAb_CoSim
Error: ** Note: loading VHPI plugin ./VProc.so
Error: ** Fatal: (init): unsupported constraint subkind 2
Error: [0x5587780ab02c] ../src/diag.c:1011 diag_femit
Error: [0x5587780ab02c] ../src/diag.c:1036 diag_emit
Error: [0x558777fe5e92] ../src/util.c:585 fatal_trace
Error: [0x5587780f09f8] ../src/vhpi/vhpi-model.c:3330 build_typeDecl
Error: [0x5587780f0bda] ../src/vhpi/vhpi-model.c:3350 cached_typeDecl
Error: [0x5587780f0bda] ../src/vhpi/vhpi-model.c:789 init_objDecl
Error: [0x5587780f0d72] ../src/vhpi/vhpi-model.c:3384 vhpi_build_decls
Error: [0x5587780f4844] ../src/vhpi/vhpi-model.c:3572 vhpi_build_design_model
Error: [0x5587780f4844] ../src/vhpi/vhpi-model.c:3583 vhpi_initialise_cb.lto_priv.0
Error: [0x558777ff01ab] ../src/rt/model.c:425 model_reset
Error: [0x558777ff01ab] ../src/rt/model.c:836 run_cmd
Error: [0x558777fe9c91] ../src/nvc.c:1923 process_command
Error: [0x558777fe3715] ../src/nvc.c:2059 main
Error: 
Error: Please report this bug at https://github.com/nickg/nvc/issues
Error: child process exited abnormally
```
Versions Used:

- OsvvmLibraries: main branch head as of 13th Mar 2024 (SHA 6545fde...)
- NVC (MSYS2): nvc 1.12-devel (1.11.0.r146.g736b6381) (Using LLVM 15.0.7)
- NVC (Linux): nvc 1.12-devel (1.11.0.r146.g736b6381) (Using LLVM 14.0.0)

I have prepared and attached a bundle that reproduces this error (see attached file at the end). It has the test bench VHDL (<tt>.vhd</tt> files in top level directory) and the <tt>VProc.so</tt> source code (in <tt>code/</tt> directory) and a <tt>compile.sh</tt> script to compile and run the C/C++ and VHDL. The <tt>VHDL_LIB</tt> directory contains precompiled OSVVM libraries. A precompiled <tt>VUser.so</tt> shared object, that would be loaded by <tt>VProc.so</tt>, is also included which is a co-simulation test that should run and pass if all else is functioning. This is compiled for MSYS2/mingw-w64.

I can't say for sure when this error was introduced. There have been no changes to OSVVM co-simulation code since October when fixes were required to track changes in the OSVVM VHDL, implying CoSim tests are not regularly run upon OSVVM releases, and I can't say for sure whether NVC tests were run at that point. My records show that NVC was passing the CoSim tests on 5th June last year.

I am about to start adding new features to OSVVM co-simulation and am trying to get all supported platforms running with the latest OSVVM code before embarking on this, and am keen to get NVC fully operational again. The tests pass for GHDL, Questa and others.

[nvc.tar.gz](https://github.com/nickg/nvc/files/14586729/nvc.tar.gz)

nickg, 2024-03-14:
Fixed by the commit above:

```
%% Log    PASSED    in manager_1: WriteResponse Scoreboard,   Received: 0   Item Number: 427 at 16620 ns
%% DONE   PASSED   CoSim_usercode_size  Passed: 800  Affirmations Checked: 800  at 16670 ns
```

closed 2024-03-14 by nickg

wyvernSemi, 2024-03-15:
All OSVVM Co-simulation tests now passing. Thanks for a quick turn around.