nickg/nvc #867: Fatal: cannot use uarray left with non-uarray type

Anselmo95, 2024-03-14:
Hello,

I got this stack trace while elaborating a design:

```
      etc...
      invalid := uarray left r90 dim 0	 <----

** Fatal: cannot use uarray left with non-uarray type
[0x491fee] ../src/diag.c:1016 diag_femit
[0x491fee] ./../src/diag.c:1040 diag_emit
[0x53988d] ./../src/util.c:585 fatal_trace
[0x4ac6a3] ./../src/vcode.c:4557 emit_uarray_op.lto_priv.515
[0x4aec1b] ./../src/vcode.c:4571 lower_range_left.part.27.lto_priv.841
[0x4d0b85] ../src/lower.c:3905 lower_array_aggregate
[0x4d0b85] ./../src/lower.c:4471 lower_aggregate.lto_priv.486
[0x4c85bf] ./../src/lower.c:5366 lower_expr.lto_priv.460
[0x4b3569] ./../src/lower.c:12428 lower_rvalue
[0x4cc328] ./../src/lower.c:1124 lower_subprogram_arg.lto_priv.815
[0x4cdf88] ./../src/lower.c:2581 lower_fcall.lto_priv.481
[0x4c855f] ./../src/lower.c:5356 lower_expr.lto_priv.460
[0x4b3569] ./../src/lower.c:12428 lower_rvalue
[0x4c1614] ./../src/lower.c:6411 lower_signal_assign
[0x4c4e5a] ./../src/lower.c:7506 lower_sequence.lto_priv.781
[0x4c4f8d] ./../src/lower.c:6527 lower_sequence.lto_priv.781
[0x4ec4ce] ./../src/lower.c:11204 elab_stmts.lto_priv.881
[0x4ec0fd] ./../src/elab.c:1397 elab_stmts.lto_priv.881
[0x4ec0fd] ./../src/elab.c:1397 elab_stmts.lto_priv.881
[0x4ec0fd] ./../src/elab.c:1397 elab_stmts.lto_priv.881
[0x4ec0fd] ./../src/elab.c:1397 elab_stmts.lto_priv.881
[0x4edcf4] ./../src/elab.c:1884 elab_top_level
[0x4ede63] ./../src/elab.c:1986 elab
[0x544a9a] ./../src/nvc.c:457 elaborate
[0x54610e] ./../src/nvc.c:1909 process_command.lto_priv.1056
[0x411c59] ./../src/nvc.c:2047 main
```

I tried to bisect and find which commit caused the regression which gave me this commit 33ba293b302a651ec0c42fee1a7345907f40b0c5, but I am a bit unsure becaue even the commit before this one fails to complete the elaboration, just with a different error.

Unfortunately I cannot share the original code and I couldn't find a way to reproduce the issue with something i can share.
I'll keep trying to create a reproducer but if you have suggestion on how to proceed or something you want me to try just let me know.

nickg, 2024-03-14:
Could you try with the patch below? That might give some clue where the error comes from.

```diff
diff --git a/src/lower.c b/src/lower.c
index 1d75c9d887cd..a61dda874db3 100644
--- a/src/lower.c
+++ b/src/lower.c
@@ -321,6 +321,9 @@ static vcode_reg_t lower_range_expr(lower_unit_t *lu, tree_t r, int *dim)
    if (prefix_reg == VCODE_INVALID_REG)
       return lower_get_type_bounds(lu, type);
 
+   if (vcode_reg_kind(prefix_reg) != VCODE_TYPE_UARRAY)
+      fmt_loc(stdout, tree_loc(r));
+
    assert(vcode_reg_kind(prefix_reg) == VCODE_TYPE_UARRAY);
    return prefix_reg;
 }
```

Anselmo95, 2024-03-14:
Thanks for the quick reply! I tried the patch and it highligthed a certain line. This is a snippet containing the logic reported by your patch, unfortunately i cannot reproduce the problem with just this little part of the code but i hope it gives you some clues nonetheless.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity a is
generic (
    N : natural := 5
);
end entity a;

architecture rtl of a is
    signal clk   : std_logic := '0';
    signal reset : std_logic := '1';
    signal a : std_logic;
    signal b : std_logic;
    signal c : unsigned(N-1 downto 0);
    signal d : unsigned(N-1 downto 0);
    signal e : unsigned(N-1 downto 0);
    signal f : std_logic_vector(N-1 downto 0);
    signal g : unsigned(N-1 downto 0);
begin

    clk   <= not clk after 1 ns;
    reset <= '0' after 2 ns;

    process
    begin
        a <= '0';
        b <= '0';
        wait for 3 ns;
        a <= '1';
        b <= '0';
        wait for 3 ns;
        a <= '0';
        b <= '1';
        wait for 3 ns;
        a <= '1';
        b <= '1';
        wait for 3 ns;
        wait;
    end process;

    d <= (others => a);
    f <= (others => b);
    e <= unsigned(f);
    c <= (others => '0') when (a = '1' and b = '1' and f(N-1) = '1') else
         (e - d) when (a = '1' and b = '1') else
         e and (e'range => a); -- This (e'range => a) is the highligthed part

    process(clk)
    begin
        if rising_edge(clk) then
            if (reset = '1') then
                g <= (others => '0');
            else
                g <= c;
            end if;
        end if;
    end process;

end architecture;

```

nickg, 2024-03-14:
> This is a snippet containing the logic reported by your patch, unfortunately i cannot reproduce the problem with just this little part of the code but i hope it gives you some clues nonetheless.
> 

So in your actual case, `e` is a signal (not a port) and its type is `unsigned(N-1 downto 0)` where `N` is some generic? Are there any generate blocks between the signal and the `(e'range => a)` expression?

Anselmo95, 2024-03-14:
In my actual case `e` is a signal but `f` is an input port, while `N` is just a constant (it's value is 5 in my case).
The other differnce is that `a` is an input port and there is some logic behind `f` and `d` but I did't put it in my example not to bring too many signlas in the example.
There is a single `for generate` block in the same file but I can't find a direct link with any of the signlas in my example, there is some inderect link though. `b` uses some signals which are registerd version of the signals used in the generate block.

I'll keep tring to reproduce the issue now that I know the line but I'm not sure i'll manage to do it soon

nickg, 2024-03-18:
I pushed an issue867 branch which has extra debug prints. Could you try again with that?

Anselmo95, 2024-03-18:
```
      |
 1526 |                             else e and (e'range => a);
      |                                                    ^^^^^^^
E'RANGE
prefix_reg kind = 4
```
I think this is the extra debug prints you added, let mre know if I missed something and sorry did't find the time to look for a reproducer yet.

Anselmo95, 2024-03-19:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pkg is
    function dummy_function return natural;
end package pkg;

package body pkg is
    function dummy_function return natural is
    begin
        return 4;
    end function;
end package body pkg;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.pkg.all;

entity ent is
constant CONST : natural  := dummy_function;
end entity ent;

architecture arch of ent is
    signal s0 : std_logic;
    signal s1 : unsigned(CONST downto 0);
    signal s2 : unsigned(CONST downto 0);
begin
    s2 <= s1 and (s1'range => s0);
end architecture arch;
```

I finally have a reproducer and it seams to me that the problme is caused by the `s1'range`, where `CONST` is declare and the fact that it is initialized by a function. If I either move the constant declaration inside the architecture or initialize the constant with a number the elaboration is fine.

nickg, 2024-03-20:
Thanks. Can you try again with the latest master branch?

Anselmo95, 2024-03-20:
Thanks, the problem is fixed ! I still have another poblem during elaboration where nvc tells me that for package `WORK.A_PKG` it cannot find a package inside another library,  `B_LIB.B_PKG`, and I am sure I have analized the latter beacuse I can see a file with the same name inside the folder `b_lib` ... Anyways I'll open another issue later on after some experiments. Thanks again !

nickg, 2024-03-21:
OK, thanks. I'll close this one now.

closed 2024-03-21 by nickg