nickg/nvc #868: VHPI: Iterating declarations inside `for generate` no longer returns any

marlonjames, 2024-03-19:
In `v1.11.3`, iterating for `vhpiDecls`, `vhpiSigDecls`, or `vhpiConstDecls` from a reference handle of kind `vhpiForGenerateK` would give an iterator handle that then gives declaration handles.

On master, these iterators no longer return any handles to declarations when iterating with `vhpi_scan`.

I don't have a minimal example, but the VHDL used by the cocotb regression test suite is here:

https://github.com/cocotb/cocotb/blob/785e0896930039023db3c2e2be47bed5375326c5/tests/designs/array_module/array_module.vhd#L149-L156

If you want, I can try to put together a test for NVC.

nickg, 2024-03-20:
I think this was broken by 7dd41079090dcf. There's still a few failures if I run the cocotb testsuite with the current master but I'm using #857 to track that. Once that's fixed I'll add cocotb's full testsuite to the CI to avoid regressions in the future (at the moment it only runs the examples).

I also implemented `vhpiParamDecl` for `vhpi_handle()` and `vhpiGenerateIndexP` for `vhpi_get` but cocotb doesn't seem to be using those at the moment.

closed 2024-03-20 by nickg

marlonjames, 2024-03-20:
Thanks for the fix!
cocotb runs its regression suite on the master branch of several open source simulators weekly: https://github.com/cocotb/cocotb/actions/workflows/experimental.yml

commit 93e95cf0 (2024-03-20): VHPI discovery in for-generate broken since 7dd41. Fixes #868