nickg/nvc #870: Relax hidden declaration to warning

Anselmo95, 2024-03-26:
Hello, 

While analysing the code below NVC errors out because the generic is hidden by the variable inside the process. While I understand that this is not a good practice would it be possible to have an option, maybe `--relaxed` or another one, to reduce this error to just a warning ?

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
generic (
    SEED1 : natural
);
port (
    clk : in std_logic
);
end entity ent;

architecture arch of ent is
begin
    process(clk)
        variable seed1 : natural := SEED1;
    begin
    end process;
end architecture arch;
```

closed 2024-03-26 by nickg

Anselmo95, 2024-03-26:
Thanks a lot for the change !

nickg, 2024-03-26:
This error comes from the rules in LRM section 12.3 "Visibility": an object declaration hides all homographs from the _start_ of its declaration but is not itself visible until the _end_ of the declaration. So it's illegal to reference an object with a given name inside a declaration with the same name.

Anyway I can skip this check with `--relaxed`, which actually just restores the behaviour before #733.

commit d1273a9b (2024-03-26): Suppress error for hidden declarations with --relaxed. Fixes #870