nickg/nvc #873: Assign same length vector with different offsets

Anselmo95, 2024-03-27:
Hello,
The code below gives me this error during simulation:
```
** Fatal: 0ms+0: index 3 outside of INTEGER range 13 downto 10
    > ent.vhd:22
    |
 22 |         mask := (data'length-1 downto 0 => en);
    |                  ^^^^^^^^^^^^^^^^^^^^^^
   Function APPLY_ENABLE [STD_LOGIC_VECTOR, STD_LOGIC return STD_LOGIC_VECTOR] at ent.vhd:18
   Process :ent:_p0 at ent.vhd:27
```
As a note the code runs fine if I modify the declaration of mask into `variable mask : std_logic_vector(data'length-1 downto 0);` or if I change the assignment to match the indexes `mask := (data'range => en);`. 
I don't know if it is important, just thought to mention this

Command:
`nvc --std=08 --work=work -L./ -a ent.vhd --relaxed && nvc --std=08 --work
=work -L./ -e ent && nvc --std=08 --work=work -L./ -r ent --stop-time=5ns`

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
generic (
    DWIDTH : natural := 4;
    OFFSET : natural := 10
);
port (
    enable : in std_logic;
    in_data : in std_logic_vector(DWIDTH+OFFSET-1 downto OFFSET);
    out_data : out std_logic_vector(DWIDTH+OFFSET-1 downto OFFSET)
);
end entity ent;

architecture arch of ent is
    function apply_enable(data : std_logic_vector; en : std_logic) return std_logic_vector is
        variable mask : std_logic_vector(data'range);
        variable data_masked : std_logic_vector(data'range);
    begin
        mask := (data'length-1 downto 0 => en);
        data_masked := data and mask;
        return data_masked;
    end function;
begin
    out_data <= apply_enable(in_data, enable);
end architecture arch;```

nickg, 2024-03-27:
Does this work?

```vhdl
mask := std_logic_vector'(data'length-1 downto 0 => en);
```

(Not able to test it at the moment.)

Anselmo95, 2024-03-27:
Yes it works! Both in the reproducer and in the original code I first had this issue

nickg, 2024-03-29:
Should be fixed in the latest master branch.

closed 2024-03-29 by nickg

Anselmo95, 2024-03-29:
As always thanks a lot for the fix !
But i gave it a try and this commit may have introduced some regression because I now have this trace during elaboration:
```** Fatal: invalid type kind T_ARRAY in range_of
[0x498c24] ../src/diag.c:1014 diag_femit
[0x498c24] ./../src/diag.c:1039 diag_emit
[0x54375d] ./../src/util.c:585 fatal_trace
[0x4df564] ./../src/common.c:964 range_of.part.8
[0x4d6853] ./../src/common.c:940 lower_array_const_size.lto_priv.498
[0x42cda9] ./../src/lower.c:3305 lower_const_array_aggregate.constprop.43
[0x42cf94] ./../src/lower.c:3331 lower_const_array_aggregate.constprop.43
[0x4d2e2d] ../src/lower.c:3753 lower_array_aggregate
[0x4d2e2d] ./../src/lower.c:4224 lower_aggregate.lto_priv.508
[0x4d395f] ./../src/lower.c:5120 lower_expr.lto_priv.481
[0x4b7bf3] ./../src/lower.c:12516 lower_rvalue
[0x4d80b0] ./../src/lower.c:1127 lower_subprogram_arg.lto_priv.786
[0x4d0675] ./../src/lower.c:2460 lower_fcall.lto_priv.503
[0x4d38ef] ./../src/lower.c:5110 lower_expr.lto_priv.481
[0x4b7bf3] ./../src/lower.c:12516 lower_rvalue
[0x4c8cf7] ./../src/lower.c:6264 lower_sequence.lto_priv.782
[0x4c855d] ./../src/lower.c:6283 lower_sequence.lto_priv.782
[0x4c855d] ./../src/lower.c:6283 lower_sequence.lto_priv.782
[0x4c855d] ./../src/lower.c:6283 lower_sequence.lto_priv.782
[0x4b9e4d] ./../src/lower.c:11241 lower_process
[0x4f311f] ./../src/elab.c:1962 elab_stmts
[0x4f795a] ./../src/elab.c:1589 elab_architecture
[0x4f40c4] ./../src/elab.c:1656 elab_stmts
[0x4f795a] ./../src/elab.c:1589 elab_architecture
[0x4f40c4] ./../src/elab.c:1656 elab_stmts
[0x4f795a] ./../src/elab.c:1589 elab_architecture
[0x4f40c4] ./../src/elab.c:1656 elab_stmts
[0x4f795a] ./../src/elab.c:1589 elab_architecture
[0x4f40c4] ./../src/elab.c:1656 elab_stmts
[0x4f490c] ./../src/elab.c:2095 elab_stmts
[0x4f490c] ./../src/elab.c:2095 elab_stmts
[0x4f795a] ./../src/elab.c:1589 elab_architecture
[0x4f40c4] ./../src/elab.c:1656 elab_stmts
[0x4f795a] ./../src/elab.c:1589 elab_architecture
[0x4f8246] ./../src/elab.c:2176 elab_top_level.isra.12
[0x4f8639] ./../src/elab.c:2238 elab
[0x54d7ab] ./../src/nvc.c:464 elaborate
[0x54f21e] ./../src/nvc.c:1922 process_command
[0x4120a9] ./../src/nvc.c:2060 main
```

I'll probably debug and try to find a reproducer/more information sometime next week and then open a new issue when I have something. Let me know if you prefer to keep using this issue instead.