nickg/nvc #879: [Bug] `fatal: load indirect with non-pointer argument` when

pidgeon777, 2024-04-18:
@nickg this is an issue that I have found while working on a project, which I haven't been able to solve.

A `fatal: load indirect with non-pointer argument` error is produced when trying to elaborate an entity.

# Steps to reproduce

1. Create the following source files:

- `TestBusSlv.vhd`

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.TestIfPkg.all;
use work.TestPkg.all;

entity TestBusSlv is
  port (
    EriCfgRegxMO  : out tERI_CFG_REG_ARRAY;
    TaxCfgRegxMO  : out tTAX_CFG_REG_ARRAY;
    TestCfgRegxMO : out tTEST_CFG_REG_ARRAY
  );
end TestBusSlv;

architecture behavioral of TestBusSlv is

  signal TaxCfgRegxM  : tTAX_CFG_REG_ARRAY;
  signal TestCfgRegxM : tTEST_CFG_REG_ARRAY;
  signal EriCfgRegxM  : tERI_CFG_REG_ARRAY;

begin

  EriCfgRegxMO  <= EriCfgRegxM;
  TaxCfgRegxMO  <= TaxCfgRegxM;
  TestCfgRegxMO <= TestCfgRegxM;

end behavioral;

```

- `TestCompLib.vhd`

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.TestIfPkg.all;
use work.TestPkg.all;

package TestCompLib is

  component TestBusSlv is
    port (
      EriCfgRegxMO  : out tERI_CFG_REG_ARRAY;
      TaxCfgRegxMO  : out tTAX_CFG_REG_ARRAY;
      TestCfgRegxMO : out tTEST_CFG_REG_ARRAY
    );
  end component;

end TestCompLib;

```

- `TestIfPkg.vhd`

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

library work;

package TestIfPkg is

  constant cVOD1_NUM : natural := 4;
  constant cVOD2_NUM : natural := 8;
  constant cVOD3_NUM : natural := 4;

end TestIfPkg;
```

- `TestModule.vhd`

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.TestIfPkg.all;
use work.TestPkg.all;
use work.TestCompLib.all;

entity TestModule is
end TestModule;

architecture behavioral of TestModule is

  signal EriCfgRegxM  : tERI_CFG_REG_ARRAY;
  signal TaxCfgRegxM  : tTAX_CFG_REG_ARRAY;
  signal TestCfgRegxM : tTEST_CFG_REG_ARRAY;

begin

  BusSlaveComp : TestBusSlv
    port map (
      EriCfgRegxMO  => EriCfgRegxM,
      TaxCfgRegxMO  => TaxCfgRegxM,
      TestCfgRegxMO => TestCfgRegxM
    );

end behavioral;

```

- `TestPkg.vhd`

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

library work;
use work.TestIfPkg.all;

package TestPkg is

  constant cTAX_NUMSIG_GROUPS : natural := 5;
  constant cTES_NUMSIG_GROUPS : natural := 5;
  constant cERI_NUMSIG_GROUPS : natural := 3;

  ----------------------------------------------------------------------------
  -- Section 1
  ----------------------------------------------------------------------------
  constant cTAX_GRPSEL_WIDTH : natural := natural(ceil(log2(real(cTAX_NUMSIG_GROUPS))));

  type tTAX_CFG_REG is record
    PgmSelxM : std_logic_vector(cTAX_GRPSEL_WIDTH-1 downto 0);
    SolSelxM : std_logic;
  end record tTAX_CFG_REG;

  constant cTAX_CFG_REG_DEFAULT : tTAX_CFG_REG :=
  (
    PgmSelxM => (others => '0'),
    SolSelxM => '0'
  );

  type tTAX_CFG_REG_ARRAY is array (0 to cVOD2_NUM-1) of tTAX_CFG_REG;

  ----------------------------------------------------------------------------
  -- Section 2
  ----------------------------------------------------------------------------
  constant cTEST_GRPSEL_WIDTH : natural := natural(ceil(log2(real(cTES_NUMSIG_GROUPS))));

  type tTEST_CFG_REG is record
    PgmSelxM : std_logic_vector(cTEST_GRPSEL_WIDTH-1 downto 0);
    SolSelxM : std_logic;
  end record tTEST_CFG_REG;

  constant cTEST_CFG_REG_DEFAULT : tTEST_CFG_REG :=
  (
    PgmSelxM => (others => '0'),
    SolSelxM => '0'
  );

  type tTEST_CFG_REG_ARRAY is array (0 to cVOD1_NUM-1) of tTEST_CFG_REG;

  ----------------------------------------------------------------------------
  -- Section 3
  ----------------------------------------------------------------------------
  constant cERI_GRPSEL_WIDTH : natural := natural(ceil(log2(real(cERI_NUMSIG_GROUPS))));

  type tERI_CFG_REG is record
    PgmSelxM  : std_logic_vector(cERI_GRPSEL_WIDTH-1 downto 0);
    SolSelxM  : std_logic;
    AnsGdexM : std_logic;
  end record tERI_CFG_REG;

  constant cERI_CFG_REG_DEFAULT : tERI_CFG_REG :=
  (
    PgmSelxM  => (others => '0'),
    SolSelxM  => '0',
    AnsGdexM => '0'
  );

  type tERI_CFG_REG_ARRAY is array (0 to cVOD3_NUM-1) of tERI_CFG_REG;

end package TestPkg;

```

2. Analyze the design:

```batch
C:\NVC\GCC\mingw64\bin\nvc.exe -M 64m --message=compact --std=1993 --work=work:nvc_lib -a TestIfPkg.vhd TestPkg.vhd TestBusSlv.vhd TestCompLib.vhd TestModule.vhd
```

3. Elaborate the design:

```batch
C:\NVC\GCC\mingw64\bin\nvc.exe -M 64m --message=compact --std=1993 --work=work:nvc_lib -e --verbose TestModule
```

As a result, the following error is produced:

```log
Name       WORK.TESTMODULE.BUSSLAVECOMP
Kind       instance
Context    WORK.TESTMODULE
Blocks     1
Registers  16
Types      17
  WORK.TESTPKG.TERI_CFG_REG$            {[*] : $<0..8>, $<0..8>, $<0..8>}
Variables  1
  ERICFGREGXMO                          // [*] : WORK.TESTPKG.TERI_CFG_REG${}, signal
Begin
   0: r0 := package init WORK.TESTCOMPLIB // P<WORK.TESTCOMPLIB>
      r1 := package init STD.STANDARD   // P<STD.STANDARD>
      r2 := package init WORK.TESTPKG   // P<WORK.TESTPKG>
      r3 := package init IEEE.STD_LOGIC_1164 // P<IEEE.STD_LOGIC_1164>
      r4 := package init IEEE.NUMERIC_STD // P<IEEE.NUMERIC_STD>
      r5 := package init WORK.TESTIFPKG // P<WORK.TESTIFPKG>
      r6 := var upref 1, ERICFGREGXM    // @<[*] : WORK.TESTPKG.TERI_CFG_REG${}> => WORK.TESTPKG.TERI_CFG_REG{}       
      r7 := load indirect r6            // [*] : WORK.TESTPKG.TERI_CFG_REG${} => WORK.TESTPKG.TERI_CFG_REG{}
      r8 := var upref 1, TAXCFGREGXM    // @<[*] : WORK.TESTPKG.TTAX_CFG_REG${}> => WORK.TESTPKG.TTAX_CFG_REG{}       
      r9 := load indirect r8            // [*] : WORK.TESTPKG.TTAX_CFG_REG${} => WORK.TESTPKG.TTAX_CFG_REG{}
      r10 := var upref 1, TESTCFGREGXM  // @<[*] : WORK.TESTPKG.TTEST_CFG_REG${}> => WORK.TESTPKG.TTEST_CFG_REG{}
      r11 := load indirect r10          // [*] : WORK.TESTPKG.TTEST_CFG_REG${} => WORK.TESTPKG.TTEST_CFG_REG{}        
      r12 := debug locus WORK.TESTMODULE.elab-29 // D<>
      r13 := const 4                    // # => 4
      r14 := uarray len r7 dim 0        // #
      length check left r13 == right r14 locus r12
      r15 := index ERICFGREGXMO         // @<[*] : WORK.TESTPKG.TERI_CFG_REG${}> => [*] : WORK.TESTPKG.TERI_CFG_REG${}
      invalid := load indirect r7        <----

note: initialising [0ms +11492kB]
note: loading top-level unit [2ms +456kB]
fatal: load indirect with non-pointer argument
[00007ff6c179a0bf]
[00007ff6c1854ed5]
[00007ff6c1799af1]
[00007ff6c18389c5]
[00007ff6c17eba08]
[00007ff6c17f2c9c]
[00007ff6c17f3a75]
[00007ff6c17f63ed]
[00007ff6c17987a9]
[00007ff6c1795cb4]
[00007ff6c18c994d] std_env_get_assert_format+0x282bd
[00007ff6c17912ee]
[00007ff6c1791406]
[00007ffca2c3257d] BaseThreadInitThunk+0x1d
[00007ffca40aaa48] RtlUserThreadStart+0x28
```

# Expected behavior

The design should elaborate without errors.

# Additional information

Other compilers and simulators like **GHDL** and the one from **Vivado** are able to elaborate the design without issues.

nickg, 2024-04-18:
Which version are you using? Can you check this still fails with either the latest master branch or 1.12 release?

```
$ nvc -M 64m --message=compact --std=1993 --work=work:nvc_lib -e --verbose TestModule
note: initialising [7ms +54332kB]
note: loading top-level unit [1ms +688kB]
note: elaborating design [3ms +5640kB]
note: saving library [0ms +344kB]
note: code generation for 15 units [73ms +13916kB]
note: linking shared library [1ms +0kB]
$ nvc --version
nvc 1.13-devel (1.12.0.r12.g47bc8413186d) (Using LLVM 15.0.6) [debug]
Copyright (C) 2011-2024  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
``

nickg, 2024-05-04:
I think this was fixed by afca01c6ef18d

closed 2024-05-04 by nickg