nickg/nvc #880: Coverage refactoring - Tracking Issue

Blebowski, 2024-04-20:
Convert all Cover Items to one Bin per item, make each item counter:
1. [ ] Branch coverage - Done only half-way, coverage reporting still uses to mask and flag.
2. [ ] FSM coverage - Each bin single item, but mask is used
3. [ ] Toggle coverage - Not done
4. [ ] Expression coverage - Not done

Fully encode bin name into the cover item hierarchy:
1. [x] FSM coverage
2. [ ] Toggle coverage
3. [ ] Branch coverage
4. [ ] Expression coverage

Other internal refactoring / improvements:
1. [ ] Make run-time counters 64 bit (32 bit toggle can overflow in realistically long simulation)
2. [ ] Make index of cover_item (the tag) 64 bit. Large designs can have 2 ^ 32 tags.
3. [ ] Convert unreachability mask to a single bit flag. Since single cover item is single bin, no "unreachability mask" is needed.
4. [ ] Convert exclude mask to single bit flag. The same as above

Changes related to the "user":
1. [ ] Change specification of exclude files not to require bin name separately -> Since bin name will be part of hierarchy, this will simplify exclude file, its loading, checking and handling.

nickg, 2024-04-20:
> Make run-time counters 64 bit (32 bit toggle can overflow in realistically long simulation)

This will also double the memory usage of the counters and isn't needed in 99% of cases. It would be better to just make the counter saturate at 2^32 - 1 instead.

Blebowski, 2024-04-20:
> This will also double the memory usage of the counters and isn't needed in 99% of cases. It would be better to just make the counter saturate at 2^32 - 1 instead.

Agreed. Any solution that does not overflow and will cause covered item to be reported as uncovered in works for me.

nickg, 2024-04-21:
> It would be better to just make the counter saturate at 2^32 - 1 instead.

I did this in 1745c08.

Blebowski, 2024-04-21:
Thanks, I have rebased my new branch where I am trying to the the coverage refactoring more gradually. It is still WIP, but I am on a much better track now :) Time wise, I am still busy, but I will try to push it a little.

closed 2024-07-18 by Blebowski