nickg/nvc #886: False error for generic initialised with non globally static expression

stieeev, 2024-05-07:
[hdl.zip](https://github.com/nickg/nvc/files/15228964/hdl.zip)

The attached example file top.vhd instantiates work.ram with a generic set to the value INIT_LINE("default.txt"), which is not a globally static expression. nvc 1.12.0 fails this with an error:

```
** Error: actual associated with generic DFLT_NUM must be a globally static expression
    > C:\awrk\hdl\top.vhd:23
    |
 23 |     DFLT_NUM => INIT_LINE("default.txt")
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^

```
Questasim 2022.3 doesn't bat an eyelid with this, not even a warning.

I'm not expert in the VHDL standards, but I guess the requirement for globally static expressions applies to ports but not to generics ?

stieeev, 2024-05-07:
Saw a Stack Overflow post that seems relevant:

https://stackoverflow.com/questions/57387726/what-is-a-globallylocally-static-primaryexpressionrangesubexpression-and

> One interesting usage is - considering an innocent looking footnote in Chapter 9.4 of the standard:
> 
> The rules for globally static expressions imply that a declared constant or a generic may be initialized with an expression that is not globally static, for example, with a call to an impure function. The resulting constant value may be globally static, even though its initial value expression is not.

nickg, 2024-05-08:
Yes you're correct, this check is overly strict.

commit 6f319eac (2024-05-08): Generic actuals do not need to be globally static. Fixes #886
closed 2024-05-08 by nickg