nickg/nvc #887: JIT throwing illegal instructions

albydnc, 2024-05-08:
I am using a docker image in CI to execute my simulations.
When I run the image locally, all tests are successful, however when I run in CI or on a different machine, I get `SIGILL`.
```
*** Caught signal 4 (SIGILL) [address=0x7f22e6631007, ip=0x7f22e6631007] ***

[0x42586d] ../src/util.c:870 signal_handler.lto_priv.0
[0x7f22f7c4970f] (/usr/lib64/libc.so.6) 
[0x7f22e6631007] OSVVM.COVERAGEPKG.CALCPERCENTCOV(II)R+0x7 [VHDL]
```
```
*** Caught signal 4 (SIGILL) [address=0x7fcaae020a06, ip=0x7fcaae020a06] ***

[0x42586d] ../src/util.c:870 signal_handler.lto_priv.0
[0x7fcabf73e70f] (/usr/lib64/libc.so.6) 
[0x7fcaae020a06] OSVVM.RANDOMPKG.RANDOMPTYPE.RANDUNSIGNED(N)25IEEE.NUMERIC_STD.UNSIGNED+0x6 [VHDL]
```
The errors are gone when I remove the `--jit` flag

nickg, 2024-05-08:
Can you try installing [capstone](https://www.capstone-engine.org/) (e.g. `libcapstone-dev` in Ubuntu/Debian) and then re-run `configure`, `make clean`, `make`, `make install`? That should print a disassembly dump of the instruction that causes the problem.

Are there any differences between the machine where it works and the one which doesn't? E.g. CPU type, model, etc.

nickg, 2024-05-13:
I also improved the crash reporting a bit so it prints a hexdump if capstone is not installed.

albydnc, 2024-05-14:
hello @nickg, here is the dump:
```
------------------------------------- OSVVM.COVERAGEPKG.CALCPERCENTCOV(II)R ----
7fcf0fe10600 0000             add byte ptr [rax], al
7fcf0fe10602 0000             add byte ptr [rax], al
7fcf0fe10604 0000             add byte ptr [rax], al
7fcf0fe10606 59               pop rcx
7fcf0fe10607 40               .byte 0x40
7fcf0fe10608 ff               .byte 0xff                          <=============
	RAX	7fcf0fe10600
	RCX	507ae0
	RDX	7ffc118d8a90
	RBX	2c54d58
	RSP	7ffc118d6760
	RBP	7ffc118d72b0
	RSI	7ffc118d7210
	RDI	2a7bce0
	R8	7ffc118d8aa0
	R9	7fcf1156fa98
	R10	2
	R11	1
	R12	c8
	R13	7ffc118d6770
	R14	0
	R15	2b2a2e0
7fcf0fe10609 ff               .byte 0xff
7fcf0fe1060a ff               .byte 0xff
7fcf0fe1060b ff               .byte 0xff
7fcf0fe1060c ff               .byte 0xff
7fcf0fe1060d ff               .byte 0xff
7fcf0fe1060e ef               out dx, eax
7fcf0fe1060f ff               .byte 0xff
7fcf0fe10610 ff               .byte 0xff
7fcf0fe10611 ff               .byte 0xff
7fcf0fe10612 ff               .byte 0xff
7fcf0fe10613 ff               .byte 0xff
7fcf0fe10614 ff               .byte 0xff
7fcf0fe10615 ff               .byte 0xff
7fcf0fe10616 ef               out dx, eax
7fcf0fe10617 7f90             jg 0x7fcf0fe105a9
7fcf0fe10619 90               nop 
7fcf0fe1061a 90               nop 
7fcf0fe1061b 90               nop 
7fcf0fe1061c 90               nop 
7fcf0fe1061d 90               nop 
7fcf0fe1061e 90               nop 
7fcf0fe1061f 90               nop 
7fcf0fe10620 f30f1efa         endbr64 
7fcf0fe10624 4883ec18         sub rsp, 0x18
7fcf0fe10628 48893424         mov qword ptr [rsp], rsi
7fcf0fe1062c 48897c2408       mov qword ptr [rsp + 8], rdi
7fcf0fe10631 c744241000000000 mov dword ptr [rsp + 0x10], 0
7fcf0fe10639 8b4110           mov eax, dword ptr [rcx + 0x10]
7fcf0fe1063c 89442414         mov dword ptr [rsp + 0x14], eax
7fcf0fe10640 488b4210         mov rax, qword ptr [rdx + 0x10]
7fcf0fe10644 4885c0           test rax, rax
7fcf0fe10647 0f8e9b000000     jle 0x7fcf0fe106e8
7fcf0fe1064d f2480f2a4208     cvtsi2sd xmm0, qword ptr [rdx + 8]
7fcf0fe10653 48be0006e10fcf7f movabs rsi, 0x7fcf0fe10600
                -0000            
7fcf0fe1065d f20f5906         mulsd xmm0, qword ptr [rsi]
7fcf0fe10661 f2480f2ac8       cvtsi2sd xmm1, rax
7fcf0fe10666 f20f5ec1         divsd xmm0, xmm1
7fcf0fe1066a 66480f7ec0       movq rax, xmm0
7fcf0fe1066f 48be0806e10fcf7f movabs rsi, 0x7fcf0fe10608
                -0000            
7fcf0fe10679 f20f100e         movsd xmm1, qword ptr [rsi]
7fcf0fe1067d 660f2ec8         ucomisd xmm1, xmm0
7fcf0fe10681 7710             ja 0x7fcf0fe10693
7fcf0fe10683 48be1006e10fcf7f movabs rsi, 0x7fcf0fe10610
[... 768 lines cut ...]
7f77c9e01ed9 31d2             xor edx, edx
7f77c9e01edb 31c9             xor ecx, ecx
7f77c9e01edd eb09             jmp 0x7f77c9e01ee8
7f77c9e01edf 90               nop 
7f77c9e01ee0 d1ee             shr esi, 1
7f77c9e01ee2 0fafff           imul edi, edi
7f77c9e01ee5 0f90c1           seto cl
7f77c9e01ee8 85f6             test esi, esi
7f77c9e01eea 7414             je 0x7f77c9e01f00
7f77c9e01eec 40f6c601         test sil, 1
7f77c9e01ef0 74ee             je 0x7f77c9e01ee0
7f77c9e01ef2 0fafc7           imul eax, edi
7f77c9e01ef5 410f90c0         seto r8b
7f77c9e01ef9 08ca             or dl, cl
7f77c9e01efb 4408c2           or dl, r8b
7f77c9e01efe ebe0             jmp 0x7f77c9e01ee0
7f77c9e01f00 c3               ret 
--------------------------------------------------------------------------------

*** Caught signal 4 (SIGILL) [address=0x7f77c9e01306, ip=0x7f77c9e01306] ***

[0x42596d] ../src/util.c:870 signal_handler.lto_priv.0
[0x7f77cf3b270f] (/usr/lib64/libc.so.6) 
[0x7f77c9e01306] OSVVM.RANDOMPKG.RANDOMPTYPE.RANDUNSIGNED(N)25IEEE.NUMERIC_STD.UNSIGNED+0x6 [VHDL]
```

nickg, 2024-05-14:
Thanks. I think I might have fixed it - could you test again with the latest master branch?

I'm a bit confused why I can't reproduce this though. Can you find out what LLVM version is inside the container (`nvc --version` will print it)?

albydnc, 2024-05-15:
nvc version output:
```
nvc 1.13-devel (29905fb) (Using LLVM 18.1.1)
```
Thanks! now the issue is fixed! :+1:

closed 2024-05-15 by albydnc

nickg, 2024-05-15:
Yes the problem is with LLVM 18. I managed to reproduce it myself now.