nickg/nvc #890: "Empty basic block" crash while compiling

nathanaelhuffman, 2024-05-16:
I've done no debugging on my end, this is fresh code and I'm sure I've got an issue with my code and will go chase that down but reporting the crash as the tool suggested:

This is using a `case?` if that helps, code is something like this in a package:
```
constant OPCODE_GET_STATUS : std_logic_vector(7 downto 0) := "00100101";
constant OPCODE_SET_CONFIGURATION : std_logic_vector(7 downto 0) := "00100010";
constant OPCODE_GET_CONFIGURATION : std_logic_vector(7 downto 0) := "00100001";
constant OPCODE_PUT_IORD_SHORT_MASK : std_logic_vector(7 downto 0) := "010000--";
constant OPCODE_PUT_IOWR_SHORT_MASK : std_logic_vector(7 downto 0) := "010000--";
constant OPCODE_PUT_MEMRD32_SHORT_MASK : std_logic_vector(7 downto 0) := "010010--";
constant OPCODE_PUT_MEMWR32_SHORT_MASK  : std_logic_vector(7 downto 0) := "010011--";

 function bytes_until_turn(opcode: std_logic_vector; cmd_length: std_logic_vector) return natural is
            variable bytes : natural := 0;
        begin
            case? opcode is
                -- opcode + crc
                when OPCODE_GET_STATUS => bytes := 1 + 1; 
                -- opcode + 16bit addr +crc
                when OPCODE_GET_CONFIGURATION => bytes := 1 + 2 + 1; 
                -- opcode + 16bit addr + 32bit data + crc
                when OPCODE_SET_CONFIGURATION => bytes := 1 + 2 + 4 + 1;
                -- opcode + 16bit addr + crc
                when OPCODE_PUT_IORD_SHORT_MASK => bytes := 1 + 2 + 1;
                -- opcode + 16bit addr + 1/2/4 data + crc
                when OPCODE_PUT_IOWR_SHORT_MASK | OPCODE_PUT_MEMRD32_SHORT_MASK => 
                    bytes := 1 + 2 + req_length_by_opcode(opcode) + 1;
                when others =>
                    assert false report "Not yet implemented";

            end case?;
            return bytes;
```

Crash report as follows:
```
Name       LIB.ESPI_PROTOCOL_PKG.BYTES_UNTIL_TURN(VV)N
Kind       function
Context    LIB.ESPI_PROTOCOL_PKG
Blocks     12
Registers  59
Types      17
Variables  1
  BYTES                                 // -2^31..2^31-1 => 0..2^31-1
Result     -2^31..2^31-1
Parameters 3
  r0    context                         // P<LIB.ESPI_PROTOCOL_PKG>
  r1    OPCODE                          // [*] : 0..8 => 0..8
  r2    CMD_LENGTH                      // [*] : 0..8 => 0..8
Begin
   0: r3 := const 0                     // -2^31..2^31-1 => 0
      r4 := const 2147483647            // -2^31..2^31-1 => 2^31-1
      r5 := const 0                     // 0..1 => 0
      BYTES := store r3
      r6 := link package NVC.IEEE_SUPPORT // P<NVC.IEEE_SUPPORT>
      fcall NVC.IEEE_SUPPORT.CHECK_MATCH_EXPRESSION(Y) r6, r1
      r7 := link package IEEE.STD_LOGIC_1164 // P<IEEE.STD_LOGIC_1164>
      r8 := const 3                     // 0..8 => 3
      r9 := var upref 1, OPCODE_GET_STATUS // @<0..8> => 0..8
      r10 := const 7                    // -2^31..2^31-1 => 7
      r11 := const 1                    // 0..1 => 1
      r12 := wrap r9 [r10 r3 r11]       // [*] : 0..8
      r13 := fcall IEEE.STD_LOGIC_1164."?="(YY)U$predef r7, r1, r12 // 0..8
      r14 := cmp r13 == r8
      cond r14 then 1 else 2
   1: r15 := const 2                    // -2^31..2^31-1 => 2
      r16 := const 0                    // -2^31..2^31-1 => 0
      r17 := const 2147483647           // -2^31..2^31-1 => 2^31-1
      r18 := const 0                    // 0..1 => 0
      BYTES := store r15
      jump 3
   2: r19 := var upref 1, OPCODE_GET_CONFIGURATION // @<0..8> => 0..8
      r20 := const 7                    // -2^31..2^31-1 => 7
      r21 := const 0                    // -2^31..2^31-1 => 0
      r22 := const 1                    // 0..1 => 1
      r23 := wrap r19 [r20 r21 r22]     // [*] : 0..8
      r24 := fcall IEEE.STD_LOGIC_1164."?="(YY)U$predef r7, r1, r23 // 0..8
      r25 := cmp r24 == r8
      cond r25 then 4 else 5
   3: Empty basic block
   4: r26 := const 4                    // -2^31..2^31-1 => 4
      r27 := const 0                    // -2^31..2^31-1 => 0
      r28 := const 2147483647           // -2^31..2^31-1 => 2^31-1
      r29 := const 0                    // 0..1 => 0
      BYTES := store r26
      jump 3
   5: r30 := var upref 1, OPCODE_SET_CONFIGURATION // @<0..8> => 0..8
      r31 := const 7                    // -2^31..2^31-1 => 7
      r32 := const 0                    // -2^31..2^31-1 => 0
      r33 := const 1                    // 0..1 => 1
      r34 := wrap r30 [r31 r32 r33]     // [*] : 0..8
      r35 := fcall IEEE.STD_LOGIC_1164."?="(YY)U$predef r7, r1, r34 // 0..8
      r36 := cmp r35 == r8
      cond r36 then 6 else 7
   6: r37 := const 8                    // -2^31..2^31-1 => 8
      r38 := const 0                    // -2^31..2^31-1 => 0
      r39 := const 2147483647           // -2^31..2^31-1 => 2^31-1
      r40 := const 0                    // 0..1 => 0
      BYTES := store r37
      jump 3
   7: r41 := var upref 1, OPCODE_PUT_IORD_SHORT_MASK // @<0..8> => 0..8
      r42 := const 7                    // -2^31..2^31-1 => 7
      r43 := const 0                    // -2^31..2^31-1 => 0
      r44 := const 1                    // 0..1 => 1
      r45 := wrap r41 [r42 r43 r44]     // [*] : 0..8
      r46 := fcall IEEE.STD_LOGIC_1164."?="(YY)U$predef r7, r1, r45 // 0..8
      r47 := cmp r46 == r8
      cond r47 then 8 else 9
   8: r48 := const 4                    // -2^31..2^31-1 => 4
      r49 := const 0                    // -2^31..2^31-1 => 0
      r50 := const 2147483647           // -2^31..2^31-1 => 2^31-1
      r51 := const 0                    // 0..1 => 0
      BYTES := store r48
      jump 3
   9: r52 := var upref 1, OPCODE_PUT_IOWR_SHORT_MASK // @<0..8> => 0..8
      r53 := const 7                    // -2^31..2^31-1 => 7
      r54 := const 0                    // -2^31..2^31-1 => 0
      r55 := const 1                    // 0..1 => 1
      r56 := wrap r52 [r53 r54 r55]     // [*] : 0..8
      r57 := fcall IEEE.STD_LOGIC_1164."?="(YY)U$predef r7, r1, r56 // 0..8
      r58 := cmp r57 == r8
      cond r58 then 10 else 11   <----
  10: Empty basic block
  11: Empty basic block

** Fatal: attempt to add to already finished block 9
[0x62c0d50ae7cc] ../src/diag.c:1011 diag_femit
[0x62c0d50ae7cc] ../src/diag.c:1036 diag_emit
[0x62c0d4fe91a2] ../src/util.c:585 fatal_trace
[0x62c0d5090d95] ../src/vcode.c:266 vcode_add_op
[0x62c0d5090d95] ../src/vcode.c:261 vcode_add_op.lto_priv.0
[0x62c0d50983f9] ../src/vcode.c:4755 emit_var_upref
[0x62c0d50723a9] ../src/lower.c:2725 lower_var_ref.lto_priv.0
[0x62c0d507afe4] ../src/lower.c:2989 lower_ref
[0x62c0d507afe4] ../src/lower.c:5198 lower_expr.lto_priv.0
[0x62c0d509483d] ../src/lower.c:12589 lower_rvalue
[0x62c0d507e704] ../src/lower.c:7203 lower_sequence.lto_priv.0
[0x62c0d508b5d4] ../src/lower.c:11000 lower_func_body
[0x62c0d50905e5] ../src/lower.c:13067 unit_registry_get
[0x62c0d506f54c] ../src/lower.c:2390 lower_context_for_call.lto_priv.0
[0x62c0d506f856] ../src/lower.c:2457 lower_fcall.lto_priv.0
[0x62c0d5079a7f] ../src/lower.c:5192 lower_expr.lto_priv.0
[0x62c0d509483d] ../src/lower.c:12589 lower_rvalue
[0x62c0d507e2f4] ../src/lower.c:6249 lower_sequence.lto_priv.0
[0x62c0d507dab2] ../src/lower.c:6365 lower_sequence.lto_priv.0
[0x62c0d5082506] ../src/lower.c:6880 lower_case_scalar
[0x62c0d5082506] ../src/lower.c:7140 lower_case.lto_priv.0
[0x62c0d507d02c] ../src/lower.c:7158 lower_sequence.lto_priv.0
[0x62c0d507dab2] ../src/lower.c:6365 lower_sequence.lto_priv.0
[0x62c0d504c790] ../src/lower.c:11323 elab_stmts.lto_priv.0
[0x62c0d504f0f1] ../src/elab.c:1589 elab_architecture.lto_priv.0
[0x62c0d504c91b] ../src/elab.c:1656 elab_stmts.lto_priv.0
[0x62c0d504f0f1] ../src/elab.c:1589 elab_architecture.lto_priv.0
[0x62c0d504c91b] ../src/elab.c:1656 elab_stmts.lto_priv.0
[0x62c0d504f0f1] ../src/elab.c:1589 elab_architecture.lto_priv.0
[0x62c0d504c91b] ../src/elab.c:1656 elab_stmts.lto_priv.0
[0x62c0d504f0f1] ../src/elab.c:1589 elab_architecture.lto_priv.0
[0x62c0d510a069] ../src/elab.c:2176 elab_top_level.constprop.0
[0x62c0d4ff039a] ../src/elab.c:2242 elab
[0x62c0d4ff039a] ../src/elab.c:464 elaborate
[0x62c0d4fecccf] ../src/nvc.c:1921 process_command
[0x62c0d4fe6a25] ../src/nvc.c:2059 main
```

nickg, 2024-05-17:
The branch with multiple choices (`OPCODE_PUT_IOWR_SHORT_MASK | OPCODE_PUT_MEMRD32_SHORT_MASK`) causes it. Should be fixed now, could you test again with the latest master branch?

nathanaelhuffman, 2024-05-17:
yep master branch runs great, thanks!

closed 2024-05-17 by nickg