nickg/nvc #894: nvc error creating instance of generic package given a constant as parameter

tmeissner, 2024-05-23:
I have a strange error when creating an instance of a generic package. When I give a parameter an integer literal, it works. When I give the parameter an constant of the same integer, it creates an nvc error like this:

```shell
nvc --std=19 -L ./ -e --cover=statement,branch,expression,fsm-state,functional axiqueuedmat

Name       WORK.AXIQUEUEDMATP.ADDCMPJOB(32WORK.AXIQUEUEDMATP.T_QUEUE_RANGE34WORK.AXIQUEUEDMATP.T_JOB_SIZE_BYTE32WORK.AXIQUEUEDMATP.MEMORY.T_DICT35WORK.AXIQUEUEDMATP.CMPQUEUE.T_QUEUEsLs32WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2Ss28WORK.AXIQUEUEDMAP.T_AXIS_S2M)
Kind       procedure
Context    WORK.AXIQUEUEDMATP
Blocks     7
Registers  274
Types      108
  WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2S$     {$<0..8>, $<0..8>, $<0..8>, WORK.AXIQUEUEDMAP.T_AXIS_CMP_TUSER_M2S${}}
  WORK.AXIQUEUEDMAP.T_AXIS_CMP_TUSER_M2S$ {$<0..8>, $<0..8>, $<0..8>, $<0..8>}
  WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2S      {0..8, 0..8, [512] : 0..8, WORK.AXIQUEUEDMAP.T_AXIS_CMP_TUSER_M2S{}}
  WORK.AXIQUEUEDMAP.T_AXIS_CMP_TUSER_M2S {[11] : 0..8, [2] : 0..8, [2] : 0..8, [16] : 0..8}
  WORK.AXIQUEUEDMAP.T_AXIS_S2M$         {$<0..8>}
  WORK.AXIQUEUEDMAP.T_AXIS_S2M          {0..8}
Variables  11
  QUEUE                                 // -2^63..2^63-1 => 0..3
  LEN                                   // -2^63..2^63-1 => 0..65535
  MEMORY                                // @<P<WORK.AXIQUEUEDMATP.MEMORY.T_DICT>> => P<WORK.AXIQUEUEDMATP.MEMORY.T_DICT>
  CMPQUEUE                              // @<P<WORK.AXIQUEUEDMATP.CMPQUEUE.T_QUEUE>> => P<WORK.AXIQUEUEDMATP.CMPQUEUE.T_QUEUE>
  CLK                                   // $<0..8> => 0..8
  AXISM2S                               // @<WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2S${}> => WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2S{}
  AXISS2M                               // @<WORK.AXIQUEUEDMAP.T_AXIS_S2M${}> => WORK.AXIQUEUEDMAP.T_AXIS_S2M{}
  V_DATA                                // [512] : 0..8 => 0..8, heap
  V_DATA_CNT                            // -2^63..2^63-1 => 0..2^63-1
  V_RANDOM                              // P<OSVVM.RANDOMPKG.RANDOMPTYPE>
  tmp266                                // WORK.AXIQUEUEDMAP.T_AXIS_CMP_TUSER_M2S{}, temp
Parameters 8
  r0    context                         // P<WORK.AXIQUEUEDMATP>
  r1    QUEUE                           // -2^63..2^63-1 => 0..3
  r2    LEN                             // -2^63..2^63-1 => 0..65535
  r3    MEMORY                          // @<P<WORK.AXIQUEUEDMATP.MEMORY.T_DICT>> => P<WORK.AXIQUEUEDMATP.MEMORY.T_DICT>
  r4    CMPQUEUE                        // @<P<WORK.AXIQUEUEDMATP.CMPQUEUE.T_QUEUE>> => P<WORK.AXIQUEUEDMATP.CMPQUEUE.T_QUEUE>
  r5    CLK                             // $<0..8> => 0..8
  r6    AXISM2S                         // @<WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2S${}> => WORK.AXIQUEUEDMAP.T_AXIS_CMP_M2S{}
  r7    AXISS2M                         // @<WORK.AXIQUEUEDMAP.T_AXIS_S2M${}> => WORK.AXIQUEUEDMAP.T_AXIS_S2M{}
Begin
   0: QUEUE := store r1
      LEN := store r2
      MEMORY := store r3
      CMPQUEUE := store r4
      CLK := store r5
      AXISM2S := store r6
      AXISS2M := store r7
      r8 := const 512                   // # => 512
      r9 := index V_DATA                // @<0..8> => 0..8
      r10 := const 0                    // 0..8 => 0
      r9 := memset r10 length r8
      r11 := debug locus WORK.AXIQUEUEDMATP-body+12690 // D<>
      r12 := const 0                    // -2^63..2^63-1 => 0
      r13 := const 9223372036854775807  // -2^63..2^63-1 => 2^63-1
      r14 := const 0                    // 0..1 => 0
      r15 := const 511                  // -2^63..2^63-1 => 511
      r16 := const 1                    // 0..1 => 1
      r17 := debug locus WORK.AXIQUEUEDMATP-body+12682 // D<>
      V_DATA_CNT := store r12
      r18 := var upref 1, path_name     // @<[*] : 0..255> => 0..255
      r19 := load indirect r18          // [*] : 0..255 => 0..255
      r20 := const 58                   // 0..255 => 58
      r21 := const 97                   // 0..255 => 97
      r22 := const 100                  // 0..255 => 100
      r23 := const 99                   // 0..255 => 99
      r24 := const 109                  // 0..255 => 109
      r25 := const 112                  // 0..255 => 112
      r26 := const 106                  // 0..255 => 106
      r27 := const 111                  // 0..255 => 111
      r28 := const 98                   // 0..255 => 98
      r29 := const 91                   // 0..255 => 91
[... 343 lines cut ...]
           G_QUEUE_DEPTH : positive := 4096 / 8 + 5;
           type G_DATA_TYPE;
           function G_TO_STRING(data : G_DATA_TYPE) return string is to_string
  );

  subtype t_queue_count is natural range 0 to G_QUEUE_COUNT - 1;

  type t_queue is protected

    procedure init (data : in  G_DATA_TYPE; logging : in boolean;
                    parent_name : string := ""; level : LogType := INFO);
    procedure push (data : in  G_DATA_TYPE; queue : in t_queue_count := 0);
    procedure pop  (data : out G_DATA_TYPE; queue : in t_queue_count := 0);
    impure function pop  (queue : in t_queue_count := 0) return G_DATA_TYPE;
    impure function empty     (queue : in t_queue_count := 0) return boolean;
    impure function full      (queue : in t_queue_count := 0) return boolean;
    impure function fillstate (queue : in t_queue_count := 0) return natural;
    impure function freeslots (queue : in t_queue_count := 0) return natural;

  end protected t_queue;

end package QueueP;
```

I will post the code of an MWE if I'm successful to create one.

nickg, 2024-05-23:
Here's a minimal way to reproduce it:

```vhdl
package QueueP is

    generic (G_QUEUE_COUNT : positive := 1;
             G_QUEUE_DEPTH : positive := 4096 / 8 + 5;
             type G_DATA_TYPE;
             function G_TO_STRING(data : G_DATA_TYPE) return string is to_string);

    subtype t_queue_count is natural range 0 to G_QUEUE_COUNT - 1;

    procedure test (x : t_queue_count);
end package QueueP;

package body QueueP is

    procedure test (x : t_queue_count) is
    begin
    end procedure;

end package body;

-------------------------------------------------------------------------------

package pack is
    constant C_QUEUE_COUNT         : positive := 4;
    constant C_QUEUE_MAX_SIZE      : positive := 65535;

    package C2hQueue is new work.QueueP
        generic map (G_QUEUE_COUNT => C_QUEUE_COUNT,
                     G_QUEUE_DEPTH => C_QUEUE_MAX_SIZE,
                     G_DATA_TYPE   => bit_vector(7 downto 0),
                     G_TO_STRING   => to_string);
end package;

-------------------------------------------------------------------------------

entity issue894 is
end entity;

use work.pack.all;

architecture test of issue894 is
begin

    tb: process is
        variable v : natural;
    begin
        C2hQueue.test(v);
        wait;
    end process;

end architecture;
```

nickg, 2024-05-24:
@tmeissner please test again with the latest master branch.

tmeissner, 2024-05-24:
Thanks. The reported error is gone now. Now I get an error with my DUT design, I'll open a separate issue for that.

tmeissner, 2024-05-24:
The error I got now with another file, even before analyzing the package above:

```
nvc --std=19 --work=work -L ./ -a ../../src/QueueHandlingE.vhd
** Fatal: type kind T_INTEGER does not have item I_ELEM
[0x5612f8a3a8e9] ../src/object.c:254 object_lookup_failed
[0x5612f8a34224] ../src/type.c:329 type_elem.cold
[0x5612f8a556cd] ../src/parse.c:1946 get_subtype_for.lto_priv.0
[0x5612f8a5bfcd] ../src/parse.c:1990 apply_subtype_attribute
[0x5612f8a5bfcd] ../src/parse.c:2101 apply_type_attribute
[0x5612f8a5bfcd] ../src/parse.c:3287 p_attribute_name
[0x5612f8a5d3af] ../src/parse.c:3719 p_name.lto_priv.0
[0x5612f8a5add6] ../src/parse.c:4382 p_primary.lto_priv.0
[0x5612f8a5f84e] ../src/parse.c:4480 p_factor
[0x5612f8a5fbb9] ../src/parse.c:4519 p_term
[0x5612f8a5fe8d] ../src/parse.c:4574 p_simple_expression
[0x5612f8a6022a] ../src/parse.c:4612 p_shift_expression
[0x5612f8a606b9] ../src/parse.c:1505 p_relation
[0x5612f8a5a65a] ../src/parse.c:4693 p_expression_with_head.lto_priv.0
[0x5612f8a5aa2a] ../src/parse.c:4731 p_expression
[0x5612f8a5aa2a] ../src/parse.c:4259 p_aggregate_or_expression
[0x5612f8a5af8d] ../src/parse.c:4368 p_primary.lto_priv.0
[0x5612f8a5f84e] ../src/parse.c:4480 p_factor
[0x5612f8a5fbb9] ../src/parse.c:4519 p_term
[0x5612f8a5fe8d] ../src/parse.c:4574 p_simple_expression
[0x5612f8a6022a] ../src/parse.c:4612 p_shift_expression
[0x5612f8a60579] ../src/parse.c:4669 p_relation
[0x5612f8a5a65a] ../src/parse.c:4693 p_expression_with_head.lto_priv.0
[0x5612f8a63a9a] ../src/parse.c:4731 p_expression
[0x5612f8a63a9a] ../src/parse.c:6782 p_condition.lto_priv.0
[0x5612f8a76a5d] ../src/parse.c:9848 p_if_statement
[0x5612f8a76a5d] ../src/parse.c:10286 p_sequential_statement.lto_priv.0
[0x5612f8a6c80f] ../src/parse.c:8059 p_sequence_of_statements.lto_priv.0
[0x5612f8a6e523] ../src/parse.c:8122 p_subprogram_body.lto_priv.0
[0x5612f8a728d7] ../src/parse.c:9285 p_block_declarative_item.lto_priv.0
[0x5612f8a7e549] ../src/parse.c:12585 p_architecture_body
[0x5612f8a7e549] ../src/parse.c:12857 p_secondary_unit
[0x5612f8a7ee07] ../src/parse.c:12883 p_design_unit
[0x5612f8a7ee07] ../src/parse.c:12997 parse
[0x5612f8abb8e2] ../src/common.c:2421 analyse_file.part.0
[0x5612f8a45beb] ../src/nvc.c:2415 analyse
[0x5612f8a45beb] ../src/common.c:1920 process_command
[0x5612f8a414d6] ../src/nvc.c:2060 main

Please report this bug at https://github.com/nickg/nvc/issues
```

This does not occur with older versions of nvc before the latest commits.

nickg, 2024-05-26:
> The error I got now with another file, even before analyzing the package above:
> 

I think this should be fixed now, could you test again?

tmeissner, 2024-05-27:
This is fixed now, thanks for your efforts 👌

closed 2024-05-27 by tmeissner

nickg, 2024-10-20:
```vhdl
     function G_TO_STRING(data : G_DATA_TYPE) return string is to_string);
```

Using `to_string` here has become an error following the fix to #1025. The LRM says:

> If an interface subprogram declaration contains an interface subprogram default in the form of a subprogram name, the subprogram name shall denote a subprogram, and the denoted subprogram and the interface subprogram shall have conforming profiles (see 4.10).

I believe this means that the name `to_string` must denote a subprogram at the time the declaration is analysed, rather than when it's instantiated. GHDL has this interpretation too. Although I think this means a named subprogram cannot be used with a generic type.