nickg/nvc #903: nvc elaboration error when enabling branch coverage

tmeissner, 2024-06-26:
I often get nvc errors during elaboration when I enable branch coverage on fair complex designs and testbenches.

```
nvc --std=19 -L ./ -e --cover=statement,expression,fsm-state,functional,branch \
        --cover-spec=Coverage.sfile axififot

*** Caught signal 11 (SEGV_MAPERR) [address=0x45, ip=0x565277bf13db] ***

[0x565277b1b835] ../src/util.c:870 signal_handler.lto_priv.0
[0x7fdd37668f8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x565277bf13db] ../src/jit/jit-irgen.c:337 j_jump.lto_priv.0
[0x565277bfb334] ../src/jit/jit-irgen.c:2177 irgen_block.lto_priv.0
[0x565277bfe59e] ../src/jit/jit-irgen.c:4234 jit_irgen
[0x565277c15606] ../src/jit/jit-llvm.c:3532 llvm_aot_compile
[0x565277bd76b1] ../src/cgen.c:276 cgen_async_work.lto_priv.0
[0x565277bd5d6d] ../src/thread.c:890 execute_task.lto_priv.0
[0x565277bd5f27] ../src/thread.c:1050 steal_task.lto_priv.0
[0x565277bd617c] ../src/thread.c:1077 worker_thread
[0x565277bd5c59] ../src/thread.c:559 thread_wrapper
[0x7fdd376b5fd3] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7fdd3773581f] (/usr/lib/x86_64-linux-gnu/libc.so.6) __clone

Please report this bug at https://github.com/nickg/nvc/issues
```

If I disable branch coverage the simulation runs without this error.

 Maybe the error message helps, as I cannot give an MWE at the moment. The design is confidential and very complex.

Blebowski, 2024-06-26:
Hi,
@tmeissner maybe if you reconfigure with `--disable-lto`, rebuild NVC and re-run, the stack-trace could give
more precise information on the place in code where the error occurs.

nickg, 2024-06-26:
Adding `--enable-debug` to `configure` might be helpful too.

nickg, 2024-06-26:
Never mind: I managed to reproduce it. I'll push a fix shortly...

commit 031f4ce2 (2024-06-26): Fix crash with return in if-statement when branch coverage enabled

Fixes #903
closed 2024-06-26 by nickg