nickg/nvc #907: External name and redeclared type definition

yikkrubz, 2024-06-30:
The code D.vhd is extracted from a testbench that is successfully simulated by Questa,
but is not elaborated by nvc. 
When the loop is removed the code elaborates without error.

nvc -a D.vhd 
nvc -e D
** Error: type of signal E is not F
    > D.vhd:23
    |
 12 |   signal e : f;
    |          ^ declaration of E with type F
 ...
 23 |     alias e is <<signal .D.C.DUT.e : f>>;
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^ external name with type F

D.vhd
----------------------------------------
library ieee;
use ieee.std_logic_1164.all;

entity A is
  port (
    clk : in std_logic
  );
end A;

architecture rtl of A is
  type f is (ONE, TWO);
  signal e : f;
begin
end architecture;

entity B is
end B;

architecture sim of B is
  type f is (ONE, TWO);
begin
  E_PROC : process
    alias e is <<signal .D.C.DUT.e : f>>;
  begin
    loop         -- elaborates if commented out
      wait on e;
    end loop;    -- elaborates if commented out
  end process;
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity C is
end C;

architecture sim of C is
  signal clk : std_logic := '1';
begin
  DUT : entity work.A(rtl)
  port map (
    clk => clk
  );

  B : entity work.B(sim); 
end architecture;

entity D is
end D;

architecture sim of D is
begin
  C : entity work.C(sim);

  process
  begin
    wait;
  end process;
end architecture;

nvc 1.13-devel (1.12.0.r127.g326e923f) (Using LLVM 18.1.3)
macos 14.5 (intel)
[D.txt](https://github.com/user-attachments/files/16045058/D.txt)

nickg, 2024-06-30:
I think this error is correct: two types with the same name and structure are distinct if they are declared in different regions. The solution here is to move the definition of `f` into a package that is used by both architectures. I've pushed a change to make the error message more clear. 

The reason you don't get an error once the loop is removed is that NVC optimises out the empty process. I should prevent this optimisation in cases like this where it changes the behaviour.

yikkrubz, 2024-06-30:
Thank you very much for your reply.

closed 2024-06-30 by yikkrubz