nickg/nvc #908: nvc elaboration error (SIGABRT)

tmeissner, 2024-07-03:
I get a runtime error during elaboration with a testbench and design. The design itself runs in other tests without problems, so I assume the problem lies in the test bench. I've configured nvc with `--enable-debug` and `--disable-lto` switches to get more information about the error:

```
$ nvc --std=19 -L ./ -e --cover queueregistert
nvc: ../src/vcode.c:158: reg_array_nth_ptr: Assertion `n < a->count' failed.

*** Caught signal 6 (SIGABRT) ***

[0x55ead2f761b9] ../src/util.c:870 signal_handler
[0x7f063da5df8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7f063daacccc] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7f063da5def1] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x7f063da48471] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x7f063da48394] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7f063da56df1] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x55ead3008efe] ../src/vcode.c:158 vcode_reg_data
[0x55ead3008efe] ../src/vcode.c:599 vcode_reg_type
[0x55ead3008efe] ../src/vcode.c:604 vcode_reg_kind
[0x55ead2fed517] ../src/lower.c:12841 lower_rvalue
[0x55ead2ffd4da] ../src/lower.c:1231 lower_subprogram_arg
[0x55ead2ffdd2e] ../src/lower.c:2328 lower_builtin.part.0
[0x55ead2fea6af] ../src/lower.c:5285 lower_expr
[0x55ead2fed509] ../src/lower.c:12836 lower_rvalue
[0x55ead2fff7c8] ../src/lower.c:4856 lower_attr_param.constprop.0
[0x55ead2feba21] ../src/lower.c:5137 lower_attr_ref
[0x55ead2feba21] ../src/lower.c:5311 lower_expr
[0x55ead2fed509] ../src/lower.c:12836 lower_rvalue
[0x55ead2feac81] ../src/lower.c:3355 lower_array_ref
[0x55ead2feac81] ../src/lower.c:5297 lower_expr
[0x55ead30056ae] ../src/lower.c:12831 lower_process
[0x55ead2fc9d3e] ../src/elab.c:1975 elab_stmts
[0x55ead2fcccd4] ../src/elab.c:1498 elab_architecture
[0x55ead2fcd2f0] ../src/elab.c:2253 elab
[0x55ead2f70bb4] ../src/nvc.c:464 elaborate
[0x55ead2f70bb4] ../src/nvc.c:1922 process_command
[0x55ead2f6d4d0] ../src/nvc.c:2060 main
```
```
$ nvc --version
nvc 1.13-devel (1.12.0.r136.g95ad5561) (Using LLVM 14.0.6) [debug]
Copyright (C) 2011-2024  Nick Gasson
```

This time the `--cover` option hasn't any influence on the error.

Maybe the error message helps, as I cannot give an MWE at the moment. The design is confidential and complex.

nickg, 2024-07-03:
Do you have some code using the `'val` attribute in the testbench?

tmeissner, 2024-07-03:
Yepp. For example:

```vhdl
subtype t_queue_range is positive range 1 to 512;
type t_queue_dir is (H2C, C2H);
type t_queue_selects is array (t_queue_dir) of natural;

signal s_queue_selects : t_queue_selects;

process is
begin
...
for queue in 0 to t_queue_range'high-1 loop
  s_queue_selects(t_queue_dir'val(page-1)) <= queue;
end loop;
...
end process;
```

nickg, 2024-07-03:
If you comment out that (and any other uses of `'val`) does it elaborate ok? If so maybe you can try the opposite (comment out everything except that use of `'val`) to see if you can get a minimal reproducer.

nickg, 2024-07-04:
Thanks for the reproducer. Could you test again?

tmeissner, 2024-07-05:
Perfect! Now all tests run with nvc. Thanks :smiley:

closed 2024-07-05 by nickg