nickg/nvc #917: generic package with impure function method.

JimLewis, 2024-07-09:
I am running NVC-1.11.2

I tried this:
```vhdl
package ScoreboardGenericPkg is
  generic (
    type ExpectedType ;
    type ActualType ;
    impure function Match(Actual : ActualType ;                    
                   Expected : ExpectedType) return boolean ;       -- is "=" ;
    function expected_to_string(A : ExpectedType) return string ;  -- is to_string ;
    function actual_to_string  (A : ActualType) return string      -- is to_string ;
  ) ;
```

This resulted in the following error:
```
nvc --std=08 -H 128m --work=osvvm:C:/tools/VHDL_LIBS/NVC-1.11.2/OSVVM.08 -L C:/tools/VHDL_LIBS/NVC-1.11.2 -a C:/OsvvmLibraries/osvvm/ScoreboardGenericPkg.vhd
** Error: unexpected impure while parsing interface declaration, expecting one of constant, signal, variable, file, identifier, type, function, procedure or package
Error:     > C:\OsvvmLibraries\osvvm\ScoreboardGenericPkg.vhd:98
Error:     |
Error:  98 |     impure function Match(Actual : ActualType ;                           -- defaults
Error:     |     ^^^^^^ this token was unexpected
** Error: unexpected ) while parsing subprogram body, expecting is
Error:      > C:\OsvvmLibraries\osvvm\ScoreboardGenericPkg.vhd:102
Error:      |
Error:  102 |   ) ;
Error:      |   ^ this token was unexpected
** Error: unexpected end while parsing subprogram declarative item, expecting one of variable, type, alias, constant, function, procedure, impure, pure, attribute, subtype, use, file or group
```

JimLewis, 2024-07-09:
This was proposed in an issue to OSVVM.   

The proposed use model claims that one can map an pure function as an actual to the impure function function formal generic.   I could not find anything in the LRM that says this is illegal.   My tests showed it worked with GHDL, Questa, and RivieraPRO.   

I stopped testing and backed out the change when I ran into the issue with NVC.

nickg, 2024-07-09:
The code above is fine, it's just an oversight.

closed 2024-07-09 by nickg

nickg, 2024-07-13:
Try updating to the 1.13.0 release here: https://github.com/nickg/nvc/releases/tag/r1.13.0

commit 8092ca1d (2024-07-09): Allow function interface declarations to be pure or impure

Fixes #917