nickg/nvc #921: Regression with alias of external name?

yikkrubz, 2024-07-19:
The current git version of nvc does not elaborate F. The code elaborates sucessfully 
with ghdl and nvc version 1.12.2. Is this code valid?

F.vhd
--------
library ieee;
use ieee.std_logic_1164.all;

package A_pkg is
  procedure h;
end package;

package body A_pkg is
  procedure h is
    alias g is << signal .F.DUT.D.B.g : std_logic_vector >>; -- not accepted by current git version of nvc, accepted by nvc 1.12.2 and ghdl
    -- alias g is << signal .F.DUT.D.g : std_logic_vector >>; -- accepted by nvc and ghdl
  begin
  end procedure;
end package body;

library ieee;
use ieee.std_logic_1164.all;

entity B is
  port (
    clk : in std_logic;
    g : out std_logic_vector(31 downto 0)
  );
end B;

architecture rtl of B is
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use work.A_pkg.all;

entity C is
  port (
    clk : in std_logic;
    g : in std_logic_vector(31 downto 0)
  );
end C;

architecture rtl of C is

begin
  process(clk)
  begin
      h;
  end process;
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity D is
  port (
    clk : in std_logic
  );
end D;

architecture rtl of D is
  signal g : std_logic_vector(31 downto 0);
begin
  B : entity work.B(rtl)
    port map(
      clk => clk,
      g => g
    );

  C : entity work.C(rtl)
  port map (
    clk => clk,
    g => g
  );
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity E is
  port (
    clk : in std_logic
  );
end E;

architecture str of E is
begin
  D : entity work.D(rtl)
  port map (
    clk => clk
  );
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use std.env.finish;

entity F is
end F;

architecture sim of F is
  signal clk : std_logic := '1';
begin

  clk <= not clk after 5 ns;

  DUT : entity work.E(str)
    port map (
      clk => clk
    );

  PROC : process
  begin
    wait for 10 ns;
    finish;
  end process;

end architecture;

-----------------
nvc -a F.vhd 
nvc -e F 
** Error: external name G not found
    > F.vhd:10
    |
 10 |     alias g is << signal .F.DUT.D.B.g : std_logic_vector >>; -- not accepted by current git version of nvc, accepted by nvc 1.12.2 and ghdl
    |                           ^^^^^^^^^^^ name G not found inside B

nvc 1.14-devel (1.13.0.r7.g7a43c3cf) (Using LLVM 18.1.8)
macos 14.5 (intel)

/Users/Peter/Downloads/tmp/vw_risc-v/F.txt

nickg, 2024-07-21:
Yes it's a regression, thanks for reporting.

yikkrubz, 2024-07-21:
Thank you very much for taking care of this issue.

I hope I am not doing something stupid.
At least on my system, it seems the regression is not fixed by the patch.

nvc -a F.vhd
nvc -e F
** Error: external name G not found
    > F.vhd:10
    |
 10 |     alias g is << signal .F.DUT.D.B.g : std_logic_vector >> ; -- not accepted by current git version of nvc, accepted by nvc 1.12.2 and ghdl
    |                           ^^^^^^^^^^^ name G not found inside B

nvc 1.14-devel (1.13.0.r18.g809b75bd) (Using LLVM 18.1.8)
macos 14.5 (intel)

nickg, 2024-07-21:
Yes sorry I only fixed it for generics not ports. Your test case passes now.

closed 2024-07-21 by nickg

yikkrubz, 2024-07-21:
Thank you very much for fixing this issue.

commit 9e25d739 (2024-07-21): Cannot reference generic in external name. Fixes #921
commit cc6858f6 (2024-07-21): Cannot reference generic in external name. Fixes #921