nickg/nvc #922: Compliance-Tests: tb_garbage_collection.all failed to suspend thread

yikkrubz, 2024-07-20:
The Compliance testsuite does no longer finish. Running the tb_garbage_collection testbench causes the testsuite to hang. The error message is: Fatal: 0ms+0: failed to suspend thread 1 (268435459) 

(18:57:39) Starting vhdl_2019.tb_garbage_collection.all
Output file: /Users/Peter/Downloads/Compliance-Tests/vunit_out/test_output/vhdl_2019.tb_garbage_collection.all_f5a4bab440042444bf62e351cae2e49c16b78be5/output.txt
/usr/local/bin/nvc --work=vhdl_2019:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vhdl_2019 --std=2019 --map=vunit_lib:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vunit_lib --map=vhdl_2008:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vhdl_2008 --map=vhdl_2019:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vhdl_2019 -H 64m -e tb_garbage_collection-tb '-grunner_cfg=active python runner : true,enabled_test_cases : ,output path : /Users/Peter/Downloads/Compliance-Tests/vunit_out/test_output/vhdl_2019.tb_garbage_collection.all_f5a4bab440042444bf62e351cae2e49c16b78be5/,tb path : /Users/Peter/Downloads/Compliance-Tests/vhdl_2019/,use_color : true' --no-save --jit -r --exit-severity=error
** Fatal: 0ms+0: failed to suspend thread 1 (268435459)
   Function BIGSTRING [return LINE] at /Users/Peter/Downloads/Compliance-Tests/vhdl_2019/tb_030.vhd:11
   Process :tb_garbage_collection:u_e030:tb at /Users/Peter/Downloads/Compliance-Tests/vhdl_2019/tb_030.vhd:20
[0x10da3b407] diag_femit+0xe27
[0x10daf1ac1] fatal_trace+0xc4
[0x10da4090c] stop_world+0x29c
[0x10da57f31] mspace_gc+0xd1
[0x10da57c7b] mspace_alloc+0x7b
[0x10da96315] jit_interp+0xf75
[0x10da958b0] jit_interp+0x510
[0x10da77d23] jit_try_vcall+0x1e3
[0x10da60e36] async_run_process+0xa6
[0x10da624c0] model_cycle+0xdb0
[0x10da61550] model_run+0x100
[0x10d91ac1e] process_command+0x3b1e
[0x10d91ae48] process_command+0x3d48
[0x10d916ae8] main+0x578

Please report this bug at https://github.com/nickg/nvc/issues

nvc 1.14-devel (1.13.0.r7.g7a43c3cf) (Using LLVM 18.1.8)
macos 14.5 (intel)

yikkrubz, 2024-07-20:
When I configure nvc with --disable-lto and --enable-debug, the test run completes. This time running tb_garbage_collection causes Assertion failed: (thread->port != MACH_PORT_NULL), function stop_world, file thread.c, line 1329, followed by a SIGABRT (see below).

(22:40:46) Starting vhdl_2019.tb_garbage_collection.all
Output file: /Users/Peter/Downloads/Compliance-Tests/vunit_out/test_output/vhdl_2019.tb_garbage_collection.all_f5a4bab440042444bf62e351cae2e49c16b78be5/output.txt
/Users/Peter/Downloads/tmp/nvc-debug/bin/nvc --work=vhdl_2019:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vhdl_2019 --std=2019 --map=vunit_lib:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vunit_lib --map=vhdl_2008:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vhdl_2008 --map=vhdl_2019:/Users/Peter/Downloads/Compliance-Tests/vunit_out/nvc/libraries/vhdl_2019 -H 64m -e tb_garbage_collection-tb '-grunner_cfg=active python runner : true,enabled_test_cases : ,output path : /Users/Peter/Downloads/Compliance-Tests/vunit_out/test_output/vhdl_2019.tb_garbage_collection.all_f5a4bab440042444bf62e351cae2e49c16b78be5/,tb path : /Users/Peter/Downloads/Compliance-Tests/vhdl_2019/,use_color : true' --no-save --jit -r --exit-severity=error
Assertion failed: (thread->port != MACH_PORT_NULL), function stop_world, file thread.c, line 1329.

*** Caught signal 6 (SIGABRT) ***

[0x10d3ed273] signal_handler+0x2e3
[0x7ff811720fdc] (/usr/lib/system/libsystem_platform.dylib) _sigtramp+0x1c
[0x7ff8116b914b] (/usr/lib/system/libsystem_kernel.dylib) __pthread_kill+0xb
[0x7ff8116f1ebd] (/usr/lib/system/libsystem_pthread.dylib) pthread_kill+0x106
[0x7ff811617a79] (/usr/lib/system/libsystem_c.dylib) abort+0x7e
[0x7ff811616d68] (/usr/lib/system/libsystem_c.dylib) err
[0x10d552e73] stop_world.cold.1+0x23
[0x10d4a6efc] stop_world+0x1ac
[0x10d4b735b] mspace_gc+0x7b
[0x10d4b7180] mspace_alloc+0xa0
[0x10d4e7de2] interp_loop+0x1182
[0x10d4e6c06] jit_interp+0x126
[0x10d4e7230] interp_loop+0x5d0
[0x10d4e6c06] jit_interp+0x126
[0x10d4cb7fe] jit_try_vcall+0x14e
[0x10d4cb071] jit_fastcall+0x51
[0x10d4c5730] async_run_process+0xb0
[0x10d4c5924] deferq_run+0x54
[0x10d4bd21c] model_cycle+0x82c
[0x10d4bc998] model_run+0x158
[0x10d3e7d77] process_command+0x2777
[0x10d3e7eee] process_command+0x28ee
[0x10d3e50e8] main+0x418
fail (P=130 S=0 F=7 T=154) vhdl_2019.tb_garbage_collection.all (0.2 s)

nickg, 2024-07-21:
I can reproduce this on Arm-based macOS 14.5 too. Do you know if this test has ever passed on macOS? I don't think that code has changed for several releases.

yikkrubz, 2024-07-21:
When I ran the compliance tests before last time, I recorded the failing tests (see below). It seems the tb_garbage_collection.all test passed. 
I am sorry I did not note the date when I ran the tests. I assume the test was working on Sep 24, 2023 when I filed an issue (Compliance-tests). 

fail vhdl_2019.tb_composites_of_protected_types.all
fail vhdl_2019.tb_access_types_to_protected_types_and_composites_of_protected_types.all
fail vhdl_2019.tb_anonymous_types.all
fail vhdl_2019.tb_relax_library_requirement_on_configurations.all
fail vhdl_2019.tb_composition_with_protected_types.all
fail vhdl_2019.tb_protected_types_with_generic_clause.all
fail vhdl_2019.tb_api_and_attributes_for_psl.PSL attributes tests (1076-2019 16.2.7)
fail vhdl_2019.tb_api_and_attributes_for_psl.PSL API tests (1076-2019 16.5.8)
fail vhdl_2019.tb_map_generics_on_subprogram_call.all
fail vhdl_2019.tb_all_interface_lists_can_be_ordered.all
fail vhdl_2019.tb_extended_ranges.all

closed 2024-07-22 by nickg

yikkrubz, 2024-07-22:
Thank you very much for fixing this issue.

commit fc301245 (2024-07-22): Fix "failed to suspend thread" error on macOS. Fixes #922
commit da8cd19b (2024-07-24): Fix "failed to suspend thread" error on macOS. Fixes #922