nickg/nvc #923: OsvvmLibraries RunAllTests SetVHDLVersion 2019 SIGSEGV

yikkrubz, 2024-07-20:
When I try to run RunAllTests.pro, shipped with OsvvmLibraries, with SetVHDLVersion set to 2019, there are elaborate errors (Segmentation faults, see below).

All testcases pass when SetVHDLVersion is set to 2008.

OsvvmLibraries, tag: 2024.05b
nvc 1.14-devel (1.13.0.r7.g7a43c3cf) (Using LLVM 18.1.8)
macos 14.5 (macos)

% build ../RunAllTests.pro
build ../RunAllTests.pro
Starting Build at time 21:19:03
set CurrentWorkingDirectory ..
source ../RunAllTests.pro
include ./Common/RunAllTests.pro 
set CurrentWorkingDirectory ../Common
source ../Common/RunAllTests.pro
include ./TbStream 
set CurrentWorkingDirectory ../Common/TbStream
source ../Common/TbStream/TbStream.pro
library osvvm_TbStreamTransactionPkg /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel
nvc --std=19 --work=osvvm_tbstreamtransactionpkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 --init
TestSuite StreamTransactionPkg
include testbench 
set CurrentWorkingDirectory ../Common/TbStream/testbench
source ../Common/TbStream/testbench/testbench.pro
analyze OsvvmTestCommonPkg.vhd
nvc --std=19 -H 128m --work=osvvm_TbStreamTransactionPkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 -L /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel -a ../Common/TbStream/testbench/OsvvmTestCommonPkg.vhd

analyze TestCtrl_e.vhd
nvc --std=19 -H 128m --work=osvvm_TbStreamTransactionPkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 -L /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel -a ../Common/TbStream/testbench/TestCtrl_e.vhd

analyze TbStream.vhd
nvc --std=19 -H 128m --work=osvvm_TbStreamTransactionPkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 -L /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel -a ../Common/TbStream/testbench/TbStream.vhd

include StreamTransactionPkg 
set CurrentWorkingDirectory ../Common/TbStream/StreamTransactionPkg
source ../Common/TbStream/StreamTransactionPkg/StreamTransactionPkg.pro
RunTest TbStream_SendGet1.vhd
TestName TbStream_SendGet1
analyze TbStream_SendGet1.vhd
nvc --std=19 -H 128m --work=osvvm_TbStreamTransactionPkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 -L /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel -a ../Common/TbStream/StreamTransactionPkg/TbStream_SendGet1.vhd

Simulation Start time 21:19:03
simulate TbStream_SendGet1
nvc --std=19 -H 128m --work=osvvm_TbStreamTransactionPkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 -L /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel -e TbStream_SendGet1
Elaborate Error: 
Elaborate Error: *** Caught signal 11 (SIGSEGV) [address=0x6018038176a8, ip=0x100ae03c9] ***
Elaborate Error: 
Elaborate Error: [0x1009ef1b4] signal_handler+0x3d4
Elaborate Error: [0x7ff811720fdc] (/usr/lib/system/libsystem_platform.dylib) _sigtramp+0x1c
Elaborate Error: [0x100ae03ca] emit_load+0x1aa
Elaborate Error: [0x100abeb37] lower_name_attr+0xa07
Elaborate Error: [0x100abd441] lower_protected_init+0x211
Elaborate Error: [0x100a9d9d2] lower_decls+0x1cf2
Elaborate Error: [0x100aba16b] lower_package+0x4b
Elaborate Error: [0x100ab9c46] unit_registry_get+0x586
Elaborate Error: [0x100b4457b] jit_fill_irbuf+0xcb
Elaborate Error: [0x100b623e9] jit_interp+0x49
Elaborate Error: [0x100b628b1] jit_interp+0x511
Elaborate Error: [0x100b628b1] jit_interp+0x511
Elaborate Error: [0x100b44d24] jit_try_vcall+0x1e4
Elaborate Error: [0x100b45778] jit_try_call+0xe8
Elaborate Error: [0x100a6db36] elab_lower+0x106
Elaborate Error: [0x100a6c856] elab_architecture+0x786
Elaborate Error: [0x100a6acdb] elab+0x96b
Elaborate Error: [0x1009e5161] process_command+0x1061
Elaborate Error: [0x1009e3ae9] main+0x579
Elaborate Error: 
Elaborate Error: Please report this bug at https://github.com/nickg/nvc/issues
Simulation Finish time 21:19:03, Elapsed time: 0:00:00 
SimulateError: See messages above in "simulate TbStream_SendGet1"
For tcl errorInfo, puts $::osvvm::SimulateErrorInfo

yikkrubz, 2024-07-20:
When I configure nvc with --disable-lto and --enable-debug, the error message is:
 Elaborate Error: Assertion failed: (var != VCODE_INVALID_VAR), function lower_name_attr, file lower.c, line 1532. (see below).

Simulation Start time 23:06:19
simulate TbStream_SendGet1
nvc --std=19 -H 128m --work=osvvm_TbStreamTransactionPkg:/Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel/OSVVM_TBSTREAMTRANSACTIONPKG.19 -L /Users/Peter/Downloads/OsvvmLibraries/build/VHDL_LIBS/NVC-1.14-devel -e TbStream_SendGet1
Elaborate Error: Assertion failed: (var != VCODE_INVALID_VAR), function lower_name_attr, file lower.c, line 1532.
Elaborate Error: 
Elaborate Error: *** Caught signal 6 (SIGABRT) ***
Elaborate Error: 
Elaborate Error: [0x10ea2f273] signal_handler+0x2e3
Elaborate Error: [0x7ff811720fdc] (/usr/lib/system/libsystem_platform.dylib) _sigtramp+0x1c
Elaborate Error: [0x7ff8116b914b] (/usr/lib/system/libsystem_kernel.dylib) __pthread_kill+0xb
Elaborate Error: [0x7ff8116f1ebd] (/usr/lib/system/libsystem_pthread.dylib) pthread_kill+0x106
Elaborate Error: [0x7ff811617a79] (/usr/lib/system/libsystem_c.dylib) abort+0x7e
Elaborate Error: [0x7ff811616d68] (/usr/lib/system/libsystem_c.dylib) err
Elaborate Error: [0x10eb80bd3] lower_name_attr.cold.2+0x23
Elaborate Error: [0x10eab3ac7] lower_name_attr+0x2d7
Elaborate Error: [0x10eab0b68] lower_protected_init+0xa8
Elaborate Error: [0x10ea9de79] lower_decls+0x14d9
Elaborate Error: [0x10eaa7cd2] lower_package+0x52
Elaborate Error: [0x10eaa78a6] unit_registry_get+0x256
Elaborate Error: [0x10eb0cc32] jit_fill_irbuf+0xe2
Elaborate Error: [0x10eb28b27] jit_interp+0x47
Elaborate Error: [0x10eb29230] interp_loop+0x5d0
Elaborate Error: [0x10eb28c06] jit_interp+0x126
Elaborate Error: [0x10eb29230] interp_loop+0x5d0
Elaborate Error: [0x10eb28c06] jit_interp+0x126
Elaborate Error: [0x10eb0d7fe] jit_try_vcall+0x14e
Elaborate Error: [0x10eb0d46b] jit_try_call+0xfb
Elaborate Error: [0x10eae14d6] eval_instance+0x36
Elaborate Error: [0x10ea7e8ca] elab_lower+0x7a
Elaborate Error: [0x10ea7cbcf] elab_architecture+0x44f
Elaborate Error: [0x10ea7c447] elab+0x4f7
Elaborate Error: [0x10ea29926] process_command+0x2326
Elaborate Error: [0x10ea270e8] main+0x418
Simulation Finish time 23:06:19, Elapsed time: 0:00:00 
SimulateError: See messages above in "simulate TbStream_SendGet1"
For tcl errorInfo, puts $::osvvm::SimulateErrorInfo

nickg, 2024-07-24:
I've verified all the OSVVM regression tests pass in 2019 mode now.

commit 4c61fc5d (2024-07-24): Fix crash with shared variable in generic package. Fixes #923
commit 12737b0f (2024-07-24): Fix crash with shared variable in generic package. Fixes #923
closed 2024-07-24 by nickg