nickg/nvc #927: Generic package instantiation with type from package

amb5l, 2024-07-23:
I get a compilation error when instantiating a generic package with a type defined in another package. To reproduce, compile the following:
```
package gen_pkg is
  generic (
    type item_t
  );
end package gen_pkg;

library ieee;
  use ieee.std_logic_1164.all;

package type_pkg is
  subtype r_item_t is std_ulogic_vector(15 downto 0);
  constant R_EMPTY : r_item_t := "XXXXXXXXXXXXXXXX";
end package type_pkg;

use work.type_pkg.all;
package r_gen_pkg is
  new work.gen_pkg generic map(item_t => r_item_t);
```
Here is an example transcript:
```
nvc -H 128M --std=2008 --work=work:work -a --relaxed bug.vhd
** Error: cannot find predefined "=" operator for type R_ITEM_T
    > D:\work\tyto2\test\hram\tb_model_hram\bug.vhd:20
    |
 20 | package r_gen_pkg is
    | ^
** Error: cannot find predefined "/=" operator for type R_ITEM_T
    > D:\work\tyto2\test\hram\tb_model_hram\bug.vhd:20
    |
 20 | package r_gen_pkg is
    | ^
```
I am running the following version of NVC on Windows 11:
`nvc 1.13-devel (1.12.0.r4.ge7190884) (Using LLVM 17.0.6)`

This code compiles correctly in ModelSim, but produces the same error in GHDL.

amb5l, 2024-07-23:
Having reported the same issue for GHDL, Tristan pointed out that a library was missing, and this fixed the "issue" in GHDL and NVC:
```
package gen_pkg is
  generic (
    type item_t
  );
end package gen_pkg;

library ieee;
  use ieee.std_logic_1164.all;

package type_pkg is
  subtype r_item_t is std_ulogic_vector(15 downto 0);
  constant R_EMPTY : r_item_t := "XXXXXXXXXXXXXXXX";
end package type_pkg;

use work.type_pkg.all;
-- THE 2 LINES BELOW FIX THE ISSUE
library ieee;
  use ieee.std_logic_1164.all;
package r_gen_pkg is
  new work.gen_pkg generic map(item_t => r_item_t);
```

closed 2024-07-23 by amb5l

nickg, 2024-07-23:
I think Modelsim may actually be correct here. There is some unusual wording in the 2008 LRM about those implicit `"="` and `"/="` generics which was changed in 2019 so they behave like user defined generic subprograms with `<>`. I'll check the LRM text tomorrow. Do you get a different error with `--std=2019`?

amb5l, 2024-07-23:
With 2019 (and NVC built from today's master branch) I see this:
```
nvc -H 128M --std=2019 --work=work:work -a --relaxed bug.vhd
** Error: no visible subprogram "=" matches signature [ITEM_T, ITEM_T return BOOLEAN]
    > D:\work\tyto2\test\hram\tb_model_hram\bug.vhd:25
    |
  3 |     type item_t
    |     ^^^^^^^^^^^ while resolving interface subprogram default for "="
 ...
 25 | package r_gen_pkg is
    | ^ error occurred here
** Error: no visible subprogram "/=" matches signature [ITEM_T, ITEM_T return BOOLEAN]
    > D:\work\tyto2\test\hram\tb_model_hram\bug.vhd:25
    |
  3 |     type item_t
    |     ^^^^^^^^^^^ while resolving interface subprogram default for "/="
 ...
 25 | package r_gen_pkg is
    | ^ error occurred here
```

nickg, 2024-07-31:
I'm not entirely convinced by the discussion in the GHDL ticket, but making the behaviour the same between 2008 and 2019 does simplify the code a lot so I've gone ahead and changed it. 

In 2008 mode there's now a more verbose error message which should help users who hit this in the future:

```
** Error: no visible subprogram "=" matches signature [ITEM_T, ITEM_T return BOOLEAN]
    > test.vhd:16
    |
  3 |     type item_t
    |     ^^^^^^^^^^^ while resolving interface subprogram default for "="
 ...
 16 | package r_gen_pkg is
    | ^ error occurred here
    |
    = Note: add a use clause that makes the operators for WORK.TYPE_PKG.R_ITEM_T visible
    = Note: the 2008 LRM is unclear on the visibility requirements for the implicit operators 
            associated with a generic type, and behaviour varies between implementations; the 
            interpretation here is consistent with the 2019 LRM
    = Help: IEEE Std 1076-2008 section 6.5.3 "Interface type declarations"
```