nickg/nvc #935: UVVM AXI-Stream exception error

erick166, 2024-08-08:
I get the following error when running a testcase with the AXI-Stream VVC from UVVM:

```
*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000024425B7A200, ip=00007FF64EB05B87] ***

[00007FF64EA46EE0] 
[00007FF64EA47299] 
[00007FF9097C0B1C] UnhandledExceptionFilter+0x1ec
[00007FF90BD796BD] RtlCopyMemory+0x2bbd
[00007FF90BD5F667] _C_specific_handler+0x97
[00007FF90BD7504F] _chkstk+0x12f
[00007FF90BCEE866] RtlFindCharInUnicodeString+0xa96
[00007FF90BD7403E] KiUserExceptionDispatcher+0x2e
[00007FF64EB05B87] 
[00007FF64EB05349] 
[00007FF64EBD9A36] _nvc_mspace_alloc+0x66
[00000244010449ED] BITVIS_VIP_AXISTREAM.TD_CMD_QUEUE_PKG.T_GENERIC_QUEUE.ADD(I17T_GENERIC_ELEMENT)+0x4ed [VHDL]
[0000024401073349] BITVIS_VIP_AXISTREAM.TD_CMD_QUEUE_PKG.T_GENERIC_QUEUE.PUT(I17T_GENERIC_ELEMENT)+0x49 [VHDL]
[0000024401073259] BITVIS_VIP_AXISTREAM.TD_CMD_QUEUE_PKG.T_GENERIC_QUEUE.PUT(17T_GENERIC_ELEMENT)+0x59 [VHDL]
[000002440102345A] BITVIS_VIP_AXISTREAM.TD_VVC_ENTITY_SUPPORT_PKG.PUT_COMMAND_ON_QUEUE(49BITVIS_VIP_AXISTREAM.VVC_CMD_PKG.T_VVC_CMD_RECORD53BITVIS_VIP_AXISTREAM.TD_CMD_QUEUE_PKG.T_GENERIC_QUEUE49BITVIS_VIP_AXISTREAM.VVC_METHODS_PKG.T_VVC_STATUSsB)+0x15a [VHDL]
[0000024401022183] BITVIS_VIP_AXISTREAM.AXISTREAM_VVC_SLV_ARRAY_TB.I_AXISTREAM_VVC_S.CMD_INTERPRETER+0x783 [VHDL]
[00007FF64EB22469] std_env_get_assert_format+0x7739
[00007FF64EB0E6A1] nvc_current_delta+0x5611
[00007FF64EB0FC3C] nvc_current_delta+0x6bac
[00007FF64EA40238] 
[00007FF64EA3E257] 
[00007FF64EA38D48] 
[00007FF64D9ED9B5] 
[00007FF64D9EDA06] 

Please report this bug at https://github.com/nickg/nvc/issues
```

Here is the testcase I'm running:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library bitvis_vip_axistream;
context bitvis_vip_axistream.vvc_context;

entity axistream_vvc_tb is
end entity axistream_vvc_tb ;

architecture tb of axistream_vvc_tb is

  constant C_DATA_WIDTH        : natural := 32;
  constant C_USER_WIDTH        : natural := 8;
  constant C_ID_WIDTH          : natural := 7;
  constant C_DEST_WIDTH        : natural := 4;
  constant C_CLK_PERIOD        : time    := 10 ns;
  constant C_MAX_BYTES         : natural := 100;
  constant C_MAX_BYTES_IN_WORD : natural := 4;

  signal clk          : std_logic := '0';
  signal axistream_if : t_axistream_if(tdata(C_DATA_WIDTH - 1 downto 0),
                                       tkeep((C_DATA_WIDTH / 8) - 1 downto 0),
                                       tuser(C_USER_WIDTH - 1 downto 0),
                                       tstrb((C_DATA_WIDTH / 8) - 1 downto 0),
                                       tid(C_ID_WIDTH - 1 downto 0),
                                       tdest(C_DEST_WIDTH - 1 downto 0));

begin

  i_axistream_vvc_m : entity bitvis_vip_axistream.axistream_vvc
    generic map(
      GC_VVC_IS_MASTER => true,
      GC_DATA_WIDTH    => C_DATA_WIDTH,
      GC_USER_WIDTH    => C_USER_WIDTH,
      GC_ID_WIDTH      => C_ID_WIDTH,
      GC_DEST_WIDTH    => C_DEST_WIDTH,
      GC_INSTANCE_IDX  => 0
    )
    port map(
      clk              => clk,
      axistream_vvc_if => axistream_if
    );

  i_axistream_vvc_s : entity bitvis_vip_axistream.axistream_vvc
    generic map(
      GC_VVC_IS_MASTER => false,
      GC_DATA_WIDTH    => C_DATA_WIDTH,
      GC_USER_WIDTH    => C_USER_WIDTH,
      GC_ID_WIDTH      => C_ID_WIDTH,
      GC_DEST_WIDTH    => C_DEST_WIDTH,
      GC_INSTANCE_IDX  => 1
    )
    port map(
      clk              => clk,
      axistream_vvc_if => axistream_if
    );

  i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;

  p_clock : clock_generator(clk, C_CLK_PERIOD);

  p_main : process
    variable v_num_bytes  : integer                            := 0;
    variable v_num_words  : integer                            := 0;
    variable v_user_array : t_user_array(0 to C_MAX_BYTES - 1) := (others => (others => '0'));
    variable v_strb_array : t_strb_array(0 to C_MAX_BYTES - 1) := (others => (others => '0'));
    variable v_id_array   : t_id_array(0 to C_MAX_BYTES - 1)   := (others => (others => '0'));
    variable v_dest_array : t_dest_array(0 to C_MAX_BYTES - 1) := (others => (others => '0'));
    variable v_iterations : natural                            := 0;

    impure function get_slv_array(
      num_bytes     : integer;
      bytes_in_word : integer
    ) return t_slv_array is
      variable v_return_array : t_slv_array(0 to num_bytes - 1)((bytes_in_word * 8) - 1 downto 0);
    begin
      for byte in 0 to num_bytes - 1 loop
        v_return_array(byte) := random(v_return_array(byte)'length);
      end loop;
      return v_return_array;
    end function get_slv_array;

    procedure transmit_and_expect(
      data_array : t_slv_array;
      user_array : t_user_array;
      strb_array : t_strb_array;
      id_array   : t_id_array;
      dest_array : t_dest_array
    ) is
    begin
      axistream_transmit(AXISTREAM_VVCT, 0, data_array, "transmit");
      axistream_expect(AXISTREAM_VVCT, 1, data_array, "expect");
    end procedure transmit_and_expect;

  begin

    await_uvvm_initialization(VOID);

    disable_log_msg(ALL_MESSAGES);
    enable_log_msg(ID_LOG_HDR);
    enable_log_msg(ID_SEQUENCER);
    disable_log_msg(AXISTREAM_VVCT, ALL_INSTANCES, ALL_MESSAGES);

    ------------------------------------------------------------
    log(ID_LOG_HDR, "Start simulation");
    ------------------------------------------------------------
    for bytes_in_word in 1 to C_MAX_BYTES_IN_WORD loop
      v_iterations := 200;
      for i in 1 to v_iterations loop
        log(ID_SEQUENCER, to_string((bytes_in_word-1) * v_iterations + i) & "- bytes_in_word: " & to_string(bytes_in_word) & ", i: " & to_string(i));
        v_num_bytes := random(1, C_MAX_BYTES / bytes_in_word);
        v_num_words := integer(ceil(real(v_num_bytes * bytes_in_word) / (real(C_DATA_WIDTH) / 8.0)));

        for byte in 0 to v_num_words - 1 loop
          v_user_array(byte) := random(v_user_array(0)'length);
          v_strb_array(byte) := random(v_strb_array(0)'length);
          v_id_array(byte)   := random(v_id_array(0)'length);
          v_dest_array(byte) := random(v_dest_array(0)'length);
        end loop;

        transmit_and_expect(get_slv_array(v_num_bytes, bytes_in_word),
                            v_user_array(0 to v_num_words - 1),
                            v_strb_array(0 to v_num_words - 1),
                            v_id_array(0 to v_num_words - 1),
                            v_dest_array(0 to v_num_words - 1));
      end loop;
      await_completion(AXISTREAM_VVCT, 0, 1 ms);
      await_completion(AXISTREAM_VVCT, 1, 1 ms);
    end loop;

    report_alert_counters(FINAL);
    log(ID_LOG_HDR, "SIMULATION COMPLETED");
    std.env.stop;
    wait;
  end process p_main;
end architecture tb;

```

nickg, 2024-08-08:
This one runs ok for me (but I need to add `-H 1g` to stop it running out of memory). 

```
UVVM:      ===========================================================================================================
=========================================================
UVVM:      >> Simulation SUCCESS: No mismatch between counted and expected serious alerts
UVVM:      ===========================================================================================================
=========================================================
UVVM:
UVVM:
UVVM:
UVVM:
UVVM: ID_LOG_HDR                    109792.5 ns  TB seq.                        SIMULATION COMPLETED
UVVM: ----------------------------------------------------------------------------------------------------------------
---------------------------------------------------------
** Note: 109792500ps+2: STOP called
   Procedure STOP [] at C:\msys64\home\nick\nvc\build\lib\std.08\env-body.vhd:32
   Process :axistream_vvc_tb:p_main at C:\msys64\home\nick\nvc\build\test.vhd:141
```

Which UVVM version are you using?

erick166, 2024-08-09:
I am using the latest version v2 2024.07.03.

I see now that I was originally getting this error message:
`bitvis_vip_axistream\src\transaction_pkg.vhd:29:1: fatal: out of memory attempting to allocate 16519272 byte object`

But then I increased the size of the simulation heap too much, I used -H4096m so I got that error. I tested with -H1g and it works as well.

nickg, 2024-08-09:
> But then I increased the size of the simulation heap too much, I used -H4096m so I got that error. I tested with -H1g and it works as well.

Yes it happens when the heap size is >= 4g, thanks.

commit d2f3f683 (2024-08-09): Fix random segfault when heap is larger than 4g. Fixes #935
commit 25621cc1 (2024-08-11): Fix random segfault when heap is larger than 4g. Fixes #935
closed 2024-08-09 by nickg