nickg/nvc #938: LLVM 17 build

Blebowski, 2024-08-09:
Hi,
I am attempting to build NVC on RHEL8 after a machine upgrade with all packages bumped at their latest version.
In such case, my current LLVM version is:
```
sudo yum list installed | grep llvm
llvm.x86_64                                   17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm-devel.x86_64                             17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm-googletest.x86_64                        17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm-libs.x86_64                              17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm-static.x86_64                            17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm-test.x86_64                              17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm-toolset.x86_64                           17.0.6-2.module+el8.10.0+21256+978ccea6            @rhel-8-for-x86_64-appstream-rpms        
llvm7.0.x86_64                                7.0.1-7.el8                                        @epel                                    
llvm7.0-devel.x86_64                          7.0.1-7.el8                                        @epel                                    
llvm7.0-libs.x86_64                           7.0.1-7.el8                                        @epel         
```
I manage to `./autogen.sh`, `./configure`, `make` and `sudo make install` without problem (nothing fails).
Launching NVC fails however:
```
nvc -help
nvc: error while loading shared libraries: libLLVM-16.so: cannot open shared object file: No such file or directory
```

I tried building with `--with-llvm` and passing path to `llvm-config-17`, that did not help.

I see that LLVM-17 is not officially supported by NVC (Readme says versions 8-16). Is there a reason for this (API change ? ) ?
Either way, I would expect `configure` to fail instead of building correctly and failing only upon tool launch.

nickg, 2024-08-09:
Did you do a `make clean` between those steps? The build process involves running the `nvc` binary so I don't see why it work when run from the build directory but not after install. You can see which LLVM it's linked against with `ldd bin/nvc`. Running `make V=1` to show the command lines might also be helpful.

> I see that LLVM-17 is not officially supported by NVC (Readme says versions 8-16). Is there a reason for this (API change ? ) ?

I just haven't updated the README since those versions were released. It works fine with LLVM 18 on macOS.

Blebowski, 2024-08-11:
I found the issue in my setup. I had network disc installation folder earlier in my path, so launching NVC was not done on the local build in `/usr/bin`. It works OK from `1.11` release above. On `1.10` I still needed to use LLVM 16, and link statically to get it the running on the "updated machine".

closed 2024-08-11 by Blebowski