nickg/nvc #943: Exception std_env_get_assert_format

avelure, 2024-08-12:
This one I'm not able to reduce to an MWE. 
When I remove a reference to a signal that uses an enum from another package the exception occurs.
Then it disappears again when I remove all references to a UVVM "disable_log_msg" for a VVC.
I'm just working on reducing a testcase, so all these packages and enums has been used with nvc for years.
I'm using nvc 1.14-devel (1.13.0.r34.g1e0063fe) (Using LLVM 18.1.6) with msys clang64
```
*****************************************************************************************************
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
*****************************************************************************************************

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000008, ip=00007FFC02861326] ***

[00007FF67AAF3320]
[00007FF67AAF36D9]
[00007FFC02C00B1C] UnhandledExceptionFilter+0x1ec
[00007FFC053596BD] RtlCopyMemory+0x2bbd
[00007FFC0533F667] _C_specific_handler+0x97
[00007FFC0535504F] _chkstk+0x12f
[00007FFC052CE866] RtlFindCharInUnicodeString+0xa96
[00007FFC0535403E] KiUserExceptionDispatcher+0x2e
[00007FFC02861326] memmove+0x126
[00007FF67AC72098] vhpi_is_printable+0x922f8
[00007FF67AC709A5] vhpi_is_printable+0x90c05
[00007FF67AC709A5] vhpi_is_printable+0x90c05
[00007FF67ABC75D7] std_env_get_assert_format+0x2d87
[00007FF67ABC7E4A] std_env_get_assert_format+0x35fa
[00007FF67AB47D6B]
[00007FF67AB453E7]
[00007FF67AAE8E59]
[00007FF67AAE2A2B]
[00007FF67AAD1311]
[00007FF67AAD1366]
[00007FFC04DC257D] BaseThreadInitThunk+0x1d
[00007FFC0530AF28] RtlUserThreadStart+0x28

Please report this bug at https://github.com/nickg/nvc/issues
```
```
(lldb) run
(rocess 80160 launched: 'C:\msys64\clang64\bin\nvc.exe' (x86_64)
Process 80160 stopped
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ffc02861326: Access violation reading location 0x00000008
    frame #0: 0x00007ffc02861326 ucrtbase.dll`memcpy + 294
ucrtbase.dll`memcpy:
->  0x7ffc02861326 <+294>: movdqu (%rdx), %xmm1
    0x7ffc0286132a <+298>: movdqu -0x10(%rdx,%r8), %xmm2
    0x7ffc02861331 <+305>: movdqu %xmm1, (%rcx)
    0x7ffc02861335 <+309>: movdqu %xmm2, -0x10(%rcx,%r8)
(lldb) bt
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ffc02861326: Access violation reading location 0x00000008
  * frame #0: 0x00007ffc02861326 ucrtbase.dll`memcpy + 294
    frame #1: 0x00007ff67ac72098 nvc.exe`jit_interp [inlined] interp_move(state=0x0000007de4cfe900, ir=<unavailable>) at jit-interp.c:761:4
    frame #2: 0x00007ff67ac71fe6 nvc.exe`jit_interp [inlined] interp_loop(state=0x0000007de4cfe900) at jit-interp.c:1010:10
    frame #3: 0x00007ff67ac7057e nvc.exe`jit_interp(f=<unavailable>, caller=<unavailable>, args=<unavailable>, tlab=<unavailable>) at jit-interp.c:1104:4
    frame #4: 0x00007ff67ac709a5 nvc.exe`jit_interp [inlined] interp_call(state=0x0000007de4cfebc0, ir=0x0000025989e14238) at jit-interp.c:638:7
    frame #5: 0x00007ff67ac7096d nvc.exe`jit_interp [inlined] interp_loop(state=0x0000007de4cfebc0) at jit-interp.c:971:10
    frame #6: 0x00007ff67ac7057e nvc.exe`jit_interp(f=<unavailable>, caller=<unavailable>, args=<unavailable>, tlab=<unavailable>) at jit-interp.c:1104:4
    frame #7: 0x00007ff67ac709a5 nvc.exe`jit_interp [inlined] interp_call(state=0x0000007de4cff040, ir=0x000002598418d460) at jit-interp.c:638:7
    frame #8: 0x00007ff67ac7096d nvc.exe`jit_interp [inlined] interp_loop(state=0x0000007de4cff040) at jit-interp.c:971:10
    frame #9: 0x00007ff67ac7057e nvc.exe`jit_interp(f=<unavailable>, caller=<unavailable>, args=<unavailable>, tlab=<unavailable>) at jit-interp.c:1104:4
    frame #10: 0x00007ff67abc75d7 nvc.exe`jit_try_vcall(j=0x0000025983d48380, f=0x0000025984181ad0, result=0x0000007de4cff510, args=0x0000007de4cff190, tlab=0x0000007de4cff180) at jit-core.c:666:7
    frame #11: 0x00007ff67abc7e4a nvc.exe`jit_try_call(j=0x0000025983d48380, handle=<unavailable>, result=0x0000007de4cff510) at jit-core.c:728:11
    frame #12: 0x00007ff67ab47d6b nvc.exe`elab_lower [inlined] eval_instance(jit=0x0000025983d48380, name=0x0000025981eadac0, context=0x0000000000000000) at eval.c:424:9
    frame #13: 0x00007ff67ab47d40 nvc.exe`elab_lower(b=<unavailable>, shape=<unavailable>, ctx=0x0000007de4cff580) at elab.c:1424:19
    frame #14: 0x00007ff67ab453e7 nvc.exe`elab_architecture(bind=0x0000025985a20030, arch=<unavailable>, config=<unavailable>, ctx=0x0000007de4cff7b0) at elab.c:1539:7
    frame #15: 0x00007ff67aae8e59 nvc.exe`process_command [inlined] elab(top=<unavailable>, jit=<unavailable>, ur=0x0000025983fd3740, cover=0x0000000000000000) at elab.c:2306:10
    frame #16: 0x00007ff67aae868c nvc.exe`process_command [inlined] elaborate(argc=10, argv=0x0000025981e2a540, state=0x0000007de4cffa40) at nvc.c:464:17
    frame #17: 0x00007ff67aae868c nvc.exe`process_command(argc=10, argv=0x0000025981e2a540, state=0x0000007de4cffa40) at nvc.c:1922:14
    frame #18: 0x00007ff67aae2a2b nvc.exe`main(argc=<unavailable>, argv=<unavailable>) at nvc.c:2060:20
    frame #19: 0x00007ff67aad1311 nvc.exe`__tmainCRTStartup at crtexe.c:267:15
[... 96 lines cut ...]
      i11 := store r48
      r49 := array ref r43 offset r47   // @<0..2> => 0..2
      r50 := link package UVVM_UTIL.ADAPTATIONS_PKG // P<UVVM_UTIL.ADAPTATIONS_PKG>
      r51 := link var r50 C_SB_MSG_ID_PANEL_DEFAULT // @<0..2> => 0..2
      r49 := copy r51 count r41
      r52 := cmp r48 == r42
      cond r52 then 4 else 3
   4: r60 := const 101                  // # => 101
      r61 := index VR_ENTERED_CNT       // @<-2^31..2^31-1> => -2^31..2^31-1
      r62 := const -1                   // -2^31..2^31-1 => -1
      r61 := memset r62 length r60
      r65 := index VR_MATCH_CNT         // @<-2^31..2^31-1> => -2^31..2^31-1
      r65 := memset r62 length r60
      r68 := index VR_MISMATCH_CNT      // @<-2^31..2^31-1> => -2^31..2^31-1
      r68 := memset r62 length r60
      r71 := index VR_DROP_CNT          // @<-2^31..2^31-1> => -2^31..2^31-1
      r71 := memset r62 length r60
      r74 := index VR_INITIAL_GARBAGE_CNT // @<-2^31..2^31-1> => -2^31..2^31-1
      r74 := memset r62 length r60
      r77 := index VR_DELETE_CNT        // @<-2^31..2^31-1> => -2^31..2^31-1
      r77 := memset r62 length r60
      r80 := index VR_OVERDUE_CHECK_CNT // @<-2^31..2^31-1> => -2^31..2^31-1
      r80 := memset r62 length r60
      return
```

nickg, 2024-08-12:
Maybe try configuring with `--enable-debug` and see if you hit any assertion failures.

avelure, 2024-08-13:
I forgot to save the problematic test, but this one seems to be the same. It is in the same T_GENERIC_SB function but the stack is slightly different.

```
------------------------------------------------------------
**redacted**.TRANSACTION_PKG.**redacted**SB_PKG.T_GENERIC_SB:
        $GALLOC       R1, #0x36b8
        RECV          R0, #0
        STORE.64      R0, [R1]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:543>, 0:13
        MOV           R2, #40
        $COPY         R2, [R1+8], [CP+0]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 0:20
        STORE.64      #0, [R1+2472]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 0:21
L1:     DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:0
        LOAD.64       R3, [R1+2472]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:1
        ADD           R4, R3, #1
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:2
        STORE.64      R4, [R1+2472]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:3
        MUL           R6, R3, #24
        LEA           R5, [R1+48]
        ADD           R7, R5, R6
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:4
        LOAD.64       R8, [R1]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:5
        MOV           R9, #24
        $MOVE         R9, [R7], [R8+8]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:6
        CMP.EQ        R4, #101
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:544>, 1:7
        JUMP.F        L1
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:545>, 2:7
        MOV           R10, #101
        $BZERO        R10, [R1+2480]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:546>, 2:12
        $GETPRIV      R11, <**redacted**.TRANSACTION_PKG.**redacted**SB_PKG.SB_QUEUE_PKG>
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:546>, 2:13
        SEND          #0, R11
        CALL          <**redacted**.TRANSACTION_PKG.**redacted**SB_PKG.SB_QUEUE_PKG.T_GENERIC_QUEUE>
        RECV          R12, #0
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:546>, 2:14
        STORE.64      R12, [R1+2584]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 2:20
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 2:25
        STORE.64      #0, [R1+2472]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 2:26
L2:     DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:0
        LOAD.64       R13, [R1+2472]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:1
        ADD           R14, R13, #85
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:2
        STORE.64      R14, [R1+2472]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:3
        LEA           R15, [R1+2592]
        ADD           R16, R15, R13
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:4
        $GETPRIV      R17, <UVVM_UTIL.ADAPTATIONS_PKG>
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:6
        MOV           R18, #85
        $MOVE         R18, [R16], [R17+1072]
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:7
        CMP.EQ        R14, #0x2189
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:549>, 3:8
        JUMP.F        L2
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:552>, 4:7
        MOV           R19, #404
        $MEMSET.32    R19, [R1+11180], #-1
        DEBUG         <C:\proj\**redacted**\tools\uvvm\bitvis_vip_scoreboard\src\generic_sb_pkg.vhd:553>, 4:13
[... 106 lines cut ...]
```

If I break on exit GDB gives
```
#0  0x00007ffc0282be20 in ucrtbase!exit () from C:\Windows\System32\ucrtbase.dll
#1  0x00007ff6fb3f8565 in fatal_exit (status=status@entry=1) at ../src/util.c:504
#2  0x00007ff6fb3f7712 in fatal_trace (fmt=0x7ff6fb5902cc <__make_binary.tbl_BINARY+1662> "assertion '%s' failed") at ../src/util.c:586
#3  0x00007ff6fb51caac in interp_loop (state=state@entry=0x14ed28) at ../src/jit/jit-interp.c:1059
#4  0x00007ff6fb51a1c3 in jit_interp (f=0x92f7620, caller=<optimized out>, args=0x14f6b0, tlab=0x14f6a0) at ../src/jit/jit-interp.c:1104
#5  0x00007ff6fb51a822 in interp_call (state=0x14f058, ir=0x9406910) at ../src/jit/jit-interp.c:638
[... cut ...]
#15 0x00007ff6fb4295f2 in elab_architecture (bind=bind@entry=0x4ff0030, arch=arch@entry=0x950000, config=config@entry=0x0, ctx=ctx@entry=0x14fc10) at ../src/elab.c:1539
#16 0x00007ff6fb428efb in elab (top=top@entry=0x950000, jit=<optimized out>, ur=0x4ea5c0, cover=cover@entry=0x0) at ../src/elab.c:2306
#17 0x00007ff6fb3f3f9b in elaborate (argc=10, argv=0x404eb0, state=0x14fe20) at ../src/nvc.c:464
#18 process_command (argc=10, argv=0x404eb0, state=state@entry=0x14fe20) at ../src/nvc.c:1922
#19 0x00007ff6fb3f17b0 in main (argc=1, argv=0x400000) at ../src/nvc.c:2060
```

nickg, 2024-08-15:
The constructor for the protected type should receive a pointer to its enclosing environment (the instance of the `generic_sb_pkg` package) but that is null here for some reason. What does the package instantiation look like? Is it similar to the following from bitvis_vip_spi?

```vhdl
  package spi_sb_pkg is new bitvis_vip_scoreboard.generic_sb_pkg
    generic map(t_element         => std_logic_vector(C_VVC_CMD_DATA_MAX_LENGTH - 1 downto 0),
                element_match     => std_match,
                to_string_element => to_string);
  use spi_sb_pkg.all;
  shared variable SPI_VVC_SB : spi_sb_pkg.t_generic_sb;
```

avelure, 2024-08-16:
No, that implementation gives a GHDL error (https://github.com/ghdl/ghdl/issues/2510) as both the package declaration and instantiation is in the same package. 
This has been updated in one of the later UVVM tags where there is a new vvc_sb_pkg.
In my custom VVC I made a quick fix which was to move the declaration to the bottom of transaction_pkg.
So transaction_pkg
```
  package ***redacted***_sb_pkg is new bitvis_vip_scoreboard.generic_sb_pkg
       generic map (t_element         => std_logic_vector(C_VVC_CMD_DATA_MAX_LENGTH-1 downto 0),
                    element_match     => std_match,
                    to_string_element => to_string);
```
vvc_methods_pkg
```
shared variable ***redacted***_VVC_SB  : ***redacted***_sb_pkg.t_generic_sb;
```

nickg, 2024-08-17:
Here's a reproducer:

```vhdl
package genpack is
    generic (type t);

    type pt is protected
        impure function get return t;
        impure function get_str return string;
        procedure put (val : t);
    end protected;
end package;

package body genpack is
    constant str : string := "hello world";

    type pt is protected body
        variable cur : t;

        impure function get return t is
        begin
            return cur;
        end function;

        procedure put (val : t) is
        begin
            cur := val;
        end procedure;

        impure function get_str return string is
        begin
            return str;                 -- Accesses parent package
        end function;
    end protected body;
end package body;

-------------------------------------------------------------------------------

package testpack is
    package int_pack is new work.genpack
        generic map ( t => integer );
    use int_pack.all;
end package;

-------------------------------------------------------------------------------

entity issue943 is
end entity;

use work.testpack.all;

architecture test of issue943 is
    shared variable sv : int_pack.pt;
begin

    check: process is
    begin
        sv.put(5);
        assert sv.get = 5;
        assert sv.get_str = "hello world";

        wait;
    end process;

end architecture;
```

nickg, 2024-08-17:
Can you test again with the latest master branch?

avelure, 2024-08-19:
Yes, it works!
Thanks for investigating.

closed 2024-08-19 by nickg