nickg/nvc #946: Crash on elaborate:   Local Package with Generic (constants)

JimLewis, 2024-08-19:
**Description**
Local Generic Package Instance with signal declarations (to reduce the rats nest of signal declarations in structural code).
There are two test cases in the simulation below.  The first uses a local generic package and uses selected names to get at signals in the package - this is the first simulation that works.   The second uses a local generic package and uses selected names to aliases defined in the package (which provide short hand names for the record elements - with appropriate AXI names)

**Expected behaviour**
Successful Simulation - works in Siemens and Aldec

**How to reproduce?**
Get the dev branch of OSVVM
     git clone --recursive --branch dev https://github.com/osvvm/OsvvmLibraries

Using OSVVM scripts do:
build $OsvvmLibraries
build $OsvvmLibraries/AXI4/Axi4/RunDemoTests_GenericSignals

**Error Report**
```
Simulation Start time 07:56:26
simulate TbAxi4_DemoMemoryReadWrite1
nvc --std=08 -H 128m --stderr=error --work=osvvm_TbAxi4_GenericSignals:C:/tools/VHDL_LIBS/NVC-1.13.0
/OSVVM_TBAXI4_GENERICSIGNALS.08 -L C:/tools/VHDL_LIBS/NVC-1.13.0 -e TbAxi4_DemoMemoryReadWrite1
Elaborate Error:
Elaborate Error: Name       OSVVM_TBAXI4_GENERICSIGNALS.TBAXI4MEMORY.AXI4PASSTHRU_1
Elaborate Error: Kind       instance
Elaborate Error: Context    OSVVM_TBAXI4_GENERICSIGNALS.TBAXI4MEMORY
Elaborate Error: Blocks     1
Elaborate Error: Registers  5
Elaborate Error: Types      5
Elaborate Error: Variables  0
Elaborate Error: Begin
Elaborate Error:    0: r0 := package init STD.STANDARD   // P<STD.STANDARD>
Elaborate Error:       r1 := package init OSVVM_AXI4.AXI4COMPONENTPKG // P<OSVVM_AXI4.AXI4COMPONENTP
KG>
Elaborate Error:       r2 := package init OSVVM.CLOCKRESETPKG // P<OSVVM.CLOCKRESETPKG>
Elaborate Error:       r3 := package init IEEE.STD_LOGIC_1164 // P<IEEE.STD_LOGIC_1164>
Elaborate Error:       r4 := package init OSVVM_AXI4.AXI4INTERFACECOMMONPKG // P<OSVVM_AXI4.AXI4INTE
RFACECOMMONPKG>
Elaborate Error:       invalid := var upref 1, invalid   <----
Elaborate Error:
Elaborate Error: ** Fatal: upref -2147483643 is not a variable
Elaborate Error: [00007ff7ec7db76f]
Elaborate Error: [00007ff7ec896aee]
Elaborate Error: [00007ff7ec91a0ec] vhpi_assert+0x31b0c
Elaborate Error: [00007ff7ec881f85]
Elaborate Error: [00007ff7ec864b72]
Elaborate Error: [00007ff7ec831bbc]
Elaborate Error: [00007ff7ec835d8f]
Elaborate Error: [00007ff7ec83856a]
Elaborate Error: [00007ff7ec8340cc]
Elaborate Error: [00007ff7ec7d91f5]
Elaborate Error: [00007ff7ec7d6737]
Elaborate Error: [00007ff7ec919a4c] vhpi_assert+0x3146c
Elaborate Error: [00007ff7ec7d12ee]
Elaborate Error: [00007ff7ec7d1406]
Elaborate Error: [00007ffe02e97374] BaseThreadInitThunk+0x14
Elaborate Error: [00007ffe0309cc91] RtlUserThreadStart+0x21
Elaborate Error:
Elaborate Error: Please report this bug at https://github.com/nickg/nvc/issues
Simulation Finish time 07:56:26, Elapsed time: 0:00:00
```

nickg, 2024-08-20:
The problem here is caused by the aliases:

```vhdl
  alias AwAddr   is AxiBus.WriteAddress.Addr ;
  alias AwProt   is AxiBus.WriteAddress.Prot ;
  ...
```

I'll push a fix shortly.

closed 2024-08-20 by nickg

JimLewis, 2024-08-20:
Thanks.   Installed and passing on my side.

commit 0087f5a7 (2024-08-20): Fix crash with signal alias in generic package. Fixes #946
commit a364158f (2024-08-23): Fix crash with signal alias in generic package. Fixes #946