nickg/nvc #963: Incorrect behavior for functions called in port map with record parameters

egorxe, 2024-08-30:
Hello, it's me with functions in ports again :). This is the original issue I've encountered, but it is rather evasive, so while trying to create a minimal example I've bumped into #945 and decided to report it first, as I thought it was related.

In this example values in record should be passed unchanged through functions and submodule but assert is triggered in NVC. On the other hand If I replace submodule_inst with commented variant where conversion function is called for intermediate vector and not in port, it works as expected. Removing "this_bit_is_important" from rec_type record also masks this issue.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity submodule is
    port (
        dat_i   : in  std_logic_vector(95 downto 0);
        dat_o   : out std_logic_vector(95 downto 0)
    );
end submodule;

architecture behav of submodule is

begin

dat_o <= dat_i;

end behav;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.std_logic_unsigned.all;

use std.env.all;

entity test2 is
end test2;

architecture behav of test2 is
    
    subtype vec32 is std_logic_vector(31 downto 0);
    subtype glue_vec is std_logic_vector(32*3-1 downto 0);
    
    type test_type is record
        x       : vec32;
        y       : vec32;
        z       : vec32;
    end record;
    
    constant ZERO_RECORD  : test_type     := ((others => '0'), (others => '0'), (others => '0'));
    
    function Record2Vec(f : test_type) return glue_vec is
    begin
        return f.z & f.y & f.x;
    end function;

    type rec_type is record
        this_bit_is_important   : std_logic;
        internal_rec            : test_type;
    end record;
    signal r   : rec_type;
    
    signal submodule_out        : glue_vec;
    signal submodule_in         : glue_vec;
    signal clk_i                : std_logic := '0';
    signal rst_i                : std_logic := '1';
begin

submodule_inst : entity work.submodule
    port map (
        dat_i   => Record2Vec(r.internal_rec),
        dat_o   => submodule_out
    );
--submodule_inst : entity work.submodule
    --port map (
        --dat_i   => submodule_in,
        --dat_o   => submodule_out
    --);
--submodule_in <= Record2Vec(r.internal_rec);

clk_i <= not clk_i after 5 ns;
rst_i <= '0' after 50 ns;

process(clk_i)
begin
    if Rising_edge(clk_i) then
        if rst_i = '1' then
            r.internal_rec <= ZERO_RECORD;
        else
            report to_string(Record2Vec(r.internal_rec));
            report to_string(submodule_out);
            assert Record2Vec(r.internal_rec) = submodule_out report "Failure" severity failure;
            finish;
        end if;
    end if;
end process;

end architecture behav;
```

nickg, 2024-08-31:
I think this should be fixed now. Can you test again with the latest master branch?

egorxe, 2024-08-31:
Yes, now it works as expected. Thanks!

closed 2024-08-31 by nickg