nickg/nvc #964: Internal DUT signal aliasing, file parsing/elaboration error

p-kuzmanovic, 2024-09-03:
I experienced an issue with fetching internal DUT signals in my TB:
[image]

It's difficult to share the code, but it seems easy to reproduce the issue.

Just by **moving** the DUT instantiation to be **above** the process that fetches the internal DUT signal the **error disappears**.

It seems like an issue with file parsing and elaboration.

nickg, 2024-09-05:
Could you test again with the latest master branch?

nickg, 2024-09-11:
Should be fixed now.

closed 2024-09-11 by nickg

p-kuzmanovic, 2024-09-18:
Sorry, I haven't been able to test the fix yet. I think I will wait for the next release and give it a try then. Thanks!