nickg/nvc #965: crash during elaboration

nathanaelhuffman, 2024-09-04:
I have not yet narrowed down a reproducer here, but here's the backtrace from a fresh build off main this afternoon. As usual, I'm sure I've got something wrong in my vhdl that's finding this corner.

```
 nvc --version
nvc 1.14-devel (1.13.0.r113.g1b5d85dd) (Using LLVM 14.0.0)
Copyright (C) 2011-2024  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```
Backtrace:
```
Thread 1 "nvc" received signal SIGSEGV, Segmentation fault.
sched_driver (m=0x555557ce9110, nexus=0x7fffe76002b0, after=<optimized out>, reject=0, value=<optimized out>, proc=<optimized out>) at ../src/rt/model.c:2646
2646          if (!insert_transaction(m, nexus, d, w, w->when, reject))
(gdb) bt
#0  sched_driver (m=0x555557ce9110, nexus=0x7fffe76002b0, after=<optimized out>, reject=0, value=<optimized out>, proc=<optimized out>) at ../src/rt/model.c:2646
#1  0x000055555567bb57 in x_sched_waveform_s (reject=0, after=0, scalar=<optimized out>, offset=<optimized out>, ss=0x7fffe7600330) at ../src/rt/model.c:3805
#2  __nvc_sched_waveform (anchor=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-exits.c:333
#3  0x000055555567e5ae in __nvc_do_exit (which=<optimized out>, anchor=0x7fffffffaba0, args=0x7fffffffc750, tlab=<optimized out>) at ../src/jit/jit-exits.c:561
#4  0x00005555556814b7 in interp_exit (ir=0x5555591db280, state=0x7fffffffabc0) at ../src/jit/jit-interp.c:831
#5  interp_loop (state=0x7fffffffabc0) at ../src/jit/jit-interp.c:1025
#6  jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#7  0x0000555555681ce4 in interp_call (ir=0x5555591d08c0, state=0x7fffffffae00) at ../src/jit/jit-interp.c:638
#8  interp_loop (state=0x7fffffffae00) at ../src/jit/jit-interp.c:968
#9  jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#10 0x0000555555681ce4 in interp_call (ir=0x55555950d2b0, state=0x7fffffffaf90) at ../src/jit/jit-interp.c:638
#11 interp_loop (state=0x7fffffffaf90) at ../src/jit/jit-interp.c:968
#12 jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#13 0x0000555555681ce4 in interp_call (ir=0x55555948c610, state=0x7fffffffb4f0) at ../src/jit/jit-interp.c:638
#14 interp_loop (state=0x7fffffffb4f0) at ../src/jit/jit-interp.c:968
#15 jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#16 0x0000555555681ce4 in interp_call (ir=0x55555947e1c8, state=0x7fffffffb670) at ../src/jit/jit-interp.c:638
#17 interp_loop (state=0x7fffffffb670) at ../src/jit/jit-interp.c:968
#18 jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#19 0x0000555555681ce4 in interp_call (ir=0x555559418918, state=0x7fffffffb8e0) at ../src/jit/jit-interp.c:638
#20 interp_loop (state=0x7fffffffb8e0) at ../src/jit/jit-interp.c:968
#21 jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#22 0x0000555555681ce4 in interp_call (ir=0x55555933cce8, state=0x7fffffffbcb0) at ../src/jit/jit-interp.c:638
#23 interp_loop (state=0x7fffffffbcb0) at ../src/jit/jit-interp.c:968
#24 jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#25 0x0000555555681ce4 in interp_call (ir=0x555558f709a0, state=0x7fffffffc610) at ../src/jit/jit-interp.c:638
#26 interp_loop (state=0x7fffffffc610) at ../src/jit/jit-interp.c:968
#27 jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-interp.c:1101
#28 0x000055555566a34c in jit_try_vcall (j=<optimized out>, f=<optimized out>, result=<optimized out>, args=<optimized out>, tlab=<optimized out>)
    at ../src/jit/jit-core.c:666
#29 0x000055555566a62f in jit_fastcall (j=<optimized out>, handle=<optimized out>, result=<optimized out>, p1=..., p2=..., tlab=<optimized out>)
    at ../src/jit/jit-core.c:760
#30 0x00005555556617cc in run_process (proc=0x555557a0cae0, m=0x555557ce9110) at ../src/rt/model.c:1009
#31 async_run_process (m=0x555557ce9110, arg=0x555557a0cae0) at ../src/rt/model.c:2709
#32 0x0000555555662364 in deferq_run (m=0x555557ce9110, dq=0x555557ce9178) at ../src/rt/model.c:390
#33 0x0000555555588859 in model_cycle (m=0x555557ce9110) at ../src/rt/model.c:3296
#34 model_run (stop_time=<optimized out>, m=0x555557ce9110) at ../src/rt/model.c:3343
#35 run_cmd (argc=<optimized out>, argv=<optimized out>, state=<optimized out>) at ../src/nvc.c:843
#36 0x0000555555582d45 in process_command (argc=5, argv=0x7fffffffd348, state=0x7fffffffd090) at ../src/nvc.c:2090
#37 0x0000555555585f3a in elaborate (argc=5, argv=0x7fffffffd348, state=0x7fffffffd090) at ../src/nvc.c:522
#38 0x0000555555582c9a in process_command (argc=10, argv=0x7fffffffd320, state=0x7fffffffd090) at ../src/nvc.c:2088
#39 0x000055555557ec57 in main (argc=<optimized out>, argv=<optimized out>) at ../src/nvc.c:2230
```

nickg, 2024-09-05:
If you add `--trace` to the arguments when running it will print the name of the process before it's executed which should help narrow down the cause.

nathanaelhuffman, 2024-09-05:
I get a bunch of stuff as the designs starts, but it looks like this is happening right after the main vunit testbench comes out of reset which isn't so helpful for narrowing this down.
```
TRACE 200ns+0: run process :espi_tb:bench
TRACE 200ns+0: clear event RESET+0 count=1
TRACE 200ns+0: _sched_waveform_s NET+0 value=3 after=0ms reject=0ms

*** Caught signal 11 (SEGV_MAPERR) [address=0x30, ip=0x56f4f3ea348a] ***
... <traceback elided>
```
I'm having an extremely tough time nailing this down, the simplest change I can make right now that breaks a testbench that otherwise runs is adding this instantiation, which is not even being interacted with in the testbench or really to anything (entity from vunit's vc components):
```
axi_lite_master_inst: entity vunit_lib.axi_lite_master
        generic map (
            bus_handle => my_bus_handle
        )
        port map (
            aclk    => '0',
            arready => '0',
            arvalid => open,
            araddr  => open,
            rready  => open,
            rvalid  => '0',
            rdata   => (others => '0'),
            rresp   => "00",
            awready => '0',
            awvalid => open,
            awaddr  => open,
            wready  => '1',
            wvalid  => open,
            wdata   => open,
            wstrb   => open,
            bvalid  => '0',
            bready  => open

        );
```
But, I don't think there's anything wrong with that entity as I'm using it in other test benches without seeing this issue.

Turning jit off by removing the --jit flag resulted in a similar backtrace, though with a bit more information. I'm making heavy use of vunit's queues and message passing and some of those peices seem to show up below.

My custom vc which implements the "check_status" method works fine until I add the above entity (which also works fine in other tbs).

```
Thread 1 "nvc" received signal SIGSEGV, Segmentation fault.
sched_driver (m=0x555555afeb40, nexus=0x7fffeb6002b0, after=<optimized out>, reject=0, value=<optimized out>, proc=<optimized out>) at ../src/rt/model.c:2646
2646          if (!insert_transaction(m, nexus, d, w, w->when, reject))
(gdb) bt
#0  sched_driver (m=0x555555afeb40, nexus=0x7fffeb6002b0, after=<optimized out>, reject=0, value=<optimized out>, proc=<optimized out>)
    at ../src/rt/model.c:2646
#1  0x000055555567bb57 in x_sched_waveform_s (reject=0, after=0, scalar=<optimized out>, offset=<optimized out>, ss=0x7fffeb600330)
    at ../src/rt/model.c:3805
#2  __nvc_sched_waveform (anchor=<optimized out>, args=<optimized out>, tlab=<optimized out>) at ../src/jit/jit-exits.c:333
#3  0x00007fffec172324 in VUNIT_LIB.EVENT_COMMON_PKG.ACTIVATE(s38VUNIT_LIB.EVENT_COMMON_PKG.ANY_EVENT_T) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#4  0x00007fffec15404d in VUNIT_LIB.EVENT_COMMON_PKG.NOTIFY(s38VUNIT_LIB.EVENT_COMMON_PKG.ANY_EVENT_TN) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#5  0x00007fffec163d62 in VUNIT_LIB.COM_COMMON_PKG.NOTIFY_NET(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#6  0x00007fffec14cd31 in VUNIT_LIB.COM_PKG.SEND(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T31VUNIT_LIB.COM_TYPES_PKG.ACTOR_T36VUNIT_LIB.COM_TYPES_PKG.MAILBOX_ID_T29VUNIT_LIB.COM_TYPES_PKG.MSG_TT) () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#7  0x00007fffec174e31 in VUNIT_LIB.COM_PKG.SEND(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T31VUNIT_LIB.COM_TYPES_PKG.ACTOR_T29VUNIT_LIB.COM_TYPES_PKG.MSG_TT) () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#8  0x00007fffec159ac9 in LIB.QSPI_VC_PKG.ENQUEUE_TX_DATA_BYTES(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T31VUNIT_LIB.COM_TYPES_PKG.ACTOR_TN27VUNIT_LIB.QUEUE_PKG.QUEUE_T) () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#9  0x00007fffec141c93 in LIB.ESPI_CONTROLLER_VC_PKG.GET_STATUS(VVB) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#10 0x00007fffec12fc8f in LIB.ESPI_TB.BENCH () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.54292.so
#11 0x000055555566a34c in jit_try_vcall (j=<optimized out>, f=<optimized out>, result=<optimized out>, args=<optimized out>, tlab=<optimized out>)
    at ../src/jit/jit-core.c:666
#12 0x000055555566a62f in jit_fastcall (j=<optimized out>, handle=<optimized out>, result=<optimized out>, p1=..., p2=..., tlab=<optimized out>)
    at ../src/jit/jit-core.c:760
#13 0x00005555556617cc in run_process (proc=0x555558170d20, m=0x555555afeb40) at ../src/rt/model.c:1009
#14 async_run_process (m=0x555555afeb40, arg=0x555558170d20) at ../src/rt/model.c:2709
#15 0x0000555555662364 in deferq_run (m=0x555555afeb40, dq=0x555555afeba8) at ../src/rt/model.c:390
#16 0x0000555555588859 in model_cycle (m=0x555555afeb40) at ../src/rt/model.c:3296
#17 model_run (stop_time=<optimized out>, m=0x555555afeb40) at ../src/rt/model.c:3343
#18 run_cmd (argc=<optimized out>, argv=<optimized out>, state=<optimized out>) at ../src/nvc.c:843
#19 0x0000555555582d45 in process_command (argc=5, argv=0x7fffffffd520, state=0x7fffffffd270) at ../src/nvc.c:2090
#20 0x0000555555585f3a in elaborate (argc=5, argv=0x7fffffffd520, state=0x7fffffffd270) at ../src/nvc.c:522
#21 0x0000555555582c9a in process_command (argc=9, argv=0x7fffffffd500, state=0x7fffffffd270) at ../src/nvc.c:2088
#22 0x000055555557ec57 in main (argc=<optimized out>, argv=<optimized out>) at ../src/nvc.c:2230
```

I can try to extract this design from our build system and provide it to you if that would help or if any additional info would be helpful, other than continuing to comment things out in search of root cause I don't have much else to go on.

nickg, 2024-09-05:
Another thing you can try is making a debug build with assertions turned on and optimisations off by passing `CFLAGS="-O0 -g" --disable-lto --enable-debug` to `configure`. Hopefully it will trigger one of the debug checks, or at least provide more useful information in GDB.

nathanaelhuffman, 2024-09-05:
Looks like we do trip an assert, does that help?

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe6e00640 (LWP 65236)]
[New Thread 0x7fffe6400640 (LWP 65237)]
[New Thread 0x7fffe5a00640 (LWP 65238)]
[New Thread 0x7fffe5000640 (LWP 65239)]
[New Thread 0x7fffdfe00640 (LWP 65240)]
[New Thread 0x7fffdf400640 (LWP 65241)]
[New Thread 0x7fffdea00640 (LWP 65242)]
[Detaching after fork from child process 65243]
** Debug: JIT code footprint: 24007 bytes
nvc: ../src/rt/model.c:2621: sched_driver: Assertion `d != NULL' failed.

Thread 1 "nvc" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737350810944) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737350810944) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737350810944) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737350810944, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff1042476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff10287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff102871b in __assert_fail_base (fmt=0x7ffff11dd130 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0x55555575cb81 "d != NULL",
    file=0x55555575c1f8 "../src/rt/model.c", line=2621, function=<optimized out>) at ./assert/assert.c:92
#6  0x00007ffff1039e96 in __GI___assert_fail (assertion=0x55555575cb81 "d != NULL", file=0x55555575c1f8 "../src/rt/model.c", line=2621,
    function=0x55555575da10 <__PRETTY_FUNCTION__.42> "sched_driver") at ./assert/assert.c:101
#7  0x000055555568480a in sched_driver (m=0x5555557c7870, nexus=0x7fffeb4002b0, after=0, reject=0, value=0x7fffffffc768, proc=0x5555581f2240)
    at ../src/rt/model.c:2621
#8  0x0000555555688af8 in x_sched_waveform_s (ss=0x7fffeb400330, offset=0, scalar=3, after=0, reject=0) at ../src/rt/model.c:3805
#9  0x00005555556afbc9 in __nvc_sched_waveform (anchor=0x7fffffffc858, args=0x7fffffffcc50, tlab=0x5555580c6630) at ../src/jit/jit-exits.c:333
#10 0x00007fffebf7be04 in VUNIT_LIB.EVENT_COMMON_PKG.ACTIVATE(s38VUNIT_LIB.EVENT_COMMON_PKG.ANY_EVENT_T) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#11 0x00007fffebf5ddfa in VUNIT_LIB.EVENT_COMMON_PKG.NOTIFY(s38VUNIT_LIB.EVENT_COMMON_PKG.ANY_EVENT_TN) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#12 0x00007fffebf6d899 in VUNIT_LIB.COM_COMMON_PKG.NOTIFY_NET(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#13 0x00007fffebf56834 in VUNIT_LIB.COM_PKG.SEND(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T31VUNIT_LIB.COM_TYPES_PKG.ACTOR_T36VUNIT_LIB.COM_TYPES_PKG.MAILBOX_ID_T29VUNIT_LIB.COM_TYPES_PKG.MSG_TT) () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#14 0x00007fffebf7e92d in VUNIT_LIB.COM_PKG.SEND(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T31VUNIT_LIB.COM_TYPES_PKG.ACTOR_T29VUNIT_LIB.COM_TYPES_PKG.MSG_TT) () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#15 0x00007fffebf63688 in LIB.QSPI_VC_PKG.ENQUEUE_TX_DATA_BYTES(s41VUNIT_LIB.EVENT_PRIVATE_PKG.BASIC_EVENT_T31VUNIT_LIB.COM_TYPES_PKG.ACTOR_TN27VUNIT_LIB.QUEUE_PKG.QUEUE_T) () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#16 0x00007fffebf4bb42 in LIB.ESPI_CONTROLLER_VC_PKG.GET_STATUS(VVB) ()
   from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#17 0x00007fffebf39938 in LIB.ESPI_TB.BENCH () from vunit_out/nvc/libraries/lib/_LIB.ESPI_TB-TB.elab.65233.so
#18 0x0000555555692907 in jit_try_vcall (j=0x5555558a8ad0, f=0x5555581f2340, result=0x7fffffffcfb0, args=0x7fffffffcc50, tlab=0x5555580c6630)
    at ../src/jit/jit-core.c:666
#19 0x00005555556930c6 in jit_fastcall (j=0x5555558a8ad0, handle=1056, result=0x7fffffffcfb0, p1=..., p2=..., tlab=0x5555580c6630)
    at ../src/jit/jit-core.c:760
#20 0x000055555567ea50 in run_process (m=0x5555557c7870, proc=0x5555581f2240) at ../src/rt/model.c:1009
#21 0x0000555555684cb1 in async_run_process (m=0x5555557c7870, arg=0x5555581f2240) at ../src/rt/model.c:2709
#22 0x000055555567c5e6 in deferq_run (m=0x5555557c7870, dq=0x5555557c78d8) at ../src/rt/model.c:390
#23 0x0000555555686ed5 in model_cycle (m=0x5555557c7870) at ../src/rt/model.c:3296
#24 0x000055555568706b in model_run (m=0x5555557c7870, stop_time=9223372036854775807) at ../src/rt/model.c:3343
#25 0x000055555556b134 in run_cmd (argc=5, argv=0x7fffffffd530, state=0x7fffffffd390) at ../src/nvc.c:843
#26 0x000055555556de09 in process_command (argc=5, argv=0x7fffffffd530, state=0x7fffffffd390) at ../src/nvc.c:2090
#27 0x000055555556a36b in elaborate (argc=5, argv=0x7fffffffd530, state=0x7fffffffd390) at ../src/nvc.c:522
#28 0x000055555556ddef in process_command (argc=9, argv=0x7fffffffd510, state=0x7fffffffd390) at ../src/nvc.c:2088
#29 0x000055555556e2b8 in main (argc=9, argv=0x7fffffffd510) at ../src/nvc.c:2230
```

nathanaelhuffman, 2024-09-06:
nvc is in good company here though, as I moved this design over into edaplayground to attempt to get more debug info and it compiles fine but also crashes rivieraPro during execution :/

I'm clearly doing something VeryBad<sup>TM</sup> with this test bench and am still happy to share the design or continue debugging in any way that is helpful.

nickg, 2024-09-06:
Can you see what the first argument to `lib.qspi_vc_pkg.enqueue_tx_data_bytes` is? It should be a signal. I'm wondering if it's a signal declared in that package and so the `espi_tb.bench` process wouldn't have created a driver for it. That should be caught at analysis time though...

nathanaelhuffman, 2024-09-06:
ha yes, I'm just sorting this out now. this was a warning in riviera pro and once fixing that, nvc runs (though riviera still crashes for presumably unrelated reasons at this point).

so the first argument to that procedure is an inout signal (vunit's "net") which is declared in a package. This procedure is wrapped by another procedure that does not have the net signal as a parameter.

I have a number of these wrapper procedures, the simplest looks like this in the broken case (net is not passed in as a signal on the wrapper procedure)
```
 procedure get_status(
        --signal net : inout network_t;  -- Adding this line back in and plumbing it for each of the wrappers prevents the crash
        variable response_code : inout std_logic_vector(7 downto 0);
        variable status : inout std_logic_vector(15 downto 0);
        variable crc_ok : inout boolean;
        ) is

        variable tx_bytes : integer   := 2;
        variable rx_bytes : integer   := 4;
        variable msg_target : actor_t := find("espi_vc");
        variable rx_queue : queue_t := new_queue;

    begin
        -- Build and send a GET STATUS message
        -- OPCODE_GET_STATUS  (1 byte)
        push_byte(tx_queue, to_integer(OPCODE_GET_STATUS));
        -- CRC (1 byte)
        push_byte(tx_queue, to_integer(crc8(tx_queue)));
        -- Turn around
        -- RESPONSE (no append) (1 byte)
        -- STATUS (2 bytes)
        -- CRC (1 byte)
        enqueue_tx_data_bytes(net, msg_target,  tx_bytes, tx_queue);
        enqueue_transaction(net, msg_target, tx_bytes, rx_bytes);
        get_rx_queue(net, msg_target, rx_queue);
        crc_ok := check_queue_crc(rx_queue); -- non-destructive to queue
        response_code := std_logic_vector(to_unsigned(pop_byte(rx_queue), 8));
        status := get_status_from_queue_and_flush(rx_queue);

    end;
```
So basically my wrapper procedures were grabbing the signal from the package but it wasn't in the wrapper procedure's  interface. Possibly adding the 2nd entity which also can send and rx on the global "net" signal is what trips things up here. 

Rivera errored on compile with something to this effect: "The signal used as the actual parameter in a subprogram call inside subprogram <wrapper_procedure> is not a formal parameter of this subprogram."

nathanaelhuffman, 2024-09-06:
So I'm happy to go and plumb net through this stuff since that seems the the correct thing to do regardless, and happy to help further if you still need a reproducer to make this error more easily caught. Thanks much for the help here and as always the fast responses to things. I really love working with nvc!

nickg, 2024-09-06:
Here's a minimal reproducer:

```vhdl
package pack is
    procedure do_something;
    signal s : bit;
end package;

package body pack is
    procedure drive_signal (signal p : out bit) is
    begin
        p <= '1';
    end procedure;

    procedure do_something is
    begin
        --s <= '0';                     -- This errors...
        drive_signal(s);                -- ... but not this
    end procedure;
end package body;

-------------------------------------------------------------------------------

entity test is
end entity;

use work.pack.all;

architecture test of test is
begin
    check: process is
    begin
        do_something;
        wait;
    end process;
end architecture;
```

The call to `drive_signal(s)` should produce an error at analysis time like Rivera.

nickg, 2024-09-06:
Can you try again with the latest master branch? It should generate an error now. You probably want to remove those debug options from configure too.

nathanaelhuffman, 2024-09-06:
yep works as expected!

```
** Error: signal NET is not a formal parameter and subprogram GET_STATUS [STD_LOGIC_VECTOR, STD_LOGIC_VECTOR, BOOLEAN] is not contained within a process statement
     > /home/nhuffman/oxide/quartz/hdl/ip/vhd/espi/sims/models/espi_controller_vc_pkg.vhd:162
     |
 162 |         enqueue_tx_data_bytes(net, msg_target,  tx_bytes, tx_queue);
     |                               ^^^
     |
     = Help: IEEE Std 1076-2008 section 10.5.2.2 "Executing a simple assignment statement"
```

nickg, 2024-09-06:
Great, thanks for confirming.

closed 2024-09-06 by nickg