nickg/nvc #969: Stack Overflow (or similar) when mapping port to function

Schottkyc137, 2024-09-09:
MRE:
```vhdl
entity mre_conversion is
  port (din : in bit);
end entity mre_conversion;

entity mre_conversion_tb is
end entity mre_conversion_tb;

architecture arch of mre_conversion is
  function f1 return bit is
  begin
    return '0';
  end f1;
begin
  
  mre_conversion_inst: entity work.mre_conversion
  port map (din => f1);
  
end architecture arch;
```
Crash report:
```
$ nvc -a mre_conversion.vhd -e mre_conversion

*** Caught signal 11 (SEGV_MAPERR) [address=0x70, ip=0x61ec423fb444] ***

[0x61ec4235a5dd] ../src/util.c:872 signal_handler.lto_priv.0
[0x753e7fa4251f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x61ec423fb444] ../src/vcode.c:346 vtype_eq
[0x61ec424882b4] ../src/vcode.c:2388 vtype_new.constprop.0
[0x61ec42400c8e] ../src/vcode.c:2585 vtype_context
[0x61ec42400c8e] ../src/vcode.c:11156 lower_func_body.lto_priv.0
[0x61ec423fabaf] ../src/lower.c:13389 unit_registry_get
[0x61ec423dcdfc] ../src/lower.c:2432 lower_context_for_call.lto_priv.0
[0x61ec423e1275] ../src/lower.c:2525 lower_fcall.lto_priv.0
[0x61ec423e4119] ../src/lower.c:5123 lower_expr.lto_priv.0
[0x61ec423fef4d] ../src/lower.c:12835 lower_rvalue
[0x61ec423b2a06] ../src/lower.c:12402 lower_instance
[0x61ec423b2a06] ../src/lower.c:1415 elab_lower.lto_priv.0
[0x61ec423b4f01] ../src/elab.c:1532 elab_architecture.lto_priv.0
[0x61ec423bbeb9] ../src/elab.c:1699 elab_stmts.lto_priv.0
[0x61ec423b4f17] ../src/elab.c:1534 elab_architecture.lto_priv.0
[0x61ec423bbeb9] ../src/elab.c:1699 elab_stmts.lto_priv.0
[0x61ec423b4f17] ../src/elab.c:1534 elab_architecture.lto_priv.0
[0x61ec423bbeb9] ../src/elab.c:1699 elab_stmts.lto_priv.0
[0x61ec423b4f17] ../src/elab.c:1534 elab_architecture.lto_priv.0
[0x61ec423bbeb9] ../src/elab.c:1699 elab_stmts.lto_priv.0
[0x61ec423b4f17] ../src/elab.c:1534 elab_architecture.lto_priv.0
[0x61ec423bbeb9] ../src/elab.c:1699 elab_stmts.lto_priv.0
[0x61ec423b4f17] ../src/elab.c:1534 elab_architecture.lto_priv.0
# ... statements repeat
[0x61ec423bbeb9] ../src/elab.c:1699 elab_stmts.lto_priv.0
[0x61ec423b4f17] ../src/elab.c:1534 elab_architecture.lto_priv.0
[0x61ec42354fa3] ../src/elab.c:2299 elab
[0x61ec42354fa3] ../src/elab.c:466 elaborate
[0x61ec42351dc9] ../src/nvc.c:2092 process_command
[0x61ec42351678] ../src/nvc.c:1776 cover_report_cmd
[0x61ec42351678] ../src/nvc.c:2122 process_command
[0x61ec4234dd86] ../src/nvc.c:2234 main
```

nickg, 2024-09-09:
The problem here is caused by the infinite recursion in the entity instantiation rather than the function call. But it shouldn't crash in any case.

Schottkyc137, 2024-09-10:
> The problem here is caused by the infinite recursion in the entity instantiation rather than the function call. But it shouldn't crash in any case.

Ah I knew the error looked different :D I posted the original issue with ports and functions in #971

commit ce5606a4 (2024-09-11): Limit maximum instantiation depth. Fixes #969
closed 2024-09-11 by nickg