nickg/nvc #970: VHDL-2019: Map generics on Subprograms

bpadalino, 2024-09-09:
I was writing an example generic method to reverse any vector.  The code works well in Riviera PRO 24.04, but nvc fails to parse it.  I understand the feature isn't implemented but figured this is at least a decent test to pass.

```vhdl
library ieee ;
    use ieee.std_logic_1164.all ;

entity test is
end entity ;

architecture arch of test is

    -- note: can't get this to parse with rivieria
    --function reverse parameter(x : array(type is (<>)) of type is private) return x'subtype;

    -- naive generic reverse algorithm
    function reverse generic (
        type t is array(type is (<>)) of type is private
    ) parameter(x : t) return t is
        subtype idx is x'index ;
        variable rv     : x'subtype := x ;
        variable inidx  : idx := x'right ;
    begin
        -- null or single element arrays are the same as the original
        if x'length < 2 then
            return rv ;
        end if ;

        -- fill in right -> left
        for outidx in x'range loop
            rv(inidx) := x(outidx) ;

            -- only increment if we aren't done
            -- note: /= not passed in via generics
            if inidx /= x'left then
                if x'ascending = true then
                    inidx := idx'pred(inidx) ;
                else
                    inidx := idx'succ(inidx) ;
                end if ;
            end if ;
        end loop ;

        return rv ;
    end function ;

    procedure display generic (
        type t is array(type is (<>)) of type is private ;
        function to_string(x : t'element) return string is <>
    ) parameter (x : t) is
        use std.textio.all ;
        variable l : line ;
    begin
        swrite(l, "(") ;
        for idx in x'range loop
            write(l, to_string(idx) & "=>" & to_string(x(idx))) ;
            if idx /= x'high then
                swrite(l, ", ") ;
            end if ;
        end loop ;
        swrite(l, ")") ;
        writeline(output, l) ;
    end procedure ;

begin

    tb : process
        variable x : integer_vector := (1, 2, 3, 4, 5, 6, 7, 8) ;
        variable y : integer_vector := reverse generic map(x'subtype)(x) ;

        variable a : std_logic_vector(15 to 20) := ('X', '-', 'U', '0', '1', 'L') ;
        variable b : a'subtype := reverse generic map(a'subtype)(a) ;

        variable c : std_logic_vector(9 to 9) := (9 => '1') ;
        variable d : c'subtype := reverse generic map(c'subtype)(c) ;

        variable n : std_logic_vector(1 to 0) ;
        variable m : n'subtype := reverse generic map(n'subtype)(n) ;

        variable p : integer_vector(4 to 5) := (1, 2) ;
        variable q : p'subtype := reverse generic map(p'subtype)(p) ;
    begin
        display generic map (y'subtype)(y) ;
        display generic map (b'subtype)(b) ;
        display generic map (d'subtype)(d) ;
        display generic map (m'subtype)(m) ;
        display generic map (q'subtype)(q) ;
        std.env.stop ;
    end process ;

end architecture ;
```

nickg, 2024-09-19:
I added some parsing support for this but the example above doesn't work yet.

nickg, 2024-09-20:
This test case is working now:

```
(0=>8, 1=>7, 2=>6, 3=>5, 4=>4, 5=>3, 6=>2, 7=>1)
(15=>L, 16=>1, 17=>0, 18=>U, 19=>-, 20=>X)
(9=>1)
()
(4=>2, 5=>1)
```

> `-- note: /= not passed in via generics`

This works because

```vhdl
generic (
        type t is array(type is (<>)) of type is private
    )
```

is equivalent to

```vhdl
generic (
        type anonymous1 is (<>);
        -- Predefined ops for anonymous1 here
        type anonymous2 is private;
        -- Predefined ops for anonymous2 here   
        type t is array(anonymous1) of anonymous2
        -- Predefined ops for t here
    )
```

So the `"="` and `"/="` operators are implicitly passed as generics for the two anonymous types as well as `t`.

closed 2024-09-20 by nickg

bpadalino, 2024-09-20:
You are absolutely spectacular.  Thank you very much!

commit 8e9b3978 (2024-09-20): Various fixes for VHDL-2019 array type generics. Fixes #970