nickg/nvc #974: Initial Value not applied

Anselmo95, 2024-09-13:
Hello, 

The following code simulates a differently on nvc then questa,vcs.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library xpm;
use xpm.vcomponents.all;

entity ent is
end entity ent;

architecture arch of ent is
    signal asynch_in_reset_n : std_logic := '0';
    signal clk : std_logic := '0';
    signal synch_out_reset_n : std_logic;
begin

    clk <= not clk after 2 ns;
    asynch_in_reset_n <= '1', '0' after 36 ns;

    xpm_cdc_async_rst_inst : xpm_cdc_async_rst
    generic map (
        DEST_SYNC_FF    => 4,
        RST_ACTIVE_HIGH => 0
    )
    port map (
        src_arst  => asynch_in_reset_n,
        dest_clk  => clk,
        dest_arst => synch_out_reset_n
    );

end architecture arch;
```

The [cdc](https://github.com/fransschreuder/xpm_vhdl/blob/master/src/xpm/xpm_cdc/hdl/xpm_cdc_async_rst.vhd) component uses the generic RST_ACTIVE_HIGH to initialize  the value of an internal signal
```vhdl
constant DEF_VALS : std_logic_vector(1 downto 0) := "01";
constant DEF_VAL : std_logic := DEF_VALS(RST_ACTIVE_HIGH);
signal arststages_ff : std_logic_vector(DEST_SYNC_FF-1 downto 0) := (others => DEF_VAL); 
```
I would expect `arststages_ff` to start at value 0xf but in the simulation it is at 0.

Commands to reproduce:
```bash
nvc --std=08 --work=work -a ent.vhd && nvc --std=08 --work=work -e ent && nvc --std=08 --work=work -r ent --format=fst --wave=wave.fst --stop-time=50ns
```
The [waves](https://github.com/user-attachments/files/16990463/wave.fst.gz) in case you need them.

nickg, 2024-09-15:
It looks like it does get initialised to 0xf but is then assigned 0 during the first time step by the process here:

```vhdl
process(dest_clk, reset_pol)
  begin
    if reset_pol = '1' then
        arststages_ff <= (others => INV_DEF_VAL);     -- This branch
    elsif rising_edge(dest_clk) then
        arststages_ff <= arststages_ff(DEST_SYNC_FF-2 downto 0) & async_path_bit;
    end if;
  end process;
```

The FST dumper only records the value after the final delta cycle in a timestep so you don't see the transition unlike Questa which shows a transition before it becomes zero.

[image]

Anselmo95, 2024-09-16:
Hello @nickg,

Thanks a lot for the answer it helped me sort out what was wrong.
You are right there are no mistakes in the simulation, I completely missed that rest_pol is in the sensitivity list and would affect the value of arststages_ff immediately. I don't know why the transaction is missing on my end for questa though.

In any case the problem here is that the VHDL version of this module doesn't match the verilog counterpart and that's why I thought it was a problem with the initial values.

I am going to open a PR to fix the logic in the XPM repo.

Thanks again and cheers

closed 2024-09-16 by Anselmo95