nickg/nvc #978: fatal error `value 9223372036854775807is too large for vhpiIntT` with VHPIDIRECT project

NikLeberg, 2024-09-18:
Hi there @nickg 

First of all, this project fo yours is amazing! Really like how fast and easy everything is.

I'm playing around with VHPIDIRECT to implement a virtual JTAG _connector_ during simulation. For GHDL and ModelSim/QuestaSim I've got VHPI(ish) to work. The same implementation as for GHDL does crash NVC though.

MWE:

_file: issue.vhd_
```vhd
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY cosim_jtag IS
    PORT (
        clk : IN STD_ULOGIC
    );
END ENTITY;

ARCHITECTURE sim OF cosim_jtag IS
    PROCEDURE tick IS
    BEGIN
        REPORT "ERROR: foreign subprogram cosim_jtag_tick not called" SEVERITY failure;
    END;
    ATTRIBUTE foreign OF tick : PROCEDURE IS "VHPIDIRECT cosim_jtag_tick";
BEGIN
    jtag_tick : PROCESS (clk)
    BEGIN
        IF rising_edge(clk) THEN
            tick;
        END IF;
    END PROCESS jtag_tick;
END ARCHITECTURE;

LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY tb IS
END ENTITY;

ARCHITECTURE sim OF tb IS
    CONSTANT CLK_PERIOD : DELAY_LENGTH := 20 ns; -- 50 MHz
    SIGNAL clk : STD_LOGIC := '1';
BEGIN
    clk <= NOT clk AFTER 0.5 * CLK_PERIOD;

    cosim_jtag_inst : ENTITY work.cosim_jtag
        PORT MAP(
            clk  => clk
        );
END ARCHITECTURE;
```

_file: issue.c
```c
void issue_tick(void) {
  // nothing
}
```

```bash
$ nvc -a ../issue.vhd
$ nvc -e tb
$ nvc -r tb
** Fatal: 20ns+0: foreign function cosim_jtag_tick not found
   Subprogram TICK [] at ../issue.vhd:15
   Process :tb:cosim_jtag_inst:jtag_tick at ../issue.vhd:20
root@4d81334b93ae:/work/test# nvc -r --load issue.so tb 
** Note: loading VHPI plugin issue.so
** Fatal: issue.so: cannot open shared object file: No such file or directory
root@4d81334b93ae:/work/test# nvc -r --load ./issue.so tb 
** Note: loading VHPI plugin ./issue.so
** Fatal: (init): value 9223372036854775807is too large for vhpiIntT
[0x4d1096] ../src/diag.c:1048 diag_femit
[0x4109c0] ../src/util.c:585 fatal_trace
[0x53365b] ../src/vhpi/vhpi-util.c:364 vhpi_int_from_native
[0x5324c3] ../src/vhpi/vhpi-model.c:3604 build_int_range
[0x531d8c] ../src/vhpi/vhpi-model.c:3912 build_typeDecl
[0x530f66] ../src/vhpi/vhpi-model.c:3939 cached_typeDecl
[0x530f66] ../src/vhpi/vhpi-model.c:1011 init_objDecl
[0x530e68] ../src/vhpi/vhpi-model.c:4051 vhpi_build_decls
[0x52f77a] ../src/vhpi/vhpi-model.c:4364 vhpi_build_design_model
[0x52f77a] ../src/vhpi/vhpi-model.c:4375 vhpi_initialise_cb
[0x4e657c] ../src/rt/model.c:431 global_event
[0x4e657c] ../src/rt/model.c:2435 model_reset
[0x40d992] ../src/nvc.c:838 run_cmd
[0x40d992] ../src/nvc.c:2085 process_command
[0x40b1d5] ../src/nvc.c:2225 main

Please report this bug at https://github.com/nickg/nvc/issues
```

The version of NVC I used is a bit behind. But I dont seem to find anyhing VHPI related that did change lately?

```bash
$ nvc --version
nvc 1.14-devel (1.13.0.r46.g059d264f) (Using LLVM 11.0.1)
$ uname -a
Linux 4d81334b93ae 5.15.153.1-microsoft-standard-WSL2 #1 SMP Fri Mar 29 23:14:13 UTC 2024 x86_64 GNU/Linux
```

Thanks!

NikLeberg, 2024-09-19:
I was able to simplify the MWE:

```vhd
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY issue978 IS
END ENTITY;

ARCHITECTURE arch OF issue978 IS
    CONSTANT CLK_PERIOD : DELAY_LENGTH := 20 ns; -- comment me out and it works?

    PROCEDURE tick (
        dummy : natural
    ) IS
    BEGIN
        REPORT "ERROR: foreign subprogram cosim_jtag_tick not called" SEVERITY failure;
    END;
    ATTRIBUTE foreign OF tick : PROCEDURE IS "VHPIDIRECT issue_tick";
BEGIN
    proc: PROCESS IS
    BEGIN
        tick(2);
        WAIT;
    END PROCESS proc;
END ARCHITECTURE;
```
and C-file:
```c
#include <stdio.h>

void issue_tick(int dummy) {
  fprintf(stderr, "issue_tick called!\n");
}
```

I was trying to debug it with gdb but got lost in the sea of `tree_t` and `item_t`. I think though, that during elaboration an invalid value of 9223372036854775807 (or `0x7FFFFFFFFFFFFFFF`) is saved from somewhere into the _work_ files. When it then tries to restore that to an _int_range_ it fails.

Very interestingly, if the line containing `CONSTANT CLK_PERIOD : DELAY_LENGTH` is commented out, then everything works. So is it something in the `ieee.std_logic_1164` that is tripping it up?

nickg, 2024-09-19:
> Very interestingly, if the line containing CONSTANT CLK_PERIOD : DELAY_LENGTH is commented out, then everything works. 

Yes the problem only occurs with subtypes of physical types.

> For GHDL and ModelSim/QuestaSim I've got VHPI(ish) to work.

Out of curiosity are you using the "standard" VHPI interface with Questa (i.e. the C function takes a single `const vhpiCbDataT *` argument) or their proprietary FLI?

closed 2024-09-19 by nickg

NikLeberg, 2024-09-20:
Thanks, my project is working now! And super fast.. what takes Questa 9min and GHDL 6min your NVC does in under 2 mins! Amazing! :rocket:

> Out of curiosity are you using the "standard" VHPI interface with Questa (i.e. the C function takes a single `const vhpiCbDataT *` argument) or their proprietary FLI?

No, sadly Questa does support way too little of VHPI to be even remotely useful (e.g. no `vhpi_register_foreignf`). I had to resort to their proprietary FLI. But with how NVC and GHDL implement VHPIDIRECT it is an 1:1 match for functions or procedures (at least for simple enumeration types like `std_ulogic` or integers). The only minor_difference_ is how the `foreign`attribute has to be specified:

```vhd
-- GHDL specific way of declaring foreign VHPIDIRECT C-function:
--  -> "VHPIDIRECT <shared_library> <c_function>"
ATTRIBUTE foreign OF tick : PROCEDURE IS "VHPIDIRECT ./cosim_jtag.so cosim_jtag_tick";
-- ModelSim/QuestaSim specific way of declaring foreign MTI FLI C-function:
--  -> "<c_function> <shared_library>"
ATTRIBUTE foreign OF tick : PROCEDURE IS "cosim_jtag_tick ./cosim_jtag.so";
```
Feel free to have a look at my project [NikLeberg/cosim_jtag](https://github.com/nikleberg/cosim_jtag).

Thanks again for fixing.

nickg, 2024-09-20:
NVC also supports `vhpi_register_foreignf` but only with the indirect binding mode at the moment (i.e. "VHPI lib func").

See this comment: https://github.com/nickg/nvc/issues/976#issuecomment-2361931354

NikLeberg, 2024-09-21:
I had a go at implementing my interface via VHPI... but somehow I can't get the out variable(s) of my procedure to update. MAy I borrow your understanding of VHPI? Can you glance over the below code and maybe spot what I do wrong?

file: test.vhd:
```vhd
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY test IS
END ENTITY;

ARCHITECTURE sim OF test IS
    PROCEDURE tick (var : OUT STD_ULOGIC) IS
    BEGIN
        REPORT "ERROR: foreign subprogram tick not called" SEVERITY failure;
    END;
    ATTRIBUTE foreign OF tick : PROCEDURE IS "VHPI test.so tick_exec";
BEGIN
    PROCESS IS
        VARIABLE var_out : STD_ULOGIC := 'X';
    BEGIN
        report " pre-tick: " & STD_ULOGIC'image(var_out);
        tick(var_out);
        report "post-tick: " & STD_ULOGIC'image(var_out);
        wait for 1 ns;
        report "post-wait: " & STD_ULOGIC'image(var_out);
        wait;
    END PROCESS;
END ARCHITECTURE;
```

file: test.c
```c
#include <vhpi_user.h>

static void exec_vhpi(const struct vhpiCbDataS *cb_data)
{
    vhpiHandleT iter, var_param;
    iter = vhpi_iterator(vhpiParamDecls, cb_data->obj);
    var_param = vhpi_scan(iter); // -> first out argument "var"
    vhpi_release_handle(iter);

    vhpiValueT value;
    value.format = vhpiLogicVal;
    value.value.enumv = vhpi1;
    vhpi_put_value(var_param, &value, vhpiDepositPropagate);
    vhpi_release_handle(var_param);
}

static void register_vhpi(const struct vhpiCbDataS *cb_data)
{
    vhpiForeignDataT foreign_data = {
        vhpiProcF,
        "test.so",   // must precisely match VHDL "foreign" attribute
        "tick_exec", // must precisely match VHDL "foreign" attribute
        NULL,
        exec_vhpi};
    vhpiHandleT cb_h = vhpi_register_foreignf(&foreign_data);
    vhpi_release_handle(cb_h);
}

void (*vhpi_startup_routines[])() = {
    register_vhpi,
    NULL};
```

```shell
gcc -shared -fPIC -o test.so test.c
nvc -a test.vhd
nvc -e test
nvc -r --load ./test.so --vhpi-debug --vhpi-trace test
** Note: loading VHPI plugin ./test.so
VHPI: vhpi_register_foreignf kind=3 libraryName=test.so modelName=tick_exec
VHPI: vhpi_release_handle handle=0x100000000:{vhpiForeignfK}
VHPI: vhpi_build_design_model building model for TEST took 0 ms
** Note: 0ms+0:  pre-tick: 'X'
   Process :test:_p0 at ../test.vhd:18
VHPI: vhpi_iterator type=vhpiParamDecls handle=0x100000001:{vhpiProcDeclK}
VHPI: vhpi_scan handle=0x100000002:{vhpiIteratorK pos=0/1}
VHPI: vhpi_release_handle handle=0x100000002:{vhpiIteratorK pos=1/1}
VHPI: vhpi_put_value handle=0x100000003:{vhpiVarParamDeclK} value_p=0x7ffd90234750 mode=vhpiDepositPropagate
VHPI: vhpi_release_handle handle=0x100000003:{vhpiVarParamDeclK}
** Note: 0ms+0: post-tick: 'X'
   Process :test:_p0 at ../test.vhd:20
** Note: 1ns+0: post-wait: 'X'
   Process :test:_p0 at ../test.vhd:22
```

I expect the value to be updated to `'1'`, but its not happening. Thanks...

nickg, 2024-09-21:
> I had a go at implementing my interface via VHPI... but somehow I can't get the out variable(s) of my procedure to update. MAy I borrow your understanding of VHPI? Can you glance over the below code and maybe spot what I do wrong?

This is a bug. I guess no one used this interface much. I fixed it in 894865a.

commit 9c210bbd (2024-09-19): Handle physical subtypes in VHPI. Fixes #978