nickg/nvc #982: error `no matching operator` when in parentheses

NikLeberg, 2024-09-19:
The following MRE fails:
```vhd
entity ent is
end ent;

architecture arch of ent is
    constant C : natural := 8;
begin
    assert ((C mod 2) /= 0); -- error
    assert (C mod 2 /= 0); -- ok
end arch;
```

with the following error message:
```bash
$ nvc -a ent.vhd
** Error: no matching operator "mod" [NATURAL, universal_integer return BOOLEAN | BIT]
   > mod.vhd:7
   |
 7 |     assert ((C mod 2) /= 0); -- error
   |              ^^^^^^^
   |
   = Note: no implicit conversion was performed on the second argument as the innermost complete context did not determine a unique numeric type
   = Help: IEEE Std 1076-2008 section 9.3.6 "Type conversions"
```

Note: The first assert has additional enclosing parentheses `(C mod 2)` and fails analyze. The second assert which has no enclosing brackets, does not.

Thanks <3

nickg, 2024-09-19:
I think this is the same as #956 which was a regression in the latest 1.13.3 release. It's fixed on the master branch and I'm going to make a new release in the next week or so.

NikLeberg, 2024-09-20:
Can confirm. With current master `nvc 1.14-devel (89e8d45)` the issue is fixed.

closed 2024-09-20 by NikLeberg