nickg/nvc #985: ** Fatal: expression cannot be folded to an integer constant when running with --dump-arrays and --wave flags

augustofg, 2024-09-19:
Stumbled into this problem when simulating a Xilinx primitive (FIFO18E1). I've reduced to a minimal reproducible example:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity folded_bug is
  generic (
    DATA_WIDTH: integer := 4;
    FIFO_SIZE: integer := 36
  );
end entity;

architecture sim of folded_bug is
  function GetWidth (
    rdwr_width : in integer
    ) return integer is
    variable func_width : integer;
  begin
    return rdwr_width;
  end;
  constant mem_width : integer := GetWidth(DATA_WIDTH);
  type Two_D_array_type is array (1 downto 0) of std_logic_vector((mem_width - 1) downto 0);
  signal mem : Two_D_array_type;
begin
  process
  begin
    mem(0) <= (others => '1');
    wait for 1 ns;
    mem(0) <= (others => '0');
    wait for 1 ns;
    std.env.finish;
  end process;
end architecture;
```
```shell
$ nvc --work=work --std=2008 -a folded_bug.vhd
$ nvc --std=2008 -e --no-collapse folded_bug
$ nvc -r --dump-arrays --exit-severity=error folded_bug --wave=folded_bug.fst --format=fst
** Fatal: expression cannot be folded to an integer constant
    > /tmp/nvc_folded_bug/folded_bug.vhd:20
    |
 20 |   type Two_D_array_type is array (1 downto 0) of std_logic_vector((mem_width - 1) downto 0);
    |                                                                    ^^^^^^^^^^^^^
```
[nvc_folded_bug.zip](https://github.com/user-attachments/files/17064012/nvc_folded_bug.zip)

nickg, 2024-09-19:
Fixed in d0f7b34113

closed 2024-09-19 by nickg

augustofg, 2024-09-20:
Nice!