nickg/nvc #986: VHPI calls to `vhpi_prinf` make nvc unresponsive

NikLeberg, 2024-09-22:
If a VHPI function/procedure calls `vhpi_printf` very frequent (e.g. on every clock edge or simulator timestep), then nvc becomes very unresponsive and hard to exit with _Ctrl+C_. Even if killed with `pkill nvc`, the _kill_ is delayed by many seconds.

On the other hand, if nvc itself is generating lots of output (as e.g. with flag `--vhpi-trace`), then it stays perfectly responsive to _Ctrl+C_.

This is simply _nice-to-have_. I have no real requirement for this to work soothly. Putting so many printf calls is rarely ideal anyway, but for crude debugging it could help a lot.

nickg, 2024-09-22:
Which OS is this on? I wasn't able to reproduce this on Linux. The signal handler sets a flag that should be checked at least every simulation cycle.

NikLeberg, 2024-09-22:
I'm running inside a ubuntu (jammy) docker container under WSL2 on a Windows 10 host.

NikLeberg, 2024-09-22:
Tested it a bit more, it is not confined to VHPI. Any `report` from VHDL code (that is frequently executed) is behaving _unresponsive_ as described.

NikLeberg, 2024-09-22:
The following MWE
```vhd
ENTITY responsive IS
END ENTITY;

ARCHITECTURE test OF responsive IS
BEGIN
    PROCESS 
    BEGIN
        WHILE true LOOP
            report "loop";
        END LOOP;
    END PROCESS;
END ARCHITECTURE;
```
takes ~15 seconds on my machine from typing _Ctrl+C_ until is stops.

nickg, 2024-09-22:
For me it's pretty much instant. Can you try `nvc -r responsive 2>/dev/null` and see if that has any effect on how long it takes to stop?

NikLeberg, 2024-09-22:
Hmm.. that is instantaneous... 
May be that the vast layers of virtualization slow down the output quite a bit.
Ok it seems to be an [issue with WSL](https://github.com/microsoft/WSL/issues/2772) then..

Sorry for the noise. 😅

nickg, 2024-09-22:
OK, might be worth trying in a different terminal too (e.g. the newer Windows Terminal).

closed 2024-09-22 by nickg