nickg/nvc #987: In nvc r1.13.1 and later, it is an error to define a constant in the same package declaration using a function declared in the package declaration

ikwzm, 2024-09-24:
As in test_ng.vhd below, constants are defined using functions declared in the same package declaration.

```VHDL:test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
package TEST_NG is
    type      PARAM_TYPE is record
                  SIZE_X  :  integer;
                  SIZE_Y  :  integer;
    end record;
    function  NEW_PARAM(X_SIZE   ,Y_SIZE   :integer) return PARAM_TYPE;
    function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE;
    constant  PARAM_1x1  :  PARAM_TYPE := NEW_PARAM(1,1);
    constant  PARAM_3x3  :  PARAM_TYPE := NEW_PARAM(-1,1,-1,1);
end TEST_NG;
package body TEST_NG is
    function  NEW_PARAM(X_SIZE   ,Y_SIZE   :integer) return PARAM_TYPE is
        variable  param  :  PARAM_TYPE;
    begin
        param.SIZE_X := X_SIZE;
        param.SIZE_Y := Y_SIZE;
        return param;
    end function;

    function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE is
        variable  param  :  PARAM_TYPE;
    begin
        param.SIZE_X := Y_HI-X_LO;
        param.SIZE_Y := X_HI-Y_LO;
        return param;
    end function;
end package body;
```

When analyze with nvc r1.14.0, it fails as follows

```console
shell$ nvc --version
nvc 1.14.0 (1.14.0.r0.g157b0f62) (Using LLVM 10.0.0)
Copyright (C) 2011-2024  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
shell$ nvc -a test_ng.vhd
** Error: subprogram NEW_PARAM [INTEGER, INTEGER return PARAM_TYPE] called before its body has been 
          elaborated
    > /mnt/d/ichiro/work/nvc/issues/new/test_ng.vhd:11
    |
  9 |     function  NEW_PARAM(X_SIZE   ,Y_SIZE   :integer) return PARAM_TYPE;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |     NEW_PARAM [INTEGER, INTEGER return PARAM_TYPE] declared here
 10 |     function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE;
 11 |     constant  PARAM_1x1  :  PARAM_TYPE := NEW_PARAM(1,1);
    |                                           ^^^^^^^^^^^^^^ error occurred here
    |
    = Help: IEEE Std 1076-2008 section 14.4.2 "Elaboration of a declaration"
** Error: subprogram NEW_PARAM [INTEGER, INTEGER, INTEGER, INTEGER return PARAM_TYPE] called before its 
          body has been elaborated
    > /mnt/d/ichiro/work/nvc/issues/new/test_ng.vhd:12
    |
 10 |     function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |     NEW_PARAM [INTEGER, INTEGER, INTEGER, INTEGER return PARAM_TYPE] declared here
 11 |     constant  PARAM_1x1  :  PARAM_TYPE := NEW_PARAM(1,1);
 12 |     constant  PARAM_3x3  :  PARAM_TYPE := NEW_PARAM(-1,1,-1,1);
    |                                           ^^^^^^^^^^^^^^^^^^^^ error occurred here
    |
    = Help: IEEE Std 1076-2008 section 14.4.2 "Elaboration of a declaration"
** Error: design unit depends on WORK.TEST_NG which was analysed with errors
    > /mnt/d/ichiro/work/nvc/issues/new/test_ng.vhd:18
    |
 18 | package body TEST_NG is
    | ^^^^^^^^^^^^^^^^^^^^
```

When analyze with nvc r1.13.1, it fails as follows

```console
shell$ nvc --version
nvc 1.13.1 (1.13.1.r0.gbd2b59df) (Using LLVM 10.0.0)
Copyright (C) 2011-2024  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
shell$ nvc -a test_ng.vhd
** Error: subprogram NEW_PARAM [INTEGER, INTEGER return PARAM_TYPE] called before its body has been 
          elaborated
    > /mnt/d/ichiro/work/nvc/issues/new/test_ng.vhd:11
    |
  9 |     function  NEW_PARAM(X_SIZE   ,Y_SIZE   :integer) return PARAM_TYPE;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |     NEW_PARAM [INTEGER, INTEGER return PARAM_TYPE] declared here
 10 |     function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE;
 11 |     constant  PARAM_1x1  :  PARAM_TYPE := NEW_PARAM(1,1);
    |                                           ^^^^^^^^^^^^^^ error occurred here
    |
    = Help: IEEE Std 1076-2008 section 14.4.2 "Elaboration of a declaration"
** Error: subprogram NEW_PARAM [INTEGER, INTEGER, INTEGER, INTEGER return PARAM_TYPE] called before its 
          body has been elaborated
    > /mnt/d/ichiro/work/nvc/issues/new/test_ng.vhd:12
    |
 10 |     function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |     NEW_PARAM [INTEGER, INTEGER, INTEGER, INTEGER return PARAM_TYPE] declared here
 11 |     constant  PARAM_1x1  :  PARAM_TYPE := NEW_PARAM(1,1);
 12 |     constant  PARAM_3x3  :  PARAM_TYPE := NEW_PARAM(-1,1,-1,1);
    |                                           ^^^^^^^^^^^^^^^^^^^^ error occurred here
    |
    = Help: IEEE Std 1076-2008 section 14.4.2 "Elaboration of a declaration"
** Error: design unit depends on WORK.TEST_NG which was analysed with errors
    > /mnt/d/ichiro/work/nvc/issues/new/test_ng.vhd:14
    |
 14 | package body TEST_NG is
    | ^^^^^^^^^^^^^^^^^^^^
```

However, When analyze with nvc r1.12.1, it succeeds as follows

```console
shell$ nvc --version
nvc 1.12.1 (1.12.1.r0.g9c861c52) (Using LLVM 10.0.0)
Copyright (C) 2011-2024  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
shell$ nvc -a test_ng.vhd
```

Have I missed something?

tmeissner, 2024-09-24:
I think, the error is correct. Questa also emits an error with the code.

```
$ vcom -2008 test.vhd
QuestaSim-64 vcom 2023.4 Compiler 2023.10 Oct  9 2023
Start time: 10:52:47 on Sep 24,2024
vcom -2008 test.vhd
-- Loading package STANDARD
-- Loading package TEXTIO
-- Loading package std_logic_1164
-- Loading package NUMERIC_STD
-- Compiling package TEST_NG
** Error (suppressible): test.vhd(11): (vcom-1594) Cannot call subprogram "NEW_PARAM" before it is elaborated.
"NEW_PARAM" is declarated at "test.vhd":9
** Error (suppressible): test.vhd(12): (vcom-1594) Cannot call subprogram "NEW_PARAM" before it is elaborated.
"NEW_PARAM" is declarated at "test.vhd":10
```

You could use a deferred constant like this:

```vhdl
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
package TEST_NG is
    type      PARAM_TYPE is record
                  SIZE_X  :  integer;
                  SIZE_Y  :  integer;
    end record;
    function  NEW_PARAM(X_SIZE   ,Y_SIZE   :integer) return PARAM_TYPE;
    function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE;
    constant  PARAM_1x1  :  PARAM_TYPE;
    constant  PARAM_3x3  :  PARAM_TYPE;
end TEST_NG;
package body TEST_NG is
    function  NEW_PARAM(X_SIZE   ,Y_SIZE   :integer) return PARAM_TYPE is
        variable  param  :  PARAM_TYPE;
    begin
        param.SIZE_X := X_SIZE;
        param.SIZE_Y := Y_SIZE;
        return param;
    end function;

    function  NEW_PARAM(X_LO,X_HI,Y_LO,Y_HI:integer) return PARAM_TYPE is
        variable  param  :  PARAM_TYPE;
    begin
        param.SIZE_X := Y_HI-X_LO;
        param.SIZE_Y := X_HI-Y_LO;
        return param;
    end function;
    constant  PARAM_1x1  :  PARAM_TYPE := NEW_PARAM(1,1);
    constant  PARAM_3x3  :  PARAM_TYPE := NEW_PARAM(-1,1,-1,1);
end package body;
```

ikwzm, 2024-09-24:
Thank you @tmeissner for the helpful advice.

However, I am disappointed, that I cannot use the description that generated the error.
This description would be more concise and easier to understand.
I would like some special option, considering that GHDL, Vivado and nvc r1.12.0 have no problems.

nickg, 2024-09-24:
The LRM text:

> For each declaration, the language rules (in particular scope and visibility rules) are such that it is either impossible or illegal to use a given item before the elaboration of its corresponding declaration. For example, it is not possible to use the name of a type for an object declaration before the corresponding type declaration is elaborated. Similarly, it is illegal to call a subprogram before its corresponding body is elaborated.

I believe the intent is to prevent circular definitions like this:

```vhdl
function f return integer;

constant c : integer := f;

function f return integer is
begin
    return c + 1;
end function;
```

Some of the code generation logic in NVC now relies on this ordering (part of the reason why I added the check) so I'm not sure it's a simple as downgrading to a warning with `--relaxed`.

ikwzm, 2024-09-25:
Thank you @tmeissner and @nickg for the helpful advice.
Your advice has been very helpful.
I will close this issue now.
Thank you very much.

closed 2024-09-25 by ikwzm