nickg/nvc #991: Error when using same library name as file

erick166, 2024-09-26:
I'm generating a new VVC using UVVM's vvc_generator.py script, which generates the following files:

 - test_bfm_pkg.vhd
 - **test_vvc.vhd**
 - transaction_pkg.vhd
 - vvc_cmd_pkg.vhd
 - vvc_context.vhd
 - vvc_methods_pkg.vhd
 - vvc_sb_pkg.vhd

If the library to which I'm compiling the files is named **test_vvc**, which is the same as one of the files, I get the following error:
`C:\uvvm\uvvm_vvc_framework\script\vvc_generator\output\test_vvc.vhd:70:35: error: no visible declaration for TEST_VVC.TD_CMD_QUEUE_PKG`

nickg, 2024-09-27:
Can you provide a self contained reproducer or the exact commands to run please?

erick166, 2024-09-27:
I think the easiest is if you have UVVM to run the script yourself. You can find the script under uvvm_vvc_framework/script/vvc_generator/
1. From that location run: python3 vvc_generator.py
2. "Enter the VVC name:" test
3. "Generate with extended features:" n
4. "Set number of concurrent channels:" 1
5. "Multiple executors?:" n

It will generate the 5 files mentioned above, you don't need to compile vvc_context.
If you compile the files to a library named "test_vvc" you will get the error.

commit aa11a9e0 (2024-09-29): Fix error when work library has same name as entity. Fixes #991
commit bafac753 (2024-10-21): Fix error when work library has same name as entity. Fixes #991
closed 2024-09-29 by nickg