nickg/nvc #993: PSL default clock unsupported when not in comment

Blebowski, 2024-09-26:
Hi,
I am trying to port:
[PSL_with_GHDL](https://github.com/tmeissner/psl_with_ghdl)

into NVC and see how far I get. Based on that I can try to prepare a table with PSL supported / unsupported features.

The first issue seems to be with default clock assignment in non-comment PSL sequences:
```
  -- All is sensitive to rising edge of clk
  default clock is rising_edge(clk);
```

Steps to reproduce:
```
git clone https://github.com/Blebowski/psl_with_ghdl
cd psl_with_ghdl
git checkout e3dd487e69eb9e9da5970e920e5d3a1741128cf8
cd sim
make psl_always
```

I get:
```
nvc --std=08 --work=lib_psl_always -a  ../src/pkg.vhd ../src/sequencer.vhd ../src/hex_sequencer.vhd
nvc --std=08 --work=lib_psl_always -a --psl ../src/psl_always.vhd
** Error: unexpected default while parsing concurrent statement, expecting one of process, postponed, component, entity, configuration, with, assert, block, if, for, case, ( or identifier
    > ../src/psl_always.vhd:26
    |
 26 |   default clock is rising_edge(clk);
    |   ^^^^^^^ this token was unexpected
make: *** [Makefile:14: psl_always] Error 1
```

Once I prefix the code with `--psl ` I am able to parse , elaborate and run the `psl_always` example.

nickg, 2024-09-26:
I don't think this is legal: a PSL clock declaration may only appear in an entity declarative item or a block declarative item. See 6.11 or the corresponding grammar rules:

> A PSL clock declaration may occur as an entity declarative item (see 3.2.3) or a block declarative item (3.3.2) and applies to certain PSL directives (if any) in the declarative region containing the PSL clock declaration. The PSL clock declaration, if any, that applies to a given PSL directive is the PSL clock declaration in the innermost declarative region containing both the given directive and a PSL clock directive. It is an error if more than one PSL clock declaration appears immediately with a given declarative region.

Blebowski, 2024-09-26:
Hmmm, I did not check that. I took the repo as "golden" standard.
Aldec also refuses it, but it also refuses it in entity declarative region.

Do you think it is something that makes sense to down-grade to warning with `--relaxed` ?
Or would you forbid it ?
If not, then I would, check latest GHDL behavior and open issue there.

Implementation-wise, I guess that would not be difficult.
It is little confusing since with `-- psl` it is allowed in the same place.

nickg, 2024-09-26:
> It is little confusing since with -- psl it is allowed in the same place.

Yeah, but it can be made consistent by rejecting it there too.

closed 2024-09-26 by nickg

tmeissner, 2024-09-27:
@nickg I assume the excerpt is from teh VHDL 2008 standard?

I think the problem is that for the PSL in comments, no real specification exists where these constructs can used as its already supported by vendors since the PSL standard exists, before VHDL 2008.

Setting restrictions also to PSL in comments will break a lot of code (at least of me), as I tend to add PSL in comments for all RTL files which are tested also in commercial simulators like Questa.

tmeissner, 2024-09-27:
And interesting, I've never read the PSL specific parts of the VHDL standard 😉 I'll try to make my psl_with_ghdl repo to comply with them 🙂

Blebowski, 2024-09-27:
I know I was the one saying the mixture is confusing, but I sort-of agree with @tmeissner here.

Some of my code also uses it quite a lot. Some simulators do accept the version in comment inside the architecture body. 
One other simulator did use to accept the version in comments, but newer versions did tighten the rules, same as NVC now.

I think if we were precise, GHDL should not accept the version that is without comment, since it contradict LRM.

I think best would be:
- Accept the version in comment anywhere, do not throw any warning / error. This is to keep compatibility with other tools.
- By default throw error on "non-comment" version in architecture body. Down-grade to "warning" with `--relaxed`. This is to keep compatibility with GHDL. I can open issue in GHDL.

nickg, 2024-09-29:
> 
> Accept the version in comment anywhere, do not throw any warning / error. This is to keep compatibility with other tools.

OK, I don't mind keeping the previous behaviour.

> 
> By default throw error on "non-comment" version in architecture body. Down-grade to "warning" with `--relaxed`. This is to keep compatibility with GHDL. I can open issue in GHDL.
> 

I don't what to do this because unlike the PSL comments there is actually a standard here and apparently only one tool that supports it so the usage can't be that widespread.

commit 88b1ac6f (2024-09-26): Only allow PSL declarations in declarative region. Fixes #993