nickg/nvc #996: Nested PSL SEREs are not supported

Blebowski, 2024-09-26:
A SERE that is nested withing another SERE can't be parsed by NVC.

Steps to reproduce:
```
git clone https://github.com/Blebowski/psl_with_ghdl
cd psl_with_ghdl
git checkout b514c001c934826661a9f4ff8f4415fd52112d45
cd sim
make psl_cover
```

I get:
```
nvc --std=08 --work=lib_psl_cover -a  ../src/pkg.vhd ../src/sequencer.vhd ../src/hex_sequencer.vhd
nvc --std=08 --work=lib_psl_cover -a --psl ../src/psl_cover.vhd
nvc --std=08 --work=lib_psl_cover -a tb_tops/psl_cover/testbench.vhd
nvc --std=08 --work=lib_psl_cover -e --cover tb_psl_cover
** Fatal: cannot handle PSL kind P_SERE in build_sere
    > ../src/psl_cover.vhd:37
    |
 37 |   -- psl COVER_1_c : cover {req; {{busy[=1]} && {not done[+]}}}
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
make: *** [Makefile:16: psl_cover] Error 1
```

Blebowski, 2024-11-16:
The actual issue is usage of `&&` (length matching SERE and).

Blebowski, 2024-11-17:
Hi @nickg , thanks for merging the two PSL merge requests.

I would like to look into this issue next. I think the change of PSL FSM lowering may not be sufficient.
Always taking each `EDGE_NEXT` allows to diverge from one state into two, but only upon clock event.
If the current FSM build logic were to process something like:
```
-- psl cover {{a;b;c}:{{x;y;z} & {k;l;m}}}
```
Then after building the `{a;b;c}` SERE, I would need to immediately (`EDGE_ESPILON`) activate two states
(since `c` is active in the same state as `x` and `k`).
Such two states will have `EDGE_EPSILON` to next state guarded by `x` and `k`. Note that
I can't do it via `EDGE_NEXT` if I want to keep the semantics of the assertion correct.

Some of the current PSL build logic relies on the assumption that the edge epsilon are processed
in `if-else` manner (e.g. the `abort` logic), so I don't want to change this behavior. 

Maybe the current model is not the most flexible. Maybe there could be a per-edge boolean flag
(`fallthrough`), that would behave like so:
```
taken = false;
for e in edges:
    if ( !has_guard(e) or (has_guard(e) and  guard(e) is met) ) {
        activate destination state (execute right away for EDGE_EPSILON, or schedule enter for EDGE_NEXT)
        taken = true
        if ( !fallthrough)
            break;
    }
if ( !taken and is ASSERT_ALWAYS)
   fire an assert.
```

Such behavior would allow keeping `if-else` logic useful for `abort`. The edge itself would decide
whether it lets edges behind it to be taken.
At the same time, it would allow arbitrary number of `EDGE_NEXT` or `EDGE_EPSILON` to be taken,
thus diverge the FSM behavior arbitrarily.

I don't know how to implement this though. Current `psl_lower` logic uses `emit_jump` for the
`EDGE_EPSILON` which does the `if-else`. Taking multiple code paths would require some sort
of `call / return` for each state entered. 

What do you think about this ?

Blebowski, 2024-11-28:
Hi @nickg ,
did you manage to have a look at this topic ?
I would appreciate any feedback.

nickg, 2024-12-01:
The `EDGE_EPSILON` transitions are mainly a convenience for constructing the FSM, it should always be possible to construct and equivalent FSM without them. I've done that and added support for `&&`. It works in simple cases at least.

closed 2024-12-01 by nickg