nickg/nvc #1003: Fatal: out of memory - problems

obruendl, 2024-09-27:
Hi Nick

I run into out-of-memory problems when running large pseudo-random simulations with VUnit. Could you have a look?

https://github.com/open-logic/open-logic/actions/runs/11076670221/job/30780395154

I see that you provide a -H option for setting the heap. However, I could not find out how to use this with VUnit. And in general ideally I wouldn't have to pass any special options for those test-cases. Neither GHDL nor Questa require a special handling. The simulations are a bit bigger but still not huge.

nickg, 2024-09-27:
VUnit has a `nvc.heap_size` option to set that.

https://vunit.github.io/py/opts.html#simulation-options

obruendl, 2024-10-15:
Thank you. This fixed the problem.

closed 2024-10-15 by obruendl