nickg/nvc #999: Install for uvvm

avelure, 2024-09-27:
I tried the install option for uvvm and it seems to have a problem with some path handling.
This is from within the clang64 mintty environment
```
$ nvc --install uvvm
mkdir -p /home/arild.velure/.cache/nvc/UVVM-v2024.07.03b
git clone --recursive https://github.com/UVVM/UVVM -b v2024.07.03b /home/arild.velure/.cache/nvc/UVVM-v2024.07.03b --depth=1
Cloning into '/home/arild.velure/.cache/nvc/UVVM-v2024.07.03b'...
remote: Enumerating objects: 758, done.
remote: Counting objects: 100% (758/758), done.
remote: Compressing objects: 100% (558/558), done.
remote: Total 758 (delta 325), reused 445 (delta 175), pack-reused 0 (from 0)
Receiving objects: 100% (758/758), 23.16 MiB | 23.70 MiB/s, done.
Resolving deltas: 100% (325/325), done.
Note: switching to '38c08582f1882b4e5b845621949657c7677b8a24'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Updating files: 100% (787/787), done.
cd /home/arild.velure/.cache/nvc/UVVM-v2024.07.03b

################################################################################
compiling: uvvm_util
################################################################################

C:\msys64\clang64\bin\nvc.exe --std=2008 --work=C:/msys64/home/arild.velure/.nvc/lib/uvvm_util.08 -LC:/msys64/home/arild.velure/.nvc/lib -a --relaxed ../src/types_pkg.vhd
: The filename, directory name, or volume label syntax is incorrect.

** Fatal: bash C:\msys64\clang64\libexec\nvc\install-uvvm.sh failed with status 1
```

I tried from the native windows11 commandline and then it seems to try to execute the install script with WSL instead of with `sh` which is in the path from msys. This might not be a nvc issue though.
```
<3>WSL (18) ERROR: CreateProcessParseCommon:711: Failed to translate C:\proj\inventas\uvvm\uvvm_vvc_framework\script\vvc_generator\output
<3>WSL (18) ERROR: CreateProcessParseCommon:757: getpwuid(0) failed 2
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\ConEmu
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\ConEmu\ConEmu
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\Common Files\Oracle\Java\javapath
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\system32
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\System32\Wbem
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\System32\WindowsPowerShell\v1.0\
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\System32\OpenSSH\
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\ProgramData\chocolatey\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\dotnet\
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\PuTTY\
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\Git\cmd
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\TortoiseGit\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\nodejs\
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Program Files\Docker\Docker\resources\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Local\Microsoft\WindowsApps
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Local\Programs\Microsoft VS Code\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Local\anaconda3
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Local\anaconda3\Scripts
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Local\anaconda3\Library\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\mentor\modeltech_pe_2023.4\win32pe
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\intelFPGA_pro\23.4\questa_fse\win64
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\msys64\ucrt64\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\msys64\clang64\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\msys64\opt\ghdl_llvm\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Microsemi\Libero_v9.2_SP4\Designer\bin
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Local\metrics-ca
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Users\arild.velure\AppData\Roaming\npm
<3>WSL (18) ERROR: UtilTranslatePathList:2866: Failed to translate c:\msys64\usr\bin
Processing fstab with mount -a failed.
Failed to mount C:\, see dmesg for more details.

<3>WSL (18) ERROR: CreateProcessEntryCommon:334: getpwuid(0) failed 2
<3>WSL (18) ERROR: CreateProcessEntryCommon:505: execvpe /bin/bash failed 2
<3>WSL (18) ERROR: CreateProcessEntryCommon:508: Create process not expected to return
** Fatal: bash C:\msys64\clang64\libexec\nvc\install-uvvm.sh failed with status 1
```

Running `sh C:\msys64\clang64\libexec\nvc\install-uvvm.sh` from the win11 commandline gives the same result as from the clang64 mintty.

If I `cd /home/arild.velure/.cache/nvc/UVVM-v2024.07.03b/uvvm_util/script/` and do the failing command displayed in the script output `nvc.exe --std=2008 --work=C:/msys64/home/arild.velure/.nvc/lib/uvvm_util.08 -LC:/msys64/home/arild.velure/.nvc/lib -a --relaxed ../src/types_pkg.vhd` it is parsed ok.

nickg, 2024-09-29:
> Running sh C:\msys64\clang64\libexec\nvc\install-uvvm.sh from the win11 commandline gives the same result as from the clang64 mintty.

How about if you try `/usr/bin/bash /msys64/clang64/libexec/nvc/install-uvvm.sh` from the clang64 prompt? Also try `which bash` and `which sh`. 

> I tried from the native windows11 commandline and then it seems to try to execute the install script with WSL instead of with sh which is in the path from msys.

This is unfortunate. It just picks up the first `bash.exe` in the path. Longer term I'd like to replace these shell scripts with TCL as the interpreter is then embedded, but that isn't a hard dependency at the moment.

avelure, 2024-11-01:
Sorry, forgot about this one.

```
$ /usr/bin/bash /c/msys64/clang64/libexec/nvc/install-uvvm.sh
cd /home/arild.velure/.cache/nvc/UVVM-v2024.07.03b

################################################################################
compiling: uvvm_util
################################################################################

nvc --std=2008 --work=C:/msys64/home/arild.velure/.nvc/lib/uvvm_util.08 -LC:/msys64/home/arild.velure/.nvc/lib -a --relaxed ../src/types_pkg.vhd
: The filename, directory name, or volume label syntax is incorrect.

$ which bash
/usr/bin/bash

$ which sh
/usr/bin/sh
```