nickg/nvc #1004: Feature request: Warning suppression and escalation to error command-line flags

soronpo, 2024-09-30:
I'm missing common command-line option that:
* Escalate warnings into errors (so that the program returns non-zero error code)
* Completely suppress specific warnings

nickg, 2024-09-30:
Can you give some examples of the warnings you want suppress and/or those you want to escalate?

soronpo, 2024-10-01:
Warning suppression example of shared variables that model TrueDPR:
```
** Warning: shared variable RAM must have protected type
    > TrueDPR\hdl\TrueDPR.vhd:28
    |
 28 |   shared variable ram : t_vecX1_std_logic_vector(0 to 2 ** ADDR_WIDTH - 1)(DATA_WIDTH - 1 downto 0);
    |                   ^^^
```
Escalation is a companion to suppression. I strive for 0 warnings build/simulations. So all warnings are either considered errors that must be fixed or alternatively suppressed and completely ignored.

Blebowski, 2024-10-05:
Related to: https://github.com/nickg/nvc/discussions/784