nickg/nvc #1008: Feature Request: Sensitivity List Check

NikLeberg, 2024-10-05:
Hi there,
I have a small feature request. Given NVC's history as a linter (a long time ago), it would be awesome if it could report a mismatch between a VHDL process's sensitivity list and the signals actually read within the process. For example, if a signal is read but not included in the list, a warning would be emitted.

MWE:
```vhd
ENTITY test IS
    PORT (
        a, b : IN BOOLEAN;
        y    : OUT BOOLEAN
    );
END ENTITY test;

ARCHITECTURE sim OF test IS
BEGIN
    PROCESS (a) IS
    BEGIN
        y <= a AND b;
    END PROCESS;
END ARCHITECTURE sim;
```

For the above entity, running `nvc -a test.vhd` would emit a warning about missing `b` signal in process sensitivity list.

For ModelSim/Questa the flag `-check_synthesis` to `vcom` does exactly this and emits:
```shell
vcom -check_synthesis test.vhd
Questa Intel Starter FPGA Edition-64 vcom 2021.2 Compiler 2021.04 Apr 14 2021
Start time: 12:52:56 on Oct 05,2024
vcom -check_synthesis test.vhd
-- Loading package STANDARD
-- Compiling entity test
-- Compiling architecture sim of test
** Warning: test.vhd(12): (vcom-1400) Synthesis Warning: Signal "b" is read in the process but is not in the sensitivity list.
End time: 12:52:56 on Oct 05,2024, Elapsed time: 0:00:00
Errors: 0, Warnings: 1
```

Thanks!

nickg, 2024-10-14:
I've added a `--check-synthesis` analysis option that does this. See [test/simp/synth1.vhd](https://github.com/nickg/nvc/blob/master/test/simp/synth1.vhd) for examples of cases it detects. I tried it on a few real world projects too and didn't notice any false positives.

closed 2024-10-14 by nickg

NikLeberg, 2024-10-15:
Amazing! Thanks for adding this. It might save some poor souls a few hours of debugging. I know it would (and will) have helped me. 😅

commit 743f9d51 (2024-10-14): Add analysis option to detect incomplete sensitivity lists

Fixes #1008