nickg/nvc #1035: Accessing elements of unconstrained record via VHPI

cameronweston, 2024-10-21:
When testing a program with cocotb, I came across an issue where I can't access the elements from record that was declared with unconstrained types. Accessing the element in python throws an error. If I use cocotb's dir function to view the record's sub-handles, the elements are not shown as is the case with a constrained record. 

tb_test.vhd
```vhdl 
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_test is
end tb_test;

architecture tb of tb_test is
  type unconstrainedRecord is record
    a : std_logic;
    b : std_logic_vector;
  end record;
  subtype uRecConstrained is unconstrainedRecord
    (
      b(1 downto 0)
    );
  type constrainedRecord is record
    a : std_logic;
    b : std_logic_vector(1 downto 0);
  end record;

  signal uRec : unconstrainedRecord(b(1 downto 0));
  signal cRec : constrainedRecord;
  signal uRecC: uRecConstrained;
begin
end tb;
```

test_tb.py
```python
from cocotb.regression import TestFactory
import logging
import cocotb
from cocotb.runner import get_runner

@cocotb.test()
async def test_rec(dut):
  log = logging.getLogger("cocotb.tb")
  log.setLevel(logging.DEBUG)
  log.debug(f"Constrained record: {dir(dut.cRec)}")
  log.debug(f"Unconstrained record: {dir(dut.uRec)}")
  log.debug(f"Unconstrained subtype: {dir(dut.uRecC)}")
  cRecb = dut.cRec.b
  #uRecb = dut.uRec.b # Causes error
  #uRecC = dut.uRecC.b # Causes error

def test_tb_runner():
  sources = ["tb_test.vhd"]

  runner = get_runner("nvc")

  runner.build(
    sources=sources,
    hdl_toplevel="tb_test",
    always=True
  )

  runner.test(hdl_toplevel="tb_test", test_module="test_tb,")
```

Output
```
0.00ns DEBUG    cocotb.tb Constrained record: ['A', 'B', '_HierarchyObject__get_sub_handle_by_name', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_child_path', '_compat_mapping', '_def_file', '_def_name', '_discover_all', '_discovered', '_fullname', '_handle', '_id', '_invalid_sub_handles', '_len', '_log', '_name', '_path', '_sub_handle_key', '_sub_handles', '_type', 'get_definition_file', 'get_definition_name']   
0.00ns DEBUG    cocotb.tb Unconstrained record: ['_HierarchyObject__get_sub_handle_by_name', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_child_path', '_compat_mapping', '_def_file', '_def_name', '_discover_all', '_discovered', '_fullname', '_handle', '_id', '_invalid_sub_handles', '_len', '_log', '_name', '_path', '_sub_handle_key', '_sub_handles', '_type', 'get_definition_file', 'get_definition_name']
0.00ns DEBUG    cocotb.tb Unconstrained subtype: ['_HierarchyObject__get_sub_handle_by_name', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_child_path', '_compat_mapping', '_def_file', '_def_name', '_discover_all', '_discovered', '_fullname', '_handle', '_id', '_invalid_sub_handles', '_len', '_log', '_name', '_path', '_sub_handle_key', '_sub_handles', '_type', 'get_definition_file', 'get_definition_name']
```

nickg, 2024-10-21:
Should be fixed now on the master branch. I'll make a 1.14.1 release this weekend hopefully.

closed 2024-10-21 by nickg

cameronweston, 2024-10-21:
As always you make quick work of fixing issues. Thank you!

commit 4cd7d022 (2024-10-21): Handle record subtypes in VHPI. Fixes #1035
commit e209132d (2024-10-26): Handle record subtypes in VHPI. Fixes #1035