nickg/nvc #1038: Fuzz testing

avelure, 2024-10-25:
I ran the fuzzing for a week and collected 32 crashing cases. 
[crashes_cmin.zip](https://github.com/user-attachments/files/17522011/crashes_cmin.zip)
There might be some duplicates in here as there often is different code paths that lead to the same crash.
The log below shows SIGABRT, but you can ignore this as this is because it is neceseary to modify the source to raise a signal to the fuzzer when something interesting occurs. You can see my patch here: https://github.com/avelure/vhdl_fuzz/blob/main/nvc_bug_abort.patch

```
# find . -maxdepth 1 -type f -exec echo "file: {}" \; -exec /usr/local/bin/nvc --std=08 --ignore-time -a --relaxed --error-limit=1 --psl {} \;
file: ./12dc86da02baf50abcdb2d9777c5ff7f6ef12056c14535fd30c1ad9e50f71411
nvc: ../src/tree.c:587: tree_kind_t tree_kind(tree_t): Assertion `t != NULL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x55555560d299] /nvc/build/../src/util.c:875 signal_handler
-->    show_stacktrace();
[0x7ffff0b1351f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x7ffff0b679fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) pthread_kill
[0x7ffff0b13475] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x7ffff0af97f2] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x7ffff0af971a] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x7ffff0b0ae95] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x55555573bafa] tree_kind
[0x55555573bafa] ../src/tree.c:655 is_subprogram
[0x55555581302f] /nvc/build/../src/names.c:792 make_visible
             return;   // Ignore ports in component, etc.
-->       else if (is_subprogram(it->container))
             return;   // Do not warn when subprogram parameters hide ports
[0x555555627498] /nvc/build/../src/parse.c:7265 p_variable_declaration
          tree_add_decl(parent, t);
-->       insert_name(nametab, t, it->ident);
          sem_check(t, nametab);
[0x5555556220ba] /nvc/build/../src/parse.c:12936 p_package_body_declarative_item
       case tSHARED:
-->       p_variable_declaration(parent);
          break;
[0x555555620858] p_package_body_declarative_part
[0x555555620858] /nvc/build/../src/parse.c:13064 p_package_body
-->    p_package_body_declarative_part(body);
[0x555555622239] /nvc/build/../src/parse.c:12987 p_package_body_declarative_item
             require_std(STD_08, "nested package declarations");
-->          tree_add_decl(parent, p_package_body(NULL));
          }
[0x555555620858] p_package_body_declarative_part
[0x555555620858] /nvc/build/../src/parse.c:13064 p_package_body
-->    p_package_body_declarative_part(body);
[0x5555556180b3] /nvc/build/../src/parse.c:13098 p_secondary_unit
       case tPACKAGE:
-->       p_package_body(unit);
          break;
[0x555555613593] p_library_unit
[0x555555613593] p_design_unit
[0x555555613593] /nvc/build/../src/parse.c:13236 parse
-->    tree_t unit = p_design_unit();
[0x555555749617] /nvc/build/../src/common.c:2468 analyse_file
             tree_t unit;
-->          while (base_errors = error_count(), (unit = parse())) {
                if (error_count() == base_errors) {
[0x5555555eeed7] analyse
[0x5555555eeed7] /nvc/build/../src/nvc.c:2135 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5555555ec042] /nvc/build/../src/nvc.c:2297 main
-->    const int ret = process_command(argc, argv, &state);
find: '/usr/local/bin/nvc' terminated by signal 6
file: ./1725b8e09f4abd33a4be1451e6b69e2f56112371ad021eeffd3deb4e2334def1
nvc: ../src/type.c:164: _Bool _type_eq(type_t, type_t, _Bool, hash_t *): Assertion `a != NULL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x55555560d299] /nvc/build/../src/util.c:875 signal_handler
[... 1132 lines cut ...]
-->    tree_t unit = p_design_unit();
[0x555555749617] /nvc/build/../src/common.c:2468 analyse_file
             tree_t unit;
-->          while (base_errors = error_count(), (unit = parse())) {
                if (error_count() == base_errors) {
[0x5555555eeed7] analyse
[0x5555555eeed7] /nvc/build/../src/nvc.c:2135 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5555555ec042] /nvc/build/../src/nvc.c:2297 main
-->    const int ret = process_command(argc, argv, &state);
find: '/usr/local/bin/nvc' terminated by signal 6
```
Second part of log comes in a new comment as there is a comment length limit.

There was also one that causes an `input buffer overflow, can't enlarge buffer because scanner uses REJECT`, but that might not be so interesting.
[input_buffer_overflow.zip](https://github.com/user-attachments/files/17522164/input_buffer_overflow.zip)

Then there were 5 cases that seem to hang the parser or the parser takes an abnormal long time to complete.
[hangs.zip](https://github.com/user-attachments/files/17522192/hangs.zip)

Currently I'm just using the analyze switch for fuzzing which is a bit slow. I guess I could use the `--syntax` switch to parse the file, but it is missing some of the analysis switches like `--psl`, `--relaxed` and `--error-limit`, though it seems quite easy to add, so maybe I can make a pullrequest.
Another option is to use persistence mode to loop the parser and just change the input buffer to speed up the testing https://github.com/AFLplusplus/AFLplusplus/blob/stable/instrumentation/README.persistent_mode.md
this also seems feasible in nvc, so I can look at this next.

avelure, 2024-10-25:
Second part of log:
```
file: ./7c96d591351ae9d499df39a80e42815cb94e68692fd17fac9e52bb9001ed70ed
nvc: ../src/parse.c:4622: tree_t p_primary(tree_t): Assertion `is_scanned_as_psl()' failed.

*** Caught signal 6 (SIGABRT) ***

[0x55555560d299] /nvc/build/../src/util.c:875 signal_handler
-->    show_stacktrace();
[0x7ffff0b1351f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x7ffff0b679fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) pthread_kill
[0x7ffff0b13475] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x7ffff0af97f2] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x7ffff0af971a] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x7ffff0b0ae95] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x55555564bc36] /nvc/build/../src/parse.c:4622 p_primary
       case tNONDETV:
-->       assert(is_scanned_as_psl());
          return p_psl_builtin_function_call();
[0x55555564efb9] p_unary_expression
[0x55555564efb9] /nvc/build/../src/parse.c:4704 p_factor
-->    tree_t operand = p_unary_expression(head);
[0x55555564bf73] /nvc/build/../src/parse.c:4743 p_term
-->    tree_t term = p_factor(head);
[0x55555564fb5f] /nvc/build/../src/parse.c:4798 p_simple_expression
       else
-->       expr = p_term(head);
[0x55555564f593] /nvc/build/../src/parse.c:4836 p_shift_expression
-->    tree_t shift = p_simple_expression(head);
[0x55555564e6c3] /nvc/build/../src/parse.c:4893 p_relation
-->    tree_t rel = p_shift_expression(head);
[0x55555564c393] /nvc/build/../src/parse.c:4917 p_expression_with_head
-->    tree_t expr = p_relation(head);
[0x55555564e270] /nvc/build/../src/parse.c:0 p_range
[0x55555566d97f] /nvc/build/../src/parse.c:2958 p_range_constraint
          {
-->          tree_t r = p_range(expr1);
             solve_types(nametab, r, constraint);
[0x55555563caa0] p_constraint
[0x55555563caa0] /nvc/build/../src/parse.c:4226 p_subtype_indication
-->       p_constraint(type);
       }
[0x55555562f76b] /nvc/build/../src/parse.c:6785 p_subtype_declaration
-->    type_t sub = p_subtype_indication();
[0x555555636f9e] p_package_declarative_item
[0x555555636f9e] p_package_declarative_part
[0x555555636f9e] /nvc/build/../src/parse.c:8820 p_package_declaration
-->    p_package_declarative_part(pack);
[0x555555615eac] /nvc/build/../src/parse.c:9390 p_primary_unit
          else
-->          p_package_declaration(unit);
          break;
[0x5555556135e3] p_library_unit
[0x5555556135e3] p_design_unit
[0x5555556135e3] /nvc/build/../src/parse.c:13236 parse
-->    tree_t unit = p_design_unit();
[0x555555749617] /nvc/build/../src/common.c:2468 analyse_file
             tree_t unit;
-->          while (base_errors = error_count(), (unit = parse())) {
                if (error_count() == base_errors) {
[0x5555555eeed7] analyse
[0x5555555eeed7] /nvc/build/../src/nvc.c:2135 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5555555ec042] /nvc/build/../src/nvc.c:2297 main
-->    const int ret = process_command(argc, argv, &state);
find: '/usr/local/bin/nvc' terminated by signal 6
file: ./829c8caa83d8e941a9cf4d5dbc823735e159a32d4ecd35e5bd95d0b13b8e34af
nvc: ../src/parse.c:4622: tree_t p_primary(tree_t): Assertion `is_scanned_as_psl()' failed.
[... 665 lines cut ...]
-->       tree_add_stmt(parent, p_concurrent_statement());
    }
[0x555555618e70] p_architecture_statement_part
[0x555555618e70] p_architecture_body
[0x555555618e70] /nvc/build/../src/parse.c:13094 p_secondary_unit
       case tARCHITECTURE:
-->       p_architecture_body(unit);
          break;
[0x555555613593] p_library_unit
[0x555555613593] p_design_unit
[0x555555613593] /nvc/build/../src/parse.c:13236 parse
-->    tree_t unit = p_design_unit();
[0x555555749617] /nvc/build/../src/common.c:2468 analyse_file
             tree_t unit;
-->          while (base_errors = error_count(), (unit = parse())) {
                if (error_count() == base_errors) {
[0x5555555eeed7] analyse
[0x5555555eeed7] /nvc/build/../src/nvc.c:2135 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5555555ec042] /nvc/build/../src/nvc.c:2297 main
-->    const int ret = process_command(argc, argv, &state);
find: '/usr/local/bin/nvc' terminated by signal 6
```

NikLeberg, 2024-10-30:
Hi @avelure! Your patch is not accessible anmore. Did you change it to private? Would be very interested in looking at your fuzzing setup.
Did you use AFL or llvm's libfuzzer?

avelure, 2024-10-31:
Sorry, I had forgotten to set the repo public. Should work now.
It uses aflplusplus.

NikLeberg, 2024-11-01:
I can reproduce the error from file `12dc86da02baf50abcdb2d9777c5ff7f6ef12056c14535fd30c1ad9e50f71411` with the following more minimized example:

```vhd
package foo_pkg is 
end foo_pkg;

package body foo_pkg is
  shared variable var : integer; 

  package bar_pkg is
  end bar_pkg;

  package body bar_pkg is
    shared variable var : integer; 
  end package body bar_pkg;
end package body foo_pkg;
```

```
root@nvc:/workspaces/nvc/fuzz_test# nvc --std=08 --ignore-time -a --relaxed --error-limit=1 --psl 12dc86d.vhd 
** Warning: shared variable VAR must have protected type
   > 12dc86d.vhd:5
   |
 5 |   shared variable var : integer; 
   |                   ^^^
nvc: ../src/tree.c:587: tree_kind: Assertion `t != NULL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x55bc207ab0c3] ../src/util.c:872 signal_handler
-->    show_stacktrace();
[0x7ff7450a131f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7ff7450fab1c] (/usr/lib/x86_64-linux-gnu/libc.so.6) pthread_kill
[0x7ff7450a126d] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x7ff7450848fe] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x7ff74508481a] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7ff745097506] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x55bc207eaabd] ../src/tree.c:587 tree_kind
    {
-->    assert(t != NULL);
       return t->object.kind;
[0x55bc2082645f] ../src/common.c:655 is_subprogram
    {
-->    switch (tree_kind(t)) {
       case T_FUNC_DECL:
[0x55bc20880750] ../src/names.c:792 warn_hidden_decl
             return;   // Ignore ports in component, etc.
-->       else if (is_subprogram(it->container))
             return;   // Do not warn when subprogram parameters hide ports
[0x55bc20880c76] ../src/names.c:874 make_visible
             else if (dd->visibility == DIRECT && (dd->mask & mask & N_OBJECT))
-->             warn_hidden_decl(s, dd, decl);
[0x55bc20881792] ../src/names.c:1038 make_visible_slow
-->    make_visible(s, name, decl, DIRECT, s);
    }
[0x55bc2088274c] ../src/names.c:1358 insert_name
    {
-->    make_visible_slow(tab->top_scope, alias ?: tree_ident(decl), decl);
    }
[0x55bc207c8b26] ../src/parse.c:7265 p_variable_declaration
          tree_add_decl(parent, t);
-->       insert_name(nametab, t, it->ident);
          sem_check(t, nametab);
[0x55bc207da5e2] ../src/parse.c:12998 p_package_body_declarative_item
       case tSHARED:
-->       p_variable_declaration(parent);
          break;
[0x55bc207da921] ../src/parse.c:13071 p_package_body_declarative_part
       while (not_at_token(tEND))
-->       p_package_body_declarative_item(unit);
    }
[0x55bc207dac8d] ../src/parse.c:13126 p_package_body
-->    p_package_body_declarative_part(body);
[0x55bc207da7b5] ../src/parse.c:13049 p_package_body_declarative_item
             require_std(STD_08, "nested package declarations");
-->          tree_add_decl(parent, p_package_body(NULL));
          }
[0x55bc207da921] ../src/parse.c:13071 p_package_body_declarative_part
       while (not_at_token(tEND))
-->       p_package_body_declarative_item(unit);
    }
[0x55bc207dac8d] ../src/parse.c:13126 p_package_body
-->    p_package_body_declarative_part(body);
[0x55bc207dae2e] ../src/parse.c:13160 p_secondary_unit
       case tPACKAGE:
-->       p_package_body(unit);
          break;
[0x55bc207daf6b] ../src/parse.c:13187 p_library_unit
          if (peek_nth(2) == tBODY)
-->          p_secondary_unit(unit);
          else
[0x55bc207db2e9] ../src/parse.c:13245 p_design_unit
       p_context_clause(unit);
-->    p_library_unit(unit);
[0x55bc207db509] ../src/parse.c:13298 parse
-->    tree_t unit = p_design_unit();
[0x55bc2082b68f] ../src/common.c:2486 analyse_file
             tree_t unit;
-->          while (base_errors = error_count(), (unit = parse())) {
                if (error_count() == base_errors) {
[0x55bc207a0f9c] ../src/nvc.c:255 analyse
          else
-->          analyse_file(argv[i], jit, state->registry);
       }
[0x55bc207a5716] ../src/nvc.c:2135 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x55bc207a5c5a] ../src/nvc.c:2297 main
-->    const int ret = process_command(argc, argv, &state);
```

NikLeberg, 2024-11-02:
I can reproduce the error from file `1725b8e09f4abd33a4be1451e6b69e2f56112371ad021eeffd3deb4e2334def1` with the following more minimized example:
```vhd
package foo is
  alias TO_OCTAL_STRING is ns;
end package foo;
```

<details>
<summary>nvc crash log</summary>
  
```
root@nvc:/workspaces/nvc/fuzz_test# nvc --std=08 -a --relaxed --psl 1725b8e.vhd
nvc: ../src/type.c:164: _type_eq: Assertion `a != NULL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x5580cb3ea0c3] ../src/util.c:872 signal_handler
-->    show_stacktrace();
[0x7f0272a1331f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7f0272a6cb1c] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./nptl/pthread_kill.c:44 __pthread_kill_implementation
[0x7f0272a6cb1c] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./nptl/pthread_kill.c:78 __pthread_kill_internal
[0x7f0272a6cb1c] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./nptl/pthread_kill.c:89 pthread_kill@@GLIBC_2.34
[0x7f0272a1326d] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/posix/raise.c:26 raise
[0x7f02729f68fe] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./stdlib/abort.c:79 abort
[0x7f02729f681a] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./assert/assert.c:94 __assert_fail_base.cold
[0x7f0272a09506] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./assert/assert.c:103 __assert_fail
[0x5580cb43107a] ../src/type.c:164 _type_eq
    {
-->    assert(a != NULL);
       assert(b != NULL);
[0x5580cb431916] ../src/type.c:264 type_eq
    {
-->    return _type_eq(a, b, false, NULL);
    }
[0x5580cb4bfdb5] ../src/names.c:894 make_visible
          else if (overload && kind == DIRECT
-->                && type_eq(type, tree_type(dd->tree))) {
             if (dd->origin != s) {
[0x5580cb4c0792] ../src/names.c:1038 make_visible_slow
-->    make_visible(s, name, decl, DIRECT, s);
    }
[0x5580cb4c174c] ../src/names.c:1358 insert_name
    {
-->    make_visible_slow(tab->top_scope, alias ?: tree_ident(decl), decl);
    }
[0x5580cb408447] ../src/parse.c:7439 p_alias_declaration
-->    insert_name(nametab, t, NULL);
       sem_check(t, nametab);
[0x5580cb40c3e9] ../src/parse.c:8709 p_package_declarative_item
       case tALIAS:
-->       p_alias_declaration(pack);
          break;
[0x5580cb40c616] ../src/parse.c:8754 p_package_declarative_part
       while (not_at_token(tEND))
-->       p_package_declarative_item(pack);
    }
[0x5580cb40c979] ../src/parse.c:8820 p_package_declaration
-->    p_package_declarative_part(pack);
[0x5580cb40e8bd] ../src/parse.c:9390 p_primary_unit
          else
-->          p_package_declaration(unit);
          break;
[0x5580cb419f79] ../src/parse.c:13189 p_library_unit
          else
-->          p_primary_unit(unit);
          break;
[0x5580cb41a2e9] ../src/parse.c:13245 p_design_unit
       p_context_clause(unit);
-->    p_library_unit(unit);
[0x5580cb41a509] ../src/parse.c:13298 parse
-->    tree_t unit = p_design_unit();
[0x5580cb46a68f] ../src/common.c:2486 analyse_file
             tree_t unit;
-->          while (base_errors = error_count(), (unit = parse())) {
                if (error_count() == base_errors) {
[0x5580cb3dff9c] ../src/nvc.c:255 analyse
          else
-->          analyse_file(argv[i], jit, state->registry);
       }
[0x5580cb3e4716] ../src/nvc.c:2135 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5580cb3e4c5a] ../src/nvc.c:2297 main
-->    const int ret = process_command(argc, argv, &state);
```
</details>

nickg, 2024-11-02:
> Currently I'm just using the analyze switch for fuzzing which is a bit slow. I guess I could use the --syntax switch to parse the file, but it is missing some of the analysis switches like --psl, --relaxed and --error-limit, though it seems quite easy to add, so maybe I can make a pullrequest.

The `--syntax` command has some issues at the moment: I don't think it works properly if a file contains multiple design units and NVC has been able to separate parsing from other checks for a long time.  I'm think I'd rather remove it and add `--no-save` flag to `-a` that would skip saving the working library to disk.

You might want to try disabling LLVM while you're doing this (`--disable-llvm` configure flag). Most Linux distributions have disabled lazy binding in recent years which means the whole LLVM library gets loaded at start-up regardless of whether it's used or not, and that thing is *huge*. Here's an example with `bin/nvc` linked against `libLLVM.so`:

```
 Performance counter stats for './bin/nvc -a ../test/regress/wait1.vhd':

        74,518,234      instructions              #    1.50  insn per cycle         
        49,552,361      cycles                                                      

       0.011137862 seconds time elapsed

       0.005594000 seconds user
       0.005594000 seconds sys
```

And with `--disable-llvm`:

```
 Performance counter stats for './x-interp/bin/nvc -a ../test/regress/wait1.vhd':

        19,419,825      instructions              #    1.53  insn per cycle         
        12,671,157      cycles                                                      

       0.003238301 seconds time elapsed

       0.003335000 seconds user
       0.000000000 seconds sys
```

So ~4x faster.

NikLeberg, 2024-11-02:
Reproducer for error from file `181a6c3f1ee6985b3b3d3f5b38b8d6f81f5c826a761045cb7740b8d93a753314` with the following more minimized example:

```vhd
ENTITY next_ent IS
END next_ent;

ARCHITECTURE arch OF next_ent IS
  TYPE integer_array IS ARRAY (NATURAL RANGE <>) OF INTEGER;
  FUNCTION func1(ia: integer_array := (1,2,3,next)) RETURN BOOLEAN;
BEGIN
END ARCHITECTURE arch;
```

NikLeberg, 2024-11-02:
Reproducer for error from file `60bc12eff12e4f7e41dc5499ffb25c74da54d4f78c77d1200d3621e7a8d6ca30` with the following (slightly) more minimized example:

```vhd
library ieee;
use ieee.std_logic_1164.all;

entity GENERIC_WHEN is
   generic( FOO  :     std_logic_vector(1 downto 0) );
   port(    IN1  : in  std_logic_vector(1 downto 0);
            OUT1 : out std_logic_vector(1 downto 0) );
end GENERIC_WHEN;

architecture BEHAVIOUR of GENERIC_WHEN is
begin
   process( IN1 )
   begin
      case IN1 is
         when FOO => OUT1 <= FOO;
         when others => null;
      end case;
   end process;
end BEHAVIOUR;
```

@nickg I don't really know how that should be fixed. The following dirty hack works, but I'm unsure what effects it causes in the depths of other codepaths.

```diff
diff --git a/src/common.c b/src/common.c
index 3b6d24b5..bebd594d 100644
--- a/src/common.c
+++ b/src/common.c
@@ -1599,7 +1599,10 @@ unsigned get_case_choice_char(tree_t value, int depth)
    case T_REF:
       {
          tree_t decl = tree_ref(value);
-         assert(tree_kind(decl) == T_CONST_DECL || tree_kind(decl) == T_ALIAS);
+         tree_kind_t kind = tree_kind(decl);
+         if (kind == T_GENERIC_DECL)
+            return ~0;
+         assert(kind == T_CONST_DECL || kind == T_ALIAS);
          assert(tree_has_value(decl));
          return get_case_choice_char(tree_value(decl), depth);
       }
```

NikLeberg, 2024-11-02:
Reproducer for error from file `8a914f40d873ff7bec49e7993899821246958566831a90b871c3bbc1081a4566` with the following more minimized example:

```vhd
package timing_pkg is
  type frequency is range real'high;
end package;
```

nickg, 2024-11-02:
> @nickg I don't really know how that should be fixed. The following dirty hack works, but I'm unsure what effects it causes in the depths of other codepaths.

This code is actually illegal, I misinterpreted the 2008 rule changes. It's the same as #1022 and I'm working on a fix.

NikLeberg, 2024-11-07:
Reproducer for error from file `d87cece15bccf62714ce1bdd82abbcdacc5f29bc31060c1dc3a56bb91295073b` with the following more minimized example:

```vhd
package foo is
end package;

package body foo is
	procedure bar is
	begin
		report ""                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        ;                                                                                                                                                                                                                                                                                                                     natural ;
	end procedure;
end package body;
```

Note: That line is _very_ long.

NikLeberg, 2024-11-08:
Reproducer for error from file `a3855bb94f65948ac0f8ca3555d79a537faeeb4cf8fd8273c3dbdea2e586c70f` with the following more minimized example:

```vhd
package numeric_system_pkg is
	impure function addes
end hello;

architecture behav of hello is
begin
	subtype quire_in_reg is ssf_type'
	subtype quire_in_mul is sss_mul_output_type;
end package;
```
Note: This is wholly not valid VHDL.

nickg, 2024-11-10:
>  I think I'd rather remove it and add --no-save flag to -a that would skip saving the working library to disk.

I've done this now so you can try adding `--no-save` to save a few milliseconds and some disk I/O.

NikLeberg, 2024-11-28:
> Then there were 5 cases that seem to hang the parser or the parser takes an abnormal long time to complete.
> [hangs.zip](https://github.com/user-attachments/files/17522192/hangs.zip)

All but one hang seems to have been resolved with #1065.

Hang from file `447e0104c3da4bc25acd5d7b20dbdb94ceed99731c823e133eafcf98a6e98d33` can be reproduced with reduced example:
```vhd
library ieee;
use ieee.std_logic_1164.all;

package hang is
  constant c0 : std_logic_vector (31 downto 0) := 32sb"1"; -- ok
  constant c1 : std_logic_vector (31 downto 0) := 33333333333333332sb"1"; -- hangs
end package;
```

nickg, 2024-12-01:
> I don't see an easy fix for the parser hang. The bit string literal seems to be valid and allowed according to the standard. Maybe we can at least add a warning? E.g.:

There's various places where I limit arrays to `INT32_MAX` elements (e.g. for signals) and anything larger than 2^32 is unlikely to work even given enough time and memory (`tree_chars` returns an `unsigned` for an example). So I'd just do something like:

```c
   if (length > INT32_MAX) {
      error_at(loc, "sorry, bit strings longer than %d elements are not supported", INT32_MAX);
      return t;
   }
```

NikLeberg, 2024-12-04:
Hi @avelure, I believe that all of the crashes you found are now fixed. The hangs as well. Could you confirm that? 
Have you continued fuzzing and have found other crashing inputs?

avelure, 2024-12-04:
Thanks for the help @NikLeberg.
I can confirm that all the cases here are fixed.
I have been doing some more fuzzing, but I wanted to wait until these were solved to avoid duplicates. I'll go through my logs to collect the new crashes.

I'll run a new campaing with `--no-save` as well. I have been trying to make the persistent fuzzing work (where the parser function is called in a loop and the input buffer is just updated), but I haven't gotten it to work yet. 

@nickg Should I make a new issue for the next fuzzing batch and then we create individual issues/PRs like now, or how do you prefer it to avoid too much mess?

NikLeberg, 2024-12-04:
> I have been trying to make the persistent fuzzing work (where the parser function is called in a loop and the input buffer is just updated), but I haven't gotten it to work yet.

I've got that persistant mode working and implemented a libFuzzer based fuzzer. Feel free to have a look over at https://github.com/NikLeberg/nvc-fuzz. It is missing a small patch I had to apply to nvc. Along the same lines as your _abort patch_, but additionally also instead of using _exit in fatal_exit I had to call pthread_exit for the threaded variant. I hope to add it soon.

It is capable of around 400 iterations every second. So still not a whole lot. But I do run it inside a docker container on WSL. And even with that it finds crashes every other minute or so.

nickg, 2024-12-05:
> @nickg Should I make a new issue for the next fuzzing batch and then we create individual issues/PRs like now, or how do you prefer it to avoid too much mess?

I think it would be best to close this one and create a separate ticket for the new round.

> It is capable of around 400 iterations every second. So still not a whole lot.

Did either of you try `--disable-llvm` to see if that affects the speed?

closed 2024-12-05 by nickg