nickg/nvc #1045: Registering VHPI ValueChange callback on field of signal with record type doesn't work

FireFox317, 2024-10-30:
I found this issue while working with cocotb and NVC.

It returns the following error:
```
     2.00ns ERROR    gpi                                VHPI Error level 3: class kind vhpiSelectedNameK is not an object declaration
                                                        FILE /home/verysecretuser/simple.vhd:11
     2.00ns ERROR    gpi                                VHPI: Unable to register a callback handle for VHPI type vhpiCbValueChange(1001)
     2.00ns ERROR    gpi                                Failed to register a value change callback
```

With the following VHDL file:
```vhdl
library IEEE;
use IEEE.std_logic_1164.all;

entity simple is
end entity;

architecture rtl of simple is
    type record_t is record
        field : std_logic;
    end record;
    signal record_signal : record_t;
begin
    record_signal.field <= '0';
end architecture;
```

And following cocotb test:
```py
import cocotb
from cocotb.triggers import RisingEdge

@cocotb.test()
async def simple_test(dut):
    await RisingEdge(dut.record_signal.field)
```

I have tested this on the latest release version 1.14.1.

Let me know if you need more information, or a way to reproduce, but thanks again for this very cool simulator!

Btw, something unrelated which I ran into while trying to make a small reproduceable is the following:
Note, if you ask cocotb to trigger on the RisingEdge of the signal itself (i.e `await RisingEdge(dut.record_signal)`), which maybe doesn't make sense at all(?), you get the following backtrace:
```
*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x7fd99f665da7] ***

[0x558272ca379d] ../src/util.c:872 signal_handler.lto_priv.0
[0x7fd9988f351f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7fd99f665da7] (python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:589 gpi_deregister_callback
[0x7fd9956fabfc] (python3.10/site-packages/cocotb/simulator.cpython-310-x86_64-linux-gnu.so) cocotb/share/lib/simulator/simulatormodule.cpp:835 _ZL10deregisterPN12_GLOBAL__N_114gpi_hdl_ObjectIP8GpiCbHdlEEP7_object
[0x7fd992f13bd9] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992ea2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd992feb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992ea2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd992feb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992ea2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd992feb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992ea2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd992feb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992ea2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd992feb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992f0b552] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) 
[0x7fd992f0bed0] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) PyObject_CallFunction
[0x7fd9957380f6] (python3.10/site-packages/cocotb/libs/libcocotb.so) cocotb/share/lib/embed/gpi_embed.cpp:326 _embed_sim_event
[0x7fd9957380f6] (python3.10/site-packages/cocotb/libs/libcocotb.so) cocotb/share/lib/embed/gpi_embed.cpp:314 _embed_sim_event
[0x7fd99f6655ff] (python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:127 _Z13gpi_embed_endv
[0x7fd9956d1060] (python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) _ZTv0_n40_N17VhpiShutdownCbHdl12run_callbackEv
[0x7fd9956d15dd] (python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiImpl.cpp:1034 handle_vhpi_callback
[0x558272db5d20] ../src/vhpi/vhpi-model.c:1553 vhpi_global_cb
[0x558272c9fc82] ../src/rt/model.c:432 model_run
[0x558272c9fc82] ../src/rt/model.c:857 run_cmd
[0x558272c9af44] ../src/nvc.c:2117 process_command
[0x558272c9e169] ../src/nvc.c:533 elaborate
[0x558272c9ae99] ../src/nvc.c:2115 process_command
[0x558272c96e16] ../src/nvc.c:2257 main

nvc 1.14.1 (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]
```

cameronweston, 2024-10-31:
I believe I'm getting a similar backtrace error for a test that was working on 1.14. 

```
*** Caught signal 11 (SEGV_MAPERR) [address=0x8, ip=0x55b78857e3a6] ***

[0x55b788458e9d] 
[0x7fd6cc50351f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x55b78857e3a6]
[0x55b78857d8a4]
[0x55b78857e0bb]
[0x55b78857e451]
[0x55b78857d8a4]
[0x55b7885767ba]
[0x55b788577b32]
[0x55b78857861b]
[0x55b78857b5f2] vhpi_handle_by_name
[0x7fd6c79f56dc] (python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiImpl.cpp:541 _ZN8VhpiImpl19native_check_createERKSsP9GpiObjHdl
[0x7fd6d34747a4] (python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:267 _ZL23gpi_get_handle_by_name_P9GpiObjHdlRKSsP16GpiImplInterface
[0x7fd6d34749b5] (python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:340 gpi_get_handle_by_name
[0x7fd6c92ea6c5] (python3.10/site-packages/cocotb/simulator.cpython-310-x86_64-linux-gnu.so) cocotb/share/lib/simulator/simulatormodule.cpp:681 _ZL18get_handle_by_namePN12_GLOBAL__N_114gpi_hdl_ObjectIP9GpiObjHdlEEP7_object
[0x7fd6c492353d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c491b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b49c7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c491b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c4987da1] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c4968f2a] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) PyObject_GetAttr
[0x7fd6c48aeb02] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c492f4f8] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b1966] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c492f4f8] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c493056c] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c4923ddb] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c491b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b1467] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c491b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b163d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c48b163d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7fd6c49fb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c491a3f3] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7fd6c491a882] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) PyObject_CallFunctionObjArgs
[0x7fd6c9377f60] (python3.10/site-packages/cocotb/libs/libcocotb.so) cocotb/share/lib/embed/gpi_embed.cpp:302 _embed_sim_init
[0x7fd6d3473618] (python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:123 _Z14gpi_embed_initiPKPKc
[0x7fd6c79f0fa1] (python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiCbHdl.cpp:888 _ZN16VhpiStartupCbHdl12run_callbackEv
[0x7fd6c79f15dd] (python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiImpl.cpp:1034 handle_vhpi_callback
[0x55b788577480]
[0x55b7885432b0]
[0x55b788461607]
[0x55b78845d7c6]
[0x55b788460442]
[0x55b78845d25a]
[0x55b788455617]
[0x7fd6cc4ead8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/nptl/libc_start_call_main.h:58 __libc_start_call_main
[0x7fd6cc4eae3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../csu/libc-start.c:392 __libc_start_main@@GLIBC_2.34
[0x55b788456a24]

nvc 1.14.1 (e10a10e) (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]
```

nickg, 2024-11-02:
@cameronweston that looks different from the original issue as it's crashing inside `vhpi_handle_by_name` whereas the other stack trace was inside cocotb itself. Are you able to find a minimal reproducer for it?

nickg, 2024-11-02:
@FireFox317 can you test again with the latest master branch? For the second crash With `RisingEdge` on the whole record, that seems to be internal to cocotb and the development version gives me a different error:

```
TypeError: RisingEdge requires a 1-bit LogicObject. Got HierarchyObject(SIMPLE.record_signal) of length 1
```

FireFox317, 2024-11-02:
Yes, the latest master branch works! Thanks a lot for the quick fix!

Yeah, I was also expecting cocotb to give me some error, but I then got the crash in the simulator. Thanks for figuring out that it's indeed cocotb who's fault it was. :)

nickg, 2024-11-02:
Thanks, I'll close this one now. @cameronweston can you open a separate ticket for your issue?

closed 2024-11-02 by nickg

cameronweston, 2024-11-02:
Yes I’ll try to come up with a simple example and open an issue.