nickg/nvc #1052: VHDL procedure with inout signal assigning to out signal not working as expected

FireFox317, 2024-11-02:
I was expecting the following test case to make it to the end, but instead the first assertion is hit:
```vhdl
library IEEE;
use IEEE.std_logic_1164.all;

entity reproduce is
end entity;

architecture rtl of reproduce is
    type record_t is record
        field : std_logic;
    end record;

    procedure test (
        signal sig_out: out std_logic;
        signal sig_inout : inout record_t
    ) is begin
        sig_out <= sig_inout.field;
    end procedure;
    
    signal sig_inout : record_t;
    signal sig_out : std_logic;
begin
    sig_inout.field <= '1';

    -- WORKS:
    -- sig_out <= sig_inout.field;

    -- DOESNT WORK:
    test(sig_out, sig_inout);

    process
    begin
        wait for 1 ns;
        assert sig_out = '1' severity failure;
        
        assert False severity failure;
    end process;

end architecture;
```

Assertion:
```
** Failure: 1ns+0: Assertion violation.
    > reproduce.vhd:33
    |
 33 |         assert sig_out = '1' severity failure;
    |                ^^^^^^^^^^^^^ 'U' = '1' is false
```

nickg, 2024-11-02:
This is correct behaviour: the `signal sig_inout : inout record_t` parameter creates a driver for every sub-element of the record. So `sig_inout.field` has one driver with `'1'` (from the concurrent assignment statement) and one with `'U'` (from the concurrent procedure call) and the resolved value is `'U'` which then propagates to `sig_out`. 

FWIW I get the same assertion failure in ModelSim:

```
# ** Failure: Assertion violation.
#    Time: 1 ns  Iteration: 0  Process: /reproduce/line__31 File: test.vhd
# Break in Process line__31 at test.vhd line 33
```

One way to avoid surprises like this is to use `std_ulogic` instead of `std_logic` for `field`. Then you get an error like:

```
** Fatal: (init): element FIELD of signal SIG_INOUT has multiple sources
    > test.vhd:19
    |
  9 |         field : std_ulogic;
    |         ^^^^^ element FIELD declared here
 ...
 19 |     signal sig_inout : record_t;
    |            ^^^^^^^^^ composite signal SIG_INOUT declared with unresolved type RECORD_T
 20 |     signal sig_out : std_logic;
 21 | begin
 22 |     sig_inout.field <= '1';
    |     ^^^^^^^^^^^^^^^^^^^^^^^ driven by process :reproduce:_p0
 ...
 28 |     test(sig_out, sig_inout);
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^ driven by process :reproduce:_p1
```

FireFox317, 2024-11-02:
Ah thanks, I'm sorry for the noise :) Learned something new for sure! Will test with another simulator the next time.

closed 2024-11-02 by FireFox317