nickg/nvc #1057: Use of open in signal declaration causes crash

cameronweston, 2024-11-04:
Hopefully I found what is causing the error that I mentioned in #1045. Passing open to an already constrained portion of a type declaration causes NVC to crash. Replacing open with 7 downto 0 (or any other range, causes the test to pass).

tb_simple.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_simple is
end tb_simple;

architecture tb of tb_simple is
  type my_rec is array (7 downto 0) of std_logic_vector;
  signal sigA             : my_rec(open)(0 downto 0); --replacing open with 7 downto 0 causes test to pass
begin
  sigA(0)(0) <= '0';
end tb;
```

test_simple.py
```python
import logging
import os
import cocotb
from cocotb.runner import get_runner

@cocotb.test()
async def run_test(dut):
  log = logging.getLogger("cocotb.tb")
  log.setLevel(logging.DEBUG)

  #sigC = dut.test_inst.c

def test_runner():
  tests_dir = os.path.dirname(__file__)
  sim = os.getenv("SIM", "nvc")

  runner = get_runner(sim)

  sources = [
    os.path.join(tests_dir,"tb_simple.vhd")
  ]

  runner.build(
    sources=sources,
    hdl_toplevel="tb_simple",
    always=True
  )

  runner.test(
    hdl_toplevel="tb_simple",
    test_module="test_simple"
  )
```

Error Message
```
*** Caught signal 11 (SEGV_MAPERR) [address=0x8, ip=0x55c2223fe3a6] ***

[0x55c2222d8e9d]
[0x7f41bf16d51f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x55c2223fe3a6]
[0x55c2223fd8a4]
[0x55c2223f67ba]
[0x55c2223f7b32]
[0x55c2223f8e67]
[0x55c2223c2b22]
[0x55c2222e133e]
[0x55c2222dd7c6]
[0x55c2222e0442]
[0x55c2222dd25a]
[0x55c2222d5617]
[0x7f41bf154d8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/nptl/libc_start_call_main.h:58 __libc_start_call_main
[0x7f41bf154e3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../csu/libc-start.c:392 __libc_start_main@@GLIBC_2.34
[0x55c2222d6a24]

nvc 1.14.1 (e10a10e) (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

cameronweston, 2024-11-04:
Looking at the error message, I think I simplified it to much. Here is the original file with an error message that looks more similar to what I posted in #1045.

tb_simple.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_simple is
end tb_simple;

architecture tb of tb_simple is
  constant clk50_period   : time      := 20 ns;
  signal clk              : std_logic := '0';
  signal sigA             : std_logic := '0';
begin
  clk       <= not clk        after clk50_period / 2;
  test_inst: entity work.simple
   port map(
      clk => clk,
      sigA => sigA
  );
end tb;
```

simple.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity simple is
  port (
    clk     : in std_logic;
    sigA    : in std_logic
  );
end simple;

architecture rtl of simple is
  signal c : std_logic := '0';
  type my_rec is array (7 downto 0) of std_logic_vector;
  signal chan      : my_rec(open)(0 downto 0);
begin
  c <= sigA;

end rtl;
```

test_simple.py
```python
import logging
import os
import cocotb
from cocotb.runner import get_runner

@cocotb.test()
async def run_test(dut):
  log = logging.getLogger("cocotb.tb")
  log.setLevel(logging.DEBUG)

  sigC = dut.test_inst.c

def test_runner():
  tests_dir = os.path.dirname(__file__)
  sim = os.getenv("SIM", "nvc")

  runner = get_runner(sim)

  sources = [
    os.path.join(tests_dir,"simple.vhd"),
[... 66 lines cut ...]
[0x7f2cbe0a2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f2cbe1eb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f2cbe0a163d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f2cbe1eb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f2cbe10a3f3] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f2cbe10a882] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) PyObject_CallFunctionObjArgs
[0x7f2cc08f7f60] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotb.so) cocotb/share/lib/embed/gpi_embed.cpp:302 _embed_sim_init
[0x7f2cca9f1618] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:123 _Z14gpi_embed_initiPKPKc
[0x7f2cbeff0fa1] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiCbHdl.cpp:888 _ZN16VhpiStartupCbHdl12run_callbackEv
[0x7f2cbeff15dd] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiImpl.cpp:1034 handle_vhpi_callback
[0x55d041633480]
[0x55d0415ff2b0]
[0x55d04151dcec]
[0x55d0415197c6]
[0x55d04151c442]
[0x55d04151925a]
[0x55d041511617]
[0x7f2cc3a68d8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/nptl/libc_start_call_main.h:58 __libc_start_call_main
[0x7f2cc3a68e3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../csu/libc-start.c:392 __libc_start_main@@GLIBC_2.34
[0x55d041512a24]

nvc 1.14.1 (e10a10e) (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

nickg, 2024-11-06:
```vhdl
  signal sigA             : my_rec(open)(0 downto 0); --replacing open with 7 downto 0 causes test to pass
```

Replacing `open` with `7 downto 0` here is actually invalid, but it wasn't detected before. I've fixed that as well as the VHPI crash.

closed 2024-11-06 by nickg

cameronweston, 2024-11-25:
I installed the latest 1.14.2 and nvc is still crashing.

```
*** Caught signal 11 (SEGV_MAPERR) [address=0x8, ip=0x55a30a4036c6] ***

[0x55a30a2ddead]
[0x7f380e5e451f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x55a30a4036c6]
[0x55a30a402bc4]
[0x55a30a3fbada]
[0x55a30a3fce52]
[0x55a30a3fd93b]
[0x55a30a400912] vhpi_handle_by_name
[0x7f380b3356dc] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiImpl.cpp:541 _ZN8VhpiImpl19native_check_createERKSsP9GpiObjHdl
[0x7f38155557a4] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:267 _ZL23gpi_get_handle_by_name_P9GpiObjHdlRKSsP16GpiImplInterface
[0x7f38155559b5] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:340 gpi_get_handle_by_name
[0x7f380b3ea6c5] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/simulator.cpython-310-x86_64-linux-gnu.so) cocotb/share/lib/simulator/simulatormodule.cpp:681 _ZL18get_handle_by_namePN12_GLOBAL__N_114gpi_hdl_ObjectIP9GpiObjHdlEEP7_object
[0x7f3808b1353d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b0b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa49c7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b0b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b77da1] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b58f2a] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) PyObject_GetAttr
[0x7f3808a9eb02] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808b1f4f8] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b2056c] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b13ddb] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b0b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa1467] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b0b3d7] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa163d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa2efd] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808aa163d] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) _PyEval_EvalFrameDefault
[0x7f3808beb3ae] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b0a3f3] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0)
[0x7f3808b0a882] (/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0) PyObject_CallFunctionObjArgs
[0x7f380b457f60] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotb.so) cocotb/share/lib/embed/gpi_embed.cpp:302 _embed_sim_init
[0x7f3815554618] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libgpi.so) cocotb/share/lib/gpi/GpiCommon.cpp:123 _Z14gpi_embed_initiPKPKc
[0x7f380b330fa1] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiCbHdl.cpp:888 _ZN16VhpiStartupCbHdl12run_callbackEv
[0x7f380b3315dd] (/home/cweston/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) cocotb/share/lib/vhpi/VhpiImpl.cpp:1034 handle_vhpi_callback
[0x55a30a3fc7a0]
[0x55a30a3c8600]
[0x55a30a2e6d0c]
[0x55a30a2e27d6]
[0x55a30a2e5452]
[0x55a30a2e226a]
[0x55a30a2da627]
[0x7f380e5cbd8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/nptl/libc_start_call_main.h:58 __libc_start_call_main
[0x7f380e5cbe3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../csu/libc-start.c:392 __libc_start_main@@GLIBC_2.34
[0x55a30a2dba34]

nvc 1.14.2 (1131f54) (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

nickg, 2024-11-25:
Yeah I didn't backport this one to 1.14 as the fix depended on some other VHPI changes which are only on master.

cameronweston, 2024-11-25:
Oh perfect. When do you plan on doing your next minor release?

nickg, 2024-11-25:
I'm not sure, maybe a month or two. If you need binaries/installer in the meantime they're available from the bottom of the GitHub Actions page: https://github.com/nickg/nvc/actions/runs/11986402585

commit 485ceddd (2024-11-06): Handle OPEN index constraint in VHPI. Fixes #1057